raspberrypi/rpi-eeprom #356: Ubuntu 21.04 Boot from NVME

VaheHakobyan1995, 2021-08-15:
I have the CM4 with emmc and i am trying to boot Ubuntuo Desktop 21.04 (RPi 4/400) from NVME connected to the PCIe of CM4 IO Board.  

On my emmc i have the Rapssbery Pi OS with all the updates and the bootloader is also updated:

pi@raspberrypi:~ $ rpi-eeprom-config
...
BOOT_ORDER=0xf25416
...

Do you have any thoughts why it doesn't boot from nvme? If i flash Ubuntu on both, nvme and emmc, it boots sometimes from nvme. But haveing raspberry pi OS on emmc i couldn't boor from nvme. 

Thanks!

peterharperuk, 2021-08-15:
Nothing immediately comes to mind. Can you enable logs and grab them from the UART?

VaheHakobyan1995, 2021-09-01:
@peterharperuk no, i don't even know how to do it :(

lurch, 2021-09-01:
If it "boots sometimes", could it be a power-supply issue whereby the NVMe drive isn't getting enough power to function properly? :shrug: 

https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#raspberry-pi-4-bootloader-configuration contains information about the BOOT_UART setting.

peterharperuk, 2021-09-01:
What do you see on the screen when it fails to boot? When it succeeds - what is the output of "sudo blkid" - this would be useful to confirm that it's actually using nvme. As @lurch says - it could be a hardware problem so the make and model of the drive and adapter might be interesting

peterharperuk, 2021-09-01:
This contains some further guidance for trouble shooting issues https://github.com/raspberrypi/rpi-eeprom/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

timg236, 2021-10-11:
Closed due to inactivity