raspberrypi/rpi-eeprom #357: pi400 WAKE_ON_GPIO not working

chasgames, 2021-08-22:
Hello

I'm trying to get a momentary button working with the pi400 on GPIO3, shutdown is working fine, but can't boot the system back up, which lead me to the bootloader release notes relating to this issue on the pi400.

"...If a button on GPIO3 really is required then it can be re-enabled by setting WAKE_ON_GPIO=2 ..."

I re-configured the bootloader with WAKE_ON_GPIO=2, updated and rebooted, and with no success.

vcgencmd bootloader_config / version:
```
[all]
BOOT_UART=0
WAKE_ON_GPIO=2
POWER_OFF_ON_HALT=0

Mar 18 2021 08:54:11
version 1b43d5b6fe5b71c300563afc0548122752a0618b (release)
timestamp 1616057651
update-time 1628986450
capabilities 0x0000001f
```

Can find others with same issue on the pi400 here;
https://www.stderr.nl/Blog/Hardware/RaspberryPi/PowerButton.html
https://raspberrypi.stackexchange.com/questions/125167/unable-to-power-on-raspberry-pi-400-by-shorting-gpio-pins-5-and-6-wake-on-gpio

Any ideas?

Thanks!

lurch, 2021-08-22:
That's possibly related to the power-button feature of the Pi400 (as mentioned [here](https://www.raspberrypi.org/blog/designing-raspberry-pi-400/) ) so ping @simon-martin-rpi

pelwell, 2021-08-22:
The developer of the WAKE_ON_GPIO=2 feature will be in possession of all the necessary facts.

timg236, 2021-08-23:
Custom power buttons aren't really supported on Pi400 because the power button actually turns off the core power rail so the bootloader will not run.

If you really want to try this then you'd need to disable the power_ctrl GPIO in arch/arm/boot/dts/bcm2711-rpi-400.dts. Although, that still won't work if you do a long button press because that's handled in hardware i.e. WAKE_ON_GPIO=2 might work in some limited circumstances. Enabling BOOT_UART logging will tell you if the bootloader ran at boot

chasgames, 2021-08-23:
Ah understood, thanks for the explanation 😊