raspberrypi/rpi-eeprom #359: RPI4 not booting anymore from GPT USB3 HD after update to bootloader 210706

callegar, 2021-09-07:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model:  RPI 4b Rev. 1.2
* Board revision (cat /proc/cpuinfo | grep Revision)  C03112
* Operating system version .  Manjaro
* Details of any hardware attached e.g. links to USB  USB 3 hub + USB 3 Toshiba HD
* Photo of the HDMI diagnostics screen, UART trace.  Issue worked around / Not repeatable

**Describe the bug**
A clear and concise description of what the bug is.

Used to have an RPI4 that was booting fine from an external USB3 HD partitioned using GTP. After the firmware upgrade to 210706 the system stopped booting reporting inability to read partition as FAT.

Tried changing the partitioning using parted to create a hybrid MBR/GPT setup.

After this the system boots again.

**To Reproduce**
Steps to reproduce the behavior:

Use bootloader 210706 with a GPT formatted disk.

**Expected behaviour**
A clear and concise description of what you expected to happen.

System should boot as it used to with previous bootloader.

**Bootloader version and configuration**
Please include the bootloader version and config.
```bash
vcgencmd bootloader_version  c258ef8fe1d ...
vcgencmd bootloader_config  untouched from default
```

**SD card boot (please complete the following information):**  
 - SD card type N/A
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.   See above

**USB boot (please complete the following information):**  
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'

**Network boot (please complete the following information):**  
Network boot bug normally require one or more of the following log types. [PiServer](https://github.com/raspberrypi/piserver) is the officially supported network boot server.

 - DHCP server configuration files e.g. dnsmasq.conf
 - Wireshark binary packet capture
 - UART logs with `uart_2ndstage=1` set in `config.txt`

**NVMe boot (please complete the following information):**

```bash
sudo apt-get install nvme-cli
sudo nvme list
sudo nvme id-ctrl -H /dev/nvme0
sudo nvme list-ns /dev/nvme0
sudo nvme id-ns -H /dev/nvme0 --namespace-id=1
```

**Additional context**
Add any other context about the problem here. 

The [Bootloader configuration](https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#raspberry-pi-4-bootloader-configuration) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

timg236, 2021-09-07:
You haven't provided any logs or details of the GPT partition table.

callegar, 2021-09-07:
Unfortunately, I cannot reproduce the problematic partition table anymore, since when the issue appeared I have followed instructions on the forum (https://www.raspberrypi.org/forums/viewtopic.php?t=241914#p1476111) suggesting the usage of a hybrid MBR/GPT which in fact fixed my issue.

Best that I can do is to copy here the terminal text since by luck I still have it on the terminal backtrace of the machine on which I have modified the disk partition table:

This was the original partition table detected with `fdisk`
```
Disk /dev/sdb: 465,78 GiB, 500107859968 bytes, 976773164 sectors
Disk model: External USB 3.0
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 6D5AA748-340E-44FB-A48B-BF945F89F367

Device      Start       End   Sectors   Size Type
/dev/sdb1    2048    440319    438272   214M Microsoft basic data
/dev/sdb2  440320 976773130 976332811 465,6G Linux filesystem
```

This was booting just fine before the last bootloader update (delivered by the rpi-eeprom-update via the stable channel). After the bootloader update it stopped booting. What I did was to use `gdisk` that reported:

```
Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
```

With the Recovery/transformation command mode of `gdisk` I have first printed the protective MBR data:
```
Disk size is 976773164 sectors (465.8 GiB)
MBR disk identifier: 0x00000000
MBR partitions:

Number  Boot  Start Sector   End Sector   Status      Code
   1                     1    976773163   primary     0xEE
```

Then I made the hybrid MBR/GPT

```
Type from one to three GPT partition numbers, separated by spaces, to be
added to the hybrid MBR, in sequence: 1
Place EFI GPT (0xEE) partition first in MBR (good for GRUB)? (Y/N): N

Creating entry for GPT partition #1 (MBR partition #1)
Enter an MBR hex code (default 07): 0c
Set the bootable flag? (Y/N): y

Unused partition space(s) found. Use one to protect more partitions? (Y/N): y
Note: Default is 0xEE, but this may confuse Mac OS X.
Enter an MBR hex code (default EE): 
```

Printing the situation then gave for the partition table
```
Disk /dev/sdb: 976773164 sectors, 465.8 GiB
Model: External USB 3.0
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 6D5AA748-340E-44FB-A48B-BF945F89F367
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 976773130
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048          440319   214.0 MiB   0700  P1
   2          440320       976773130   465.6 GiB   8300  P2
```
and  (for the MBR)
```
Disk size is 976773164 sectors (465.8 GiB)
MBR disk identifier: 0x00000000
MBR partitions:

Number  Boot  Start Sector   End Sector   Status      Code
   1      *           2048       440319   primary     0x0C
   2                     1         2047   primary     0xEE
   4                440320    976773163   primary     0xEE
```
where maybe it was a mistake on my side to get to two EE type partitions.

However the system now boots again.

Now, it is unclear to me if I was previously working with a formally unsupported setup (something that was not expected to work from start but for some reason worked). But certainly the bootloader update broke it, so I think it was worth noticing it here to report to the developers and to provide an indication to other users that might have experienced the same issue.

I also do not know if my fix/workaround is entirely correct. From a practical point of view it works.

timg236, 2021-09-07:
Thanks for the information. It's hard to know what the problem is without the bootloader or firmware error messages

It seems to work ok for me with stable bootloader-version firmware. Will leave bug open in case anyone else sees GPT regressions since this must be something more subtle than GPT not working at all

```
pi@raspberrypi:~$ vcgencmd bootloader_version
Jul  6 2021 11:44:53
version c258ef8fe1d2334a750078b17dab5e2c1a1787fc (release)
timestamp 1625568293
update-time 1631012226
capabilities 0x0000007f
pi@raspberrypi:~$ sudo vcgencmd version
Aug 31 2021 14:52:47
Copyright (c) 2012 Broadcom
version 67615e950e1e28b92dfae6303cf7a8b879a8908f (clean) (release) (start)
pi@raspberrypi:~$ sudo fdisk -l /dev/sda
GPT PMBR size mismatch (16785407 != 60063743) will be corrected by write.
Disk /dev/sda: 28.7 GiB, 30752636928 bytes, 60063744 sectors
Disk model: Ultra
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 07904148-AC93-41E9-8B32-F8344A03E766

Device       Start      End  Sectors  Size Type
/dev/sda1     2048  1050623  1048576  512M Microsoft basic data
/dev/sda2  1050624 16783359 15732736  7.5G Linux filesystem

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048         1050623   512.0 MiB   0700  boot
   2         1050624        16783359   7.5 GiB     8300

pi@raspberrypi:~$ sudo gdisk /dev/sda
GPT fdisk (gdisk) version 1.0.3

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
```

Bobberty, 2021-09-12:
Just to add here.  I have the same problem.  Pure GPT partition is not seen by the 210706 update.

timg236, 2021-09-12:
Please capture the error log and details of partition layout from gdisk. We can't guess the details of your particular setup.

Bobberty, 2021-09-13:
SD part info from a different machine.

Interesting discovery.....  The GUID for the GPT DISK UUID reported by the RPI log file is different than the fdisk output.
FDISK Output:
C2684EE8-AC34-4087-9DC9-9F6EF4800DB0
RPI Log Output:
e84e68c2-34ac-8740-9dc9-9f6ef4800db0

Notice that the first, second, third and fourth number groups have the octets reversed.  The fifth set is proper.
Fun question, should all octets be reversed or should they all match.  The FDISK was performed on an Intel Linux boxen.

Further, assuming that the RPI log reports the GPT partition, the UUID doesn't seem to match anything.

fdisk --list:
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: C2684EE8-AC34-4087-9DC9-9F6EF4800DB0

Device            Start      End  Sectors  Size Type
/dev/mmcblk0p1     2048  1050623  1048576  512M EFI System
/dev/mmcblk0p2  1050624 60237823 59187200 28.2G Linux filesystem
/dev/mmcblk0p3 60332032 62332927  2000896  977M Linux swap
/dev/mmcblk0p4 60237824 60332031    94208   46M Microsoft basic data

blkid:
...
/dev/mmcblk0p1: SEC_TYPE="msdos" UUID="A70A-533E" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="dcdf4867-c835-4466-8e4b-6315ef3f02c5"
/dev/mmcblk0p2: UUID="a4e6ac99-98a1-4384-ba68-0672d863cc34" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="0b3a7018-d3ee-48be-b076-a41489fd202d"
/dev/mmcblk0p3: UUID="33bc767b-3994-4355-a466-f2b4482287b5" TYPE="swap" PARTUUID="d7e84cc5-c1d1-424b-9d1b-62556a306c44"
/dev/mmcblk0p4: UUID="149E-E8CF" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="023fad4f-eacb-4ab1-95eb-ce817c5313bb"
/dev/sda3: PARTLABEL="Microsoft reserved partition" PARTUUID="c77134e3-fd46-40da-a48f-ba4fafed4757"
...

netconsole:
RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [16]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,62333951 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT e84e68c234ac87409dc99f6ef4800db0 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
PM_RSTS: 0x00001000
GPT e84e68c234ac87409dc99f6ef4800db0 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
Boot mode: RESTART (0f) order 0
Restart 0 max -1
Boot mode: SD (01) order f
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [16]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,62333951 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT e84e68c234ac87409dc99f6ef4800db0 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
PM_RSTS: 0x00001000
GPT e84e68c234ac87409dc99f6ef4800db0 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
Boot mode: RESTART (0f) order 0

pelwell, 2021-09-13:
> RPI Log Output:
e84e68c2-34ac-8740-9dc9-9f6ef4800db0

Unless we are looking in different places, the Pi log actually says:
> GPT e84e68c234ac87409dc99f6ef4800db0
 
which is subtly different. Without the '-'s it's a simple byte-wise hexadecimal dump. The first half is meant to be interpreted as little-endian integers, making 0xc2 the most significant byte, 0xe8 the least significant byte, and the entire first word 0xc2684ee8.

Bobberty, 2021-09-13:
Ok...  Little Endian vs Big Endian. Got it.  Still not booting from GPT partition on SD Card.

timg236, 2021-09-21:
Do you ever see the 'Trying partition 0' log entry? It's possible that those messages were dropped in netconsole but it's slightly different to what I see in UART logs.

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c258ef8f DATE: Jul  6 2021 TIME: 11:44:56 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1625568293 0xab5a4158 0x00b03115 0x00134f15
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x1 total-size: 16 Gbit 3200
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [2]
CID: 00275048534432353630da1588bf0107
CSD: 400e00325b590007511f7f800a400000
SD: bus-width: 4 spec: 2 SCR: 0x02458003 0x01000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,491028479 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
GPT 5485bf9a57cdb94d8e3e9331bb13e204 000000001 01d447fff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 1024 c-count 130812 c-size 8 r-dir 2 r-sec 0 r-entries 0
FAT32 clusters 130812
PM_RSTS: 0x00001000
Trying partition: 0
GPT 5485bf9a57cdb94d8e3e9331bb13e204 000000001 01d447fff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0001007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 1024 c-count 130812 c-size 8 r-dir 2 r-sec 0 r-entries 0
FAT32 clusters 130812
Read config.txt bytes     1784 hnd 0x00000000
Read start4.elf bytes  2228768 hnd 0x00000000
Read fixup4.dat bytes     5446 hnd 0x00000000
Firmware: d7f29d96450abfc77cd6cf011af1faf1e03e5e56 Apr 30 2021 13:45:52
0x00b03115 0x00000000 0x000000ff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

timg236, 2021-09-23:
@callegar @Bobberty 

The latest beta release fixes the display of GPT partition type GUIDs and adds more logging to hopefully determine why the bootloader is ignoring your EFI partition.

Please could you give this beta release a try? It's not in APT yet so to install

```
wget https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/pieeprom-2021-09-23.bin
sudo rpi-eeprom-config --edit pieeprom-2021-09-23.bin
sudo reboot
```

Bobberty, 2021-09-27:
I do see the Trying partition 0 errors on the console.
It seems that the same problem persists.  And with the beta there is an added hitch...
The NETCONSOLE seems never initialize.

timg236, 2021-09-27:
Netconsole should be working again in https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/pieeprom-2021-09-27.bin

Bobberty, 2021-09-27:
Seems to fail with a 'Matched GUID bootable-part-idx 0 want partition 0'
Current Failed Logfile with eeprom 2021-09-27:

RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [16]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,4294967295 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [16]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,4294967295 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
Restart 0 max -1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [17]

timg236, 2021-09-27:
Interesting, it looks like it accepts it as a GPT partition but not a FAT partition.

If you could use 'dd' to capture the first chunk of the disk it would be easier to debug
e.g

dd if=/dev/sdd of=sd-start.img bs=512 count=32831

Bobberty, 2021-09-27:
In my troubleshooting I may have come across this before.  Does the RPI bootloader require a proper 32bit FAT or will 12/16 FAT work as well?

Bobberty, 2021-09-27:
Link to a 16MB file with the requested DD....
https://ki.tc/f/brambles.foreshadows.biodegradable

timg236, 2021-09-27:
The bootloader supports FAT12 (only recently), FAT16 and FAT32 but was losing the default FAT type in this case for the EFI system type within the GPT partition. Please could you give this experimental fix a try.
[pieeprom-efi-fat-type.zip](https://github.com/raspberrypi/rpi-eeprom/files/7237850/pieeprom-efi-fat-type.zip)

Bobberty, 2021-09-27:
I've got another task...  I'll do this within the hour.

Bobberty, 2021-09-27:
Had the system boot through the EDKII on the first shot.  EDKII didn't load the OS. 
Rebooted and found a good long rainbow.  Nothing past the rainbow.
With Ethernet(Netconsole) unplugged, the diagnostic looks good, then seems to try and boot the EDKII and hits the rainbow.
With Ethernet(Netconsole) plugged in, the system goes straight to rainbow with the below console log.

Experimental 2021-09-27:

RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [16]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,4294967295 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
FAT16 clusters 8167
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
FAT16 clusters 8167
Stopping network

timg236, 2021-09-27:
It might be worth trying the firmware from here https://github.com/raspberrypi/firmware/issues/1465#issuecomment-814005398

The other possibility is start.elf not supporting fat16 from within an EFI partition in native GPT

Is the EDK2 trying to reboot with a different partition id?

Bobberty, 2021-09-27:
EDKII sees the ESP(fat 12/16/32) partitions when using netboot.
The experimental start4.elf/fixup4 reacts the same way when used with the experimental firmware (Rainbows forever)

Bobberty, 2021-09-27:
Just rolled back to the 2021-09-27 beta with the start4.elf/fixup.  Back to not finding the proper ESP Fat.

timg236, 2021-09-27:
AFAIK the EFI partitions are assumed to be FAT32 by start4.elf

timg236, 2021-09-28:
If you use pieeprom-efi-fat-type.zip and this attached firmware it should boot.
[fw-gpt-fat16-efi.zip](https://github.com/raspberrypi/rpi-eeprom/files/7243341/fw-gpt-fat16-efi.zip)

Not sure when it will make it into an official release and I think the short term advice is to use FAT32 for EFI partitions.

Bobberty, 2021-09-28:
Failed...  And to add a little salt to the wound, I set the boot order to 0xf2.  So, I have may have successfully created a permanent only boot system.  Any suggestions on correcting my *slight* oversight?

lurch, 2021-09-28:
[RPi Imager](https://www.raspberrypi.org/software/) contains a "standalone option" for flashing the EEPROM from a blank SD card which will reset all (eeprom) config values to their default settings. But after doing that you'll obviously need to re-flash any beta EEPROM image / config you require.

Bobberty, 2021-09-28:
Gotta love the RPI...  I have a 64bit Linux that can netboot through EDKII.  Spent some time trying to get the vchiq running.  Then threw the towel at the rip and dropped the /boot from a working SD card into the tftp server, while putting the SD card in the socket.  Worked like a charm.

On another note, replacing the start4/fixup4 and using the experimental firmware failed to the rainbow again.
Below is the log for attempting to boot....

RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [17]
SD: bus-width: 4 spec: 2 SCR: 0x02358447 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,4294967295 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
FAT16 clusters 8167
GPT 593b2bdf4d798a48972b19dd0d9644e8 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 c12a7328-f81f-11d2-ba4b-00a0c93ec93b 000000040..00000803f
Matched GUID bootable-part-idx 0 want partition 0
FAT16 clusters 8167
Stopping network

timg236, 2021-09-29:
Start.elf has always assumed that FAT partitions within GPT are FAT32 when loaded from sdcard. There’s a future tweak to make that legacy fs code probe the fat type properly

timg236, 2021-09-30:
The latest rpi-update firmware should now support FAT16 within GPT ESD partitions

timg236, 2021-10-05:
With the latest rpi-update + beta bootloader release this should now work.

timg236, 2021-10-11:
Fixed in latest beta

sallevan, 2021-11-24:
> Not sure when it will make it into an official release and I think the short term advice is to use FAT32 for EFI partitions.

My 2p:
```
"The EFI firmware must support the FAT32, FAT16, and FAT12 variants of the EFI file system.
What variant of EFI FAT to use is defined by the size of the media. The rules defining the
relationship between media size and FAT variants is defined in the specification for the EFI
file system".
```

UEFI Specification, version 2.5, page 537, paragraph 12.3.1.1.

timg236, 2021-11-24:
It’s not UEFI