raspberrypi/rpi-eeprom #361: RPI4 4GB won't enumerate JMicron 567 IC based external HDD

borisbme, 2021-09-21:
Hi!
I have an issue with the RPI4 and JMicron 567 based external HDDs

Some data first:

I boot from a 16 GB SD card
NAME        FSTYPE LABEL  UUID                                 FSAVAIL FSUSE% MOUNTPOINT
mmcblk0                                                                       
├─mmcblk0p1 vfat   boot   5DE4-665C                               204M    19% /boot
└─mmcblk0p2 ext4   rootfs 7295bbc3-bbc2-4267-9fa0-099e10ef5bf0   12.2G    11% /

The OS is: 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

Kernel verison:
Linux raspberrypi 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux

The EEPROM version:
BOOTLOADER: up to date
CURRENT: Tue  6 Jul 10:44:53 UTC 2021 (1625568293)
LATEST: Tue  6 Jul 10:44:53 UTC 2021 (1625568293)
RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
Use raspi-config to change the release.

VL805_FW: Using bootloader EEPROM
VL805: up to date
CURRENT: 000138a1
LATEST: 000138a1

The EEPROM config:
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
USB_MSD_PWR_OFF_TIME=0

The UART Boot log:
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.09.21 23:51:25 =~=~=~=~=~=~=~=~=~=~=~=
[ 1305.099741] reboot: Restarting system

PM_RSTS: 0x00000020
RPi: BOOTLOADER release VERSION:c258ef8f DATE: Jul  6 2021 TIME: 11:44:56 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1625568293 0xbe3321ed 0x00c03114 0x000cb235
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 1
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [19]
CID: 00035344534331364780ac93cb35012c
CSD: 400e00325b59000076b27f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358443 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,30583808 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
[... 376 lines cut ...]
[  195.984977] Btrfs loaded, crc32c=crc32c-generic
[  195.986398] BTRFS: device fsid 4d356723-64bb-4fec-b114-e067e5b07aa7 devid 2 transid 583 /dev/sdb scanned by systemd-udevd (979)

[  242.281897] usb 2-1: new SuperSpeed Gen 1 USB device number 38 using xhci_hcd
[  242.303356] usb 2-1: New USB device found, idVendor=152d, idProduct=0578, bcdDevice= 1.00
[  242.303376] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  242.303389] usb 2-1: Product: USB3.0 External HDD
[  242.303401] usb 2-1: Manufacturer: JMicron
[  242.303413] usb 2-1: SerialNumber: 0000AB1236AA
[  242.318941] scsi host2: uas
[  242.320470] scsi 2:0:0:0: Direct-Access     WDC WD40 EFZX-68AWUN0     8101 PQ: 0 ANSI: 6
[  242.322131] sd 2:0:0:0: Attached scsi generic sg2 type 0
[  247.620066] sd 2:0:0:0: [sdc] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[  247.620088] sd 2:0:0:0: [sdc] 4096-byte physical blocks
[  247.620333] sd 2:0:0:0: [sdc] Write Protect is off
[  247.620351] sd 2:0:0:0: [sdc] Mode Sense: 53 00 00 08
[  247.620809] sd 2:0:0:0: [sdc] Disabling FUA
[  247.620827] sd 2:0:0:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  247.621610] sd 2:0:0:0: [sdc] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[  247.662294] sd 2:0:0:0: [sdc] Attached SCSI disk
[  247.832007] BTRFS: device fsid 4d356723-64bb-4fec-b114-e067e5b07aa7 devid 1 transid 583 /dev/sdc scanned by systemd-udevd (1005)

And I do an lsusb and the result is:
Bus 002 Device 058: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 059: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge

lurch, 2021-09-22:
Have you tried disabling UAS? https://github.com/raspberrypi/rpi-eeprom/issues/360

timg236, 2021-09-22:
It sounds like the JMicron doesn't handle the USB disconnect on reset properly. You could try forcing a long power off by adding USB_MSD_PWR_OFF_TIME=5000.
Alternatively, you can try adding a powered hub.
Closing and hardware issue.

closed 2021-09-22 by timg236