raspberrypi/rpi-eeprom #367: Raspberry Pi 4 B does not leave HDMI rainbow screen (no longer boots Linux)

stapelberg, 2021-09-25:
**Mandatory information**
* Raspberry Pi 4 model B Rev 1.1
* Cannot provide board revision, as my (upstream) Linux 5.14 kernel does not show the `Revision` field in `/proc/cpuinfo` (the Raspberry Pi kernel has custom patches for that apparently?)
* I’m using https://gokrazy.org/ as operating system (custom Linux system)
* No hardware attached, only power and ethernet
* The HDMI diagnostics screen never shows up, boot is stuck at the rainbow screen
* I don’t know how to enable a UART trace because I don’t have the `rpi-eeprom` tool on my system.

**Describe the bug**

Previously, I was using the rpi-eeprom files at commit https://github.com/raspberrypi/rpi-eeprom/commit/f19990103e4483ebb47be5ea897c134

I recently upgraded to the rpi-eeprom files at commit  https://github.com/raspberrypi/rpi-eeprom/commit/7e6a961a2427ca3f5ab9c5f22a249a7, and the new recovery.bin breaks booting for me.

**To Reproduce**

I’m using the following command to overwrite my SD card with the gokrazy system, see https://gokrazy.org/quickstart/ for details:
```
gokr-packer \
  -overwrite=/dev/sdx \
  -serial_console=disabled \
  github.com/gokrazy/fbstatus \
  github.com/gokrazy/hello \
  github.com/gokrazy/serial-busybox
```

I added `replace github.com/gokrazy/rpi-eeprom => /home/michael/go/src/github.com/gokrazy/rpi-eeprom` to my `go.mod` so that my working copy on disk is used as rpi-eeprom files.

In my working copy `/home/michael/go/src/github.com/gokrazy/rpi-eeprom`, I checked out https://github.com/gokrazy/rpi-eeprom/pull/15 which brings in the new `recovery.bin`.

The release notes say that a bug with large FAT partitions was fixed — perhaps my setup is triggering a new edge case with the new code? Can you share what was changed specifically?

**Expected behaviour**

My Raspberry Pi 4 B should boot :)

**SD card boot (please complete the following information):**  

I’m booting from a SanDisk Ultra 16 GB A1.

This is the MBR partition table:
```
arch# fdisk -t mbr -l /dev/sda 
Disk /dev/sda: 14,84 GiB, 15931539456 bytes, 31116288 sectors
Disk model: SD  Transcend   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x73b10a8b

Device     Boot Start    End Sectors  Size Id Type
/dev/sda1  *     8192 212991  204800  100M  c W95 FAT32 (LBA)
/dev/sda2           1   8191    8191    4M ee GPT

Partition table entries are not in disk order.

```

This is the GPT partition table:
```
arch# fdisk -l /dev/sda
Disk /dev/sda: 14,84 GiB, 15931539456 bytes, 31116288 sectors
Disk model: SD  Transcend   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 60C24CC1-F3F9-427A-8199-73B10A8B0000

Device       Start      End  Sectors  Size Type
/dev/sda1     8192   212991   204800  100M EFI System
/dev/sda2   212992  1236991  1024000  500M Linux root (ARM-64)
/dev/sda3  1236992  2260991  1024000  500M Linux filesystem
/dev/sda4  2260992 31116253 28855262 13,8G Linux filesystem
```

https://github.com/gokrazy/tools/commit/bea67498b515201ff669f4aa7f3aa0b8476e7ce6#diff-bbf749dae63236ca522b58d7f6c0782af6451dc8ebbb54fe48647ed0a0f7237e contains more details about the gokrazy partitioning scheme (link goes to the implementation, and the commit message at the very top is also relevant).

peterharperuk, 2021-09-25:
We just reduced the size of recovery.bin so that it fits in 3 cluster entries if the cluster size is 32K. Does recovery.bin exist in the same location as start4.elf? It should only be there when doing an eeprom update.

timg236, 2021-09-25:
If it's showing the rainbow splash screen then the firmware has loaded start.elf so this isn't an rpi-eeprom issue.

stapelberg, 2021-09-25:
> We just reduced the size of recovery.bin so that it fits in 3 cluster entries if the cluster size is 32K. 

Hmm, perhaps there is an edge case in my code related to padding or similar that’s triggered by the new file size. Will take a look.

> Does recovery.bin exist in the same location as start4.elf? It should only be there when doing an eeprom update.

For simplicity, I’m just always placing recovery.bin there, even if the upgrade is then skipped.

> If it's showing the rainbow splash screen then the bootloader has loaded start.elf so this isn't an rpi-eeprom issue.

But then how come that the only change between working/broken is the new recovery.bin? It’s definitely the trigger for this issue, even if the problem turns out to be located somewhere else…

timg236, 2021-09-25:
It sounds like you have upgraded the start.elf file or Linux. recovery.bin by itself will just run, see there is no upgraded and rename itself to recovery.000 before resetting the chip.

stapelberg, 2021-09-26:
> It sounds like you have upgraded the start.elf file or Linux.

I understand why it sounds like that, but I have double-checked and can assure you that the only file system change is recovery.bin (the md5sum of all other files stays identical).

>  recovery.bin by itself will just run, see there is no upgraded and rename itself to recovery.000 before resetting the chip.

Yes. Unfortunately, this is exactly where I’m seeing data corruption with the new `recovery.bin`.

To rule out any errors in my file system writing code, I have copied my /boot files into a tar archive and am now using the Linux FAT driver to create the FAT file system on my SD card:
```
% sudo mkfs.vfat /dev/sda1
% pmount sda1
% cd /media/sda1
% tar xf /tmp/boot.tar
% cd
% pumount sda1
```

## working `recovery.bin`

Here is a diff of the root directory entries with the old, working `recovery.bin`:

```diff
--- /tmp/workingrecovery-beforeboot.txt	2021-09-26 11:04:12.883047772 +0200
+++ /tmp/workingrecovery-afterboot.txt	2021-09-26 11:04:57.959147842 +0200
@@ -104,9 +104,9 @@
 00000670  6f 00 6d 00 2e 00 73 00  69 00 00 00 67 00 00 00  |o.m...s.i...g...|
 00000680  70 69 65 65 70 72 6f 6d  73 69 67 01 00 00 00 00  |pieepromsig.....|
 00000690  00 00 00 00 00 00 1c 37  39 53 90 7a 41 00 00 00  |.......79S.zA...|
-000006a0  41 72 00 65 00 63 00 6f  00 76 00 0f 00 07 65 00  |Ar.e.c.o.v....e.|
+000006a0  e5 72 00 65 00 63 00 6f  00 76 00 0f 00 07 65 00  |.r.e.c.o.v....e.|
 000006b0  72 00 79 00 2e 00 62 00  69 00 00 00 6e 00 00 00  |r.y...b.i...n...|
-000006c0  72 65 63 6f 76 65 72 79  62 69 6e 01 00 00 00 00  |recoverybin.....|
+000006c0  52 45 43 4f 56 45 52 59  30 30 30 01 00 00 00 00  |RECOVERY000.....|
 000006d0  00 00 00 00 00 00 55 48  3a 53 91 7a c0 9f 01 00  |......UH:S.z....|
 000006e0  41 76 00 6c 00 38 00 30  00 35 00 0f 00 de 2e 00  |Av.l.8.0.5......|
 000006f0  62 00 69 00 6e 00 00 00  ff ff 00 00 ff ff ff ff  |b.i.n...........|
```

## broken `recovery.bin`

And here is a diff of the root directory entries before/after boot with the broken `recovery.bin`:

```diff
--- /tmp/brokenrecovery-rewritten-beforeboot.txt	2021-09-26 12:14:18.199863515 +0200
+++ /tmp/brokenrecovery-rewritten-afterboot.txt	2021-09-26 12:15:02.259307410 +0200
@@ -60,7 +60,7 @@
 000003b0  64 00 61 00 74 00 00 00  ff ff 00 00 ff ff ff ff  |d.a.t...........|
 000003c0  46 49 58 55 50 20 20 20  44 41 54 20 00 50 83 51  |FIXUP   DAT .P.Q|
 000003d0  3a 53 3a 53 00 00 e6 30  36 53 87 00 6f 1c 00 00  |:S:S...06S..o...|
-000003e0  41 66 00 69 00 78 00 75  00 70 00 0f 00 67 5f 00  |Af.i.x.u.p...g_.|
+000003e0  e5 66 00 69 00 78 00 75  00 70 00 0f 00 67 5f 00  |.f.i.x.u.p...g_.|
 000003f0  64 00 62 00 2e 00 64 00  61 00 00 00 74 00 00 00  |d.b...d.a...t...|
 00000400  46 49 58 55 50 5f 44 42  44 41 54 20 00 50 83 51  |FIXUP_DBDAT .P.Q|
 00000410  3a 53 3a 53 00 00 e6 30  36 53 8b 00 1b 28 00 00  |:S:S...06S...(..|
@@ -80,9 +80,9 @@
 000004f0  6f 00 6d 00 2e 00 75 00  70 00 00 00 64 00 00 00  |o.m...u.p...d...|
 00000500  50 49 45 45 50 52 4f 4d  55 50 44 20 00 51 83 51  |PIEEPROMUPD .Q.Q|
 00000510  3a 53 3a 53 00 00 1c 37  39 53 9b 00 00 00 08 00  |:S:S...79S......|
-00000520  41 72 00 65 00 63 00 6f  00 76 00 0f 00 df 65 00  |Ar.e.c.o.v....e.|
+00000520  e5 72 00 65 00 63 00 6f  00 76 00 0f 00 df 65 00  |.r.e.c.o.v....e.|
 00000530  72 00 79 00 2e 00 62 00  69 00 00 00 6e 00 00 00  |r.y...b.i...n...|
-00000540  52 45 43 4f 56 45 52 59  42 49 4e 20 00 51 83 51  |RECOVERYBIN .Q.Q|
+00000540  52 45 43 4f 56 45 52 59  30 30 30 20 00 51 83 51  |RECOVERY000 .Q.Q|
 00000550  3a 53 3a 53 00 00 2c 49  3a 53 9b 01 d0 4c 01 00  |:S:S..,I:S...L..|
 00000560  41 73 00 74 00 61 00 72  00 74 00 0f 00 cc 34 00  |As.t.a.r.t....4.|
 00000570  63 00 64 00 2e 00 65 00  6c 00 00 00 66 00 00 00  |c.d...e.l...f...|
```

Here is the corresponding directory listing (`ls`) diff:
[... 33 lines cut ...]
 -r-------- 1 michael michael  14K 2021-09-23 08:07 bcm2710-rpi-cm3.dtb
--r-------- 1 michael michael  27K 2021-09-23 08:07 bcm2711-rpi-4-b.dtb
+-r-------- 1 michael michael  27K 2021-09-23 08:07 bcm271~4.dtb
 -r-------- 1 michael michael  52K 2021-09-22 08:07 bootcode.bin
 -r-------- 1 michael michael  198 2021-09-26 11:09 cmdline.txt
 -r-------- 1 michael michael  103 2021-09-26 11:09 config.txt
@@ -18,7 +18,7 @@
 drwx------ 3 michael michael 2,0K 2049-01-01 01:00 loader
 -r-------- 1 michael michael   65 2021-09-25 08:56 pieeprom.sig
 -r-------- 1 michael michael 512K 2021-09-25 08:56 pieeprom.upd
--r-------- 1 michael michael  84K 2021-09-26 11:09 recovery.bin
+-r-------- 1 michael michael  84K 2021-09-26 11:09 RECOVERY.000
 -r-------- 1 michael michael 783K 2021-09-22 08:07 start4cd.elf
 -r-------- 1 michael michael 3,6M 2021-09-22 08:07 start4db.elf
 -r-------- 1 michael michael 2,2M 2021-09-22 08:07 start4.elf

```

Without its device tree file, the kernel cannot boot.

## full directory entry contents

Here are the full root directory entry hex contents for further debugging:  [dirents.tar.gz](https://github.com/raspberrypi/rpi-eeprom/files/7231192/dirents.tar.gz)

Let me know if you need any other details, but all steps to reproduce should be in this issue already :)

timg236, 2021-09-27:
Thanks! Originally, it sounded like a GPT issue I was looking in the wrong place. I've attached candidate fix for the recovery.bin which would cause start.elf to hang because it fails to get the right DTB.
Please could you give this a try
[recovery.zip](https://github.com/raspberrypi/rpi-eeprom/files/7235985/recovery.zip)
?

stapelberg, 2021-09-27:
Thank you for the fix! I can confirm that with the recovery.bin you linked to, my system boots as expected and only the recovery.bin entry is touched in the FAT directory entries :)

timg236, 2021-09-27:
@stapelberg Thanks for verifying that so quickly. We'll try to get the official fix into the rpi-eeprom beta channel this week. It's currently running through some soak tests.

timg236, 2021-10-05:
The latest rpi-eeprom beta release now contains a recovery.bin file with this fix.

closed 2021-10-05 by timg236