raspberrypi/rpi-eeprom #378: CM4: Booting from NVMe works fine, but reboot got stuck with "failed to open nvme"

ctschach, 2021-11-13:
**Mandatory information**
* Raspberry Pi model
CM4 with 4 GB Memory and no eMMC - attached to https://www.waveshare.com/wiki/CM4-IO-BASE-A
* Board revision (cat /proc/cpuinfo | grep Revision)
Revision        : c03140
* Operating system version .
Rasbian Bullseye

**Describe the bug**
The system boots without any trouble from the attached NVMe disk. No problem in running the machine, no hiccups, everything works fine. However, a simple "reboot now" got stuck which the following error on the boot console:

`root@raspberrypi:/home/pi# reboot now
[  437.212975] reboot: Restar
PM_RSTS: 0x00001020
RPi: BOOTLOADER release VERSION:c258ef8f DATE: Jul  6 2021 TIME: 11:44:56 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1625568293 0x11471181 0x00c03140 0x000c658a
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
Boot mode: NVME (06) order f1
Failed to open device: 'nvme'
Boot mode: SD (01) order f
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: RESTART (0f) order 0
Restart 0 max -1
Boot mode: NVME (06) order f1
Failed to open device: 'nvme'
Boot mode: SD (01) order f
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
`

This loops endless. The only way to restart the machine is by power-cycling. 

**Bootloader version and configuration**
Please include the bootloader version and config.
```bash
* vcgencmd bootloader_version
timestamp 1625568293
update-time 1636809771
capabilities 0x0000007f
* vcgencmd bootloader_config
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
[... 599 lines cut ...]
[    4.332495] systemd[1]: Mounted RPC Pipe File System.
[    4.338851] systemd[1]: Mounted Kernel Debug File System.
[    4.345500] systemd[1]: Mounted Kernel Trace File System.
[    4.353231] systemd[1]: Finished Restore / save the current clock.
[    4.362535] systemd[1]: Finished Create list of static device nodes for the current kernel.
[    4.373289] systemd[1]: modprobe@configfs.service: Succeeded.
[    4.381367] systemd[1]: Finished Load Kernel Module configfs.
[    4.386570] i2c /dev entries driver
[    4.392925] systemd[1]: modprobe@drm.service: Succeeded.
[    4.400119] systemd[1]: Finished Load Kernel Module drm.
[    4.417615] systemd[1]: modprobe@fuse.service: Succeeded.
[    4.424396] systemd[1]: Finished Load Kernel Module fuse.
[    4.447390] systemd[1]: Finished Load Kernel Modules.
[    4.475632] systemd[1]: Mounting FUSE Control File System...
[    4.496909] systemd[1]: Mounting Kernel Configuration File System...
[    4.539834] systemd[1]: Started File System Check Daemon to report status.
[    4.571499] systemd[1]: Starting Apply Kernel Variables...
[    4.587911] systemd[1]: Mounted FUSE Control File System.
[    4.594292] systemd[1]: Started Journal Service.

Raspbian GNU/Linux 11 raspberrypi ttyAMA0

raspberrypi login:

```

lurch, 2021-11-13:
Sounds quite similar to https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/188 ?

peterharperuk, 2021-11-16:
I've tried to reproduce this but it seems to work for me. I've tried a LITE and 8GB EMMC CM4. I'm using the waveshare board mentioned in https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/188 that has the resistor fix, but I don't know why that would be relevant. I'm using a different make of NVME disk.

peterharperuk, 2021-11-16:
@ctschach if I sent you a version of the bootloader with some extra logging - would you be prepared to update the eeprom on your CM4 using USBBOOT and repro the problem?

ctschach, 2021-11-16:
> @ctschach if I sent you a version of the bootloader with some extra logging - would you be prepared to update the eeprom on your CM4 using USBBOOT and repro the problem?

Absolutely....

peterharperuk, 2021-11-16:
Ok, first attempt... [pieeprom.original1.zip](https://github.com/raspberrypi/rpi-eeprom/files/7548452/pieeprom.original1.zip)

cd usbboot\recovery
cp pieeprom.original1.bin pieeprom.original.bin
./update-pieeprom.sh
../rpiboot.exe -d .

switch [RPI]boot switch on and connect your PC to the USB type-C interface to program the eeprom with this image. Switch off the boot switch and power cycle and repro the problem. If you can attach any UART logging that would be helpful

ctschach, 2021-11-17:
So here is what happened: I flashed the new EEPROM and booted the RPi from the NVMe card. This ended up in a kernel panic (see below). Since then the NVMe became unusable. If I boot from an SD card, I see the NVMe disk, but an fsck giving me error. Putting it into an external enclosure and attaching it to another Raspi did not work either. 

The interesting thing: It worked directly before changing the EEPROM. I don't think that this could have destroyed the NVMe disk - maybe just some unfortunate coincidences.

I've now ordered two new (and different) disks which should arrive early next week. I'll test them and report back.

´´´
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.78-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1484 SMP Thu Nov 11 16:36:06 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2072384
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=E4:5F:01:62:3B:5C vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=0507689b-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000012400000-0x0000000016400000] (64MB)
[    0.000000] Memory: 7809312K/8296448K available (10240K kernel code, 1366K rwdata, 3192K rodata, 2048K init, 883K bss, 159456K reserved, 327680K cma-reserved, 7510016K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34341 entries in 101 pages
[    0.000000] ftrace: allocated 101 pages with 4 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=1
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000034] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000094] bcm2835: system timer (irq = 26)
[    0.000746] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000767] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000789] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000807] Switching to timer-based delay loop, resolution 18ns
[    0.001070] Console: colour dummy device 80x30
[    0.001834] printk: console [tty1] enabled
[    0.001903] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001958] pid_max: default: 32768 minimum: 301
[    0.002135] LSM: Security Framework initializing
[    0.002332] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002378] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003855] cgroup: Disabling memory control group subsystem
[    0.004114] CPU: Testing write buffer coherency: ok
[    0.004589] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005814] Setting up static identity map for 0x200000 - 0x20003c
[    0.006028] rcu: Hierarchical SRCU implementation.
[    0.006972] smp: Bringing up secondary CPUs ...
[    0.008200] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.009560] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.010956] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.011116] smp: Brought up 1 node, 4 CPUs
[... 334 lines cut ...]
[... cut ...]
[    4.443358] [<c0b8e594>] (rest_init) from [<c1000ab4>] (arch_call_rest_init+0x18/0x1c)
[    4.451291]  r5:c1205048 r4:c1356068
[    4.454880] [<c1000a9c>] (arch_call_rest_init) from [<c1001098>] (start_kernel+0x568/0x59c)
[    4.463250] [<c1000b30>] (start_kernel) from [<00000000>] (0x0)
[    4.469183]  r10:30c5387d r9:410fd083 r8:2eff3500 r7:00000c42 r6:30c0387d r5:00000000
[    4.477027]  r4:c1000330
[    4.479588] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(259,2) ]---
´´´

peterharperuk, 2021-11-17:
Oh dear...
I have one of your disks on order so I'll be testing that. I really can't imagine why that bootloader would cause a problem. It's the same code as the stable build with debug logging enabled in a couple of modules.

peterharperuk, 2021-11-18:
I've got the SSD you mentioned and can reproduce the problem you were seeing.

peterharperuk, 2021-11-18:
I can repro the issue on a RPI IO board so it doesn't seem to be an issue with the Waveshare board.

ctschach, 2021-11-18:
> I can repro the issue on a RPI IO board so it doesn't seem to be an issue with the Waveshare board.

No sure if I'm allowed to say: "great that you are facing the same problem" :-)

But definitely good that it's wasn't my fault. My other two disks are still in transit. I'll definitely test them both and will report back.

peterharperuk, 2021-11-19:
The hardware is indicating a pcie error which seems to "appear" after initialisation. To recover we have to re-initialise the hardware. I have a change that fixes the issue for me - can you give it a try to confirm?
[pieeprom.original2.zip](https://github.com/raspberrypi/rpi-eeprom/files/7569564/pieeprom.original2.zip)

ctschach, 2021-11-19:
Are you sure this is the correct EEPROM? I 

I do see an "Unknown boot mode 0x6. Trying SD." in the output on the console. I though the I need boot-mode "6" for NVME boot, but looks like this eeprom you just gave me do not support it.

```
SIG pieeprom.sig b440356d2b26f3b6434ea2750239d7e8e239023dea95447092de1297549e5959 1637330510
Reading EEPROM: 524288
Writing EEPROM
++++++++++++++++++++++++++++++..++++++++++++++++++++++++++++++++++++++++++++++++++++++.........................................+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:ca381952 DATE: Nov 19 2021 TIME: 10:22:04 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1637317322 0x72d970ff 0x00d03140 0x000b33c8
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
PCI reset
PCI reset
USB xHC init failed
Boot mode: SD (01) order f6
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode:  (06) order f
Unknown boot mode 0x6. Trying SD.
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
```

peterharperuk, 2021-11-19:
So sorry to waste your time. I picked an old branch. Here's the right one, and I've actually tested it! It should show up as version e0c6d67c
[pieeprom.original3.zip](https://github.com/raspberrypi/rpi-eeprom/files/7570853/pieeprom.original3.zip)

ctschach, 2021-11-19:
Okay, that one works better...much better :-)
No problem with a reboot at all. 

But this is a different brand of NVMe card. The other one is already on it's way back to the vendor. Not sure it I get a replacement or a refund. 

```
SIG pieeprom.sig 09c6c294f0270e91b5d4f591c83090c27bc18fb1a421d9027fd2a252a86d053b 1637336099
Reading EEPROM: 524288
Writing EEPROM
++++++++++++++++++++++++++++++.................................................................................................+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:e0c6d67c DATE: Nov 19 2021 TIME: 14:17:52 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1637331471 0x72d970ff 0x00d03140 0x000c6c09
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
Boot mode: SD (01) order f6
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: NVME (06) order f
VID 0x1e0f MN KBG40ZNS512G NVMe KIOXIA 512GB
NVME on
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,124203008 type: 0x83
MBR: 0x076f500c,875479716 type: 0x83
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0 r-entries 0
FAT32 clusters 516190
PM_RSTS: 0x00001000
Trying partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0 r-entries 0
FAT32 clusters 516190
Read config.txt bytes     2090 hnd 0x00000000
Read start4.elf bytes  2241504 hnd 0x00000000
Read fixup4.dat bytes     5411 hnd 0x00000000
Firmware: b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 Oct 29 2021 10:47:33
0x00d03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021
[... 118 lines cut ...]
Firmware: b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 Oct 29 2021 10:47:33
0x00d03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[...]
``

peterharperuk, 2021-11-19:
Thanks, much appreciated. I'll keep this open until a fix is released.

peterharperuk, 2021-12-03:
The fix for this should be in released now.

closed 2021-12-03 by peterharperuk