raspberrypi/rpi-eeprom #383: Please drop hard dependency on xxd

HiassofT, 2021-12-01:
**Describe the bug**
Running `rpi-eeprom-update -a` on LibreELEC fails with an error because xxd is not installed
```
xxd not found. Try installing the xxd package.
```

xxd and also openssl only seem to be needed by rpi-eeprom-digest if RSA key signing is requested, which does not seem to be the case when just running `rpi-eeprom-update -a`

It would be nice if you could make these dependencies optional and check them only when RSA key signing is used (or even better also replace xxd with something from coreutils)

timg236, 2021-12-01:
I've changed this to a weak dependency check so that xxd/openssl are only required if RSA key signing (which is only relevant for secure-boot)

The purpose of this script is to be an easily replaceable part of the rpi-eeprom tools for people who want to use things other than openssl to manage their private keys i.e. it's a working example and if anyone wants something else they are free to fork this script :)

HiassofT, 2021-12-01:
Thanks a lot for the quick fix, it's working fine again!

closed 2021-12-01 by HiassofT

timg236, 2021-12-01:
Great, Apologies for the breakage :)

MichaIng, 2022-11-06:
@timg236 
Your commit https://github.com/raspberrypi/rpi-eeprom/commit/eb3bb76a81f0512280769a0f87b926db1cf06129 reintroduced the hard dependency without adding them to package dependencies. So in upgrade, users may now face a falling EEPROM update, which may stay unrecognised as long as it's not done manually but relied on the background update on boot.

The new signature verification still is only done in combination with a signing key, so the previous conditional check still makes sense?

timg236, 2022-11-07:
See https://github.com/raspberrypi/rpi-eeprom/issues/450#issuecomment-1305340330