raspberrypi/rpi-eeprom #388: Update Bootloader EEPROM causing RPi to not boot anything

Eshanepicfighter, 2021-12-17:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

**Mandatory information**
* Raspberry Pi model
Raspberry Pi 4 Model B

* Board revision (cat /proc/cpuinfo | grep Revision)
C0 Revision

* Operating system version .
Libreelec 19.2, Latest OSMC, Latest RPI Os 
* Details of any hardware attached e.g. links to USB 
Saberent usb to SATA adapter

* Photo of the HDMI diagnostics screen, UART trace.
Unable to provide

**Describe the bug**
After updating the rpi eeprom using RPI-EEPROM-UPDATE, I was unable to boot into any os following 

**To Reproduce**
Download and install the latest beta firmware, flash it to the pi, and try booting any os

**Expected behaviour**
The pi was able to boot into the os

**Bootloader version and configuration**
Bootloader version 2021-12-02

**USB boot:**  
Partitions do show up in both linux and windows.

lurch, 2021-12-17:
Is it only USB-booting that's stopped working? Are you still able to boot Raspberry Pi OS from an SD card?

Eshanepicfighter, 2021-12-17:
Usb Boot, Network Boot, and SD Card booting doesn't work. 

I've double checked the configuration on network booting and it works fine with my other pi's

timg236, 2021-12-17:
Please use the Raspberry Pi Imager to flash the bootloader via a spare SD card. This will reset the bootloader to the latest release.

Eshanepicfighter, 2021-12-17:
Using raspberry pi imager, I flashed the latest stable firmware to the spare sd card I had. Booted the pi, got the text of "firmware not found" on my monitor

timg236, 2021-12-17:
With USB or SD?

Eshanepicfighter, 2021-12-17:
I tried with SD, and usb. Same result

timg236, 2021-12-17:
We'll need a screenshot and or logs to show what's different in this setup.

Eshanepicfighter, 2021-12-17:
[image]

Eshanepicfighter, 2021-12-17:
That’s with the boot loader flashed onto the as card

Eshanepicfighter, 2021-12-17:
Odd... on Windows the root directory exists and is mounted correctly but on linux it's not there

timg236, 2021-12-17:
> Using raspberry pi imager, I flashed the latest stable firmware to the spare sd card I had. Booted the pi, got the text of "firmware not found" on my monitor

This photo shows a different date to the latest default release from the Raspberry Pi Imager. That's April 2021

Eshanepicfighter, 2021-12-17:
Huh.... I just clicked the misc --> bootloader --> USB (or sd card)  Boot

Eshanepicfighter, 2021-12-17:
Update: After unpluging and replugging the drive in windows, the drive appear or mount.

timg236, 2021-12-17:
> Odd... on Windows the root directory exists and is mounted correctly but on linux it's not there
> Update: After unpluging and replugging the drive in windows, the drive appear or mount.

It sounds like one of the copies of the FAT table is corrupted. Linux and the bootloader tend to look at the primary. Chkdsk will often solve this but you'll need to check that the firmware files have not been truncated.

Eshanepicfighter, 2021-12-17:
And how would check if the firmware files haven't been truncated? I just downloaded them straight from github or used rpi-eeprom-update or used the imager application

timg236, 2021-12-17:
> And how would check if the firmware files haven't been truncated? I just downloaded them straight from github or used rpi-eeprom-update or used the imager application

If you have a spare-sd card then use the Imager to flash a Raspberry Pi OS lite image to that to use a reference OS image

Eshanepicfighter, 2021-12-17:
Hmmm I currently don’t have a spare sd card but I do have a spare hard drove