raspberrypi/rpi-eeprom #389: Fails to boot using secure USB drive

SOCIT, 2021-12-19:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
pieeprom-2021-12-02.bin 2021-12-02: Promote the 2021-12-02 beta release to LATEST/STABLE
* Raspberry Pi model 4
* Board revision (cat /proc/cpuinfo | grep Revision) N/A
* Operating system version . N/A
* Details of any hardware attached e.g. links to USB  Apricorn AEIGIS Secure Key
* Photo of the HDMI diagnostics screen, UART trace. N/A

**Describe the bug**
Some kind of USB reset is sent when booting from USB.   This relocks the AEGIS drive causing the boot to fail. 

**To Reproduce**
Plug an Aegis secure key into the pi with the OS on it.  
Unlock the key
Power on the PI
Key locks
Unlock again
Pi Boots - if your timing was precise and your keylength is short. 

**Expected behaviour**
Expected the PI to boot from USB and not reset the USB bus 

**Bootloader version and configuration**
pieeprom-2021-12-02.bin 2021-12-02: Promote the 2021-12-02 beta release to LATEST/STABLE

```

**SD card boot (please complete the following information):**  
 - SD card type: N/A
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.  N/A

**USB boot (please complete the following information):**  
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'

Boots fine with any standard USB drive

lurch, 2021-12-19:
You _might_ be able to tweak this behaviour with the https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_PWR_OFF_TIME setting?

SOCIT, 2021-12-19:
> You _might_ be able to tweak this behaviour with the https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_PWR_OFF_TIME setting?

Unfortunately no, that provides no change.  

Somewhere a reset signal is issued, or something which is extremely simple to anyone who knows what it is, attempting to provide continuous power to the USB key via a powered hub or hardwiring from GPIO also didn't resolve the issue.

pelwell, 2021-12-20:
Have you successfully booted some other device from the secure drive? Resetting a USB device during boot doesn't sound unreasonable to me.

peterharperuk, 2021-12-20:
Can you confirm that you didn't have the problem with the 6 Jul release? You can downgrade like this...

sudo rpi-eeprom-update -f /lib/firmware/raspberrypi/bootloader/stable/pieprom-2021-07-06.bin

Uart output or at least a screen grab of the display would help determine if something else is going on - like USB boot being retried. It might be worth posting your eeprom config - the output from "rpi-eeprom-config"

timg236, 2021-12-20:
> Have you successfully booted some other device from the secure drive? Resetting a USB device during boot doesn't sound unreasonable to me.

Yes. USB (and PCIe) is always reset during power on phase of the bootloader and immediately prior to starting Linux. This is unavoidable so I don’t this this will be suitable as a boot device.

SOCIT, 2021-12-20:
> Yes. USB (and PCIe) is always reset during power on phase of the bootloader and immediately prior to starting Linux. This is unavoidable so I don’t this this will be suitable as a boot device.

So the drive will need to be physically unlocked twice regardless.  Is there any way to make the PI initially boot from an SD card, run through the power reset to USB/PCIe then wait for/load from USB?  

This also seems to affect the drive when it's plugged into an externally powered USB hub which I would have thought would keep the drive powered on/unlocked through the power reset.  I admittedly don't know enough about the protocol to know why this is.

timg236, 2021-12-20:
> So the drive will need to be physically unlocked twice regardless. Is there any way to make the PI initially boot from an SD card, run through the power reset to USB/PCIe then wait for/load from [...]

I’d suggest using this for user data and mounting it after boot with the OS on a normal storage device

SOCIT, 2021-12-20:
I'd prefer to keep data security as more of a priority than current hardware/software limitations.

lurch, 2021-12-20:
I guess you could have the `/boot` partition on the SD-card, but then have the entire root partition on your "secure USB device"? You'd just need to edit the `cmdline.txt` on the boot partition to tell Linux which partition to mount as root.
But I've never used any "secure" devices, so I've no idea whether this will actually work or not.

> This also seems to affect the drive when it's plugged into an externally powered USB hub which I would have thought would keep the drive powered on/unlocked through the power reset.

I guess (this isn't my area of expertise) that at the very least the USB drive has to be enumerated by the bootloader running on the VPU (in order to load `start4.elf` and `kernel.img`), and then enumerated again by the Linux kernel running on the CPU? :man_shrugging:

timg236, 2021-12-20:
The Linux USB stack and root file system mounting is independent of the firmware and bootloader. AFAIK there’s no API to tell Linux to wait for some authentication but maybe there’s something that the USB stick manufacturer supports.
Closing since this is outside the scope of the Pi4 bootloader.

closed 2021-12-20 by timg236