raspberrypi/rpi-eeprom #391: Feature request: output all options

jacobq, 2021-12-20:
It seems that the current behavior is for some options using the default value (e.g. `BOOT_ORDER`) to be omitted when running `rpi-eeprom-config`. It would be nice if all options could be captured, e.g. `rpi-eeprom-config --show-all-options pieeprom.bin > bootconf.txt`. As an example use case, I have a script that runs after boot to make sure that the bootloader is configured to try booting from USB-MSD (for recovery) before microSD card (for normal operation). Since there's no way of knowing ahead of time whether the config will have `BOOT_ORDER` or not it uses `grep` to check, adding some otherwise unnecessary complexity. This also makes me wonder what happens when new options are introduced or defaults are changed: will my system still work or will I have to discover and revert those changes? Knowing that all options are specified (and not left to default logic) in my configuration seems like it would offer reassurance for scenarios like that.

<details>
<summary>Example script</summary>

```bash
#!/bin/bash
# Update bootloader / firmware
# https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

# Raises USB boot priority above microSD card to aid in recovery
# lowest nybble to highest: 1 = SD card, 4 = USB mass storage, f = restart/loop/retry
DESIRED_BOOT_ORDER=0xf14
FORCE_UPDATE=0

BOOT_ORDER=$(vcgencmd bootloader_config|grep ^BOOT_ORDER=|awk -F= '{ print $2 }')
BOOTLOADER_DIR=/lib/firmware/raspberrypi/bootloader/stable
BOOTLOADER_BASE_IMG=pieeprom-2021-12-02.bin

while getopts n:c: option
do
    case "${option}"
        in
        o)DESIRED_BOOT_ORDER=${OPTARG};
        f)FORCE_UPDATE=1;
    esac
done

echo "FORCE_UPDATE=$FORCE_UPDATE"
echo "BOOT_ORDER=$BOOT_ORDER"
echo "DESIRED_BOOT_ORDER=$DESIRED_BOOT_ORDER"

if [[ "$FORCE_UPDATE" == "1" || "$BOOT_ORDER" != "$DESIRED_BOOT_ORDER" ]]; then
  echo "Updating firmware/bootloader BOOT_ORDER to $DESIRED_BOOT_ORDER"
  cd $BOOTLOADER_DIR
  cp $BOOTLOADER_BASE_IMG pieeprom.bin
  rpi-eeprom-config pieeprom.bin > bootconf.txt
  # ^^^ may omit BOOT_ORDER option, so we check below using grep  
  grep -qi BOOT_ORDER bootconf.txt \
      && sed -i -E "s/BOOT_ORDER=\S*/BOOT_ORDER=$DESIRED_BOOT_ORDER/g" bootconf.txt \
      || echo "BOOT_ORDER=$DESIRED_BOOT_ORDER" >> bootconf.txt
  cat bootconf.txt # Output for debug / logs
  rpi-eeprom-config --out pieeprom-custom.bin --config bootconf.txt pieeprom.bin
  rpi-eeprom-update -d -f ./pieeprom-custom.bin
  cd -
fi
```

</details>

timg236, 2021-12-20:
The script doesn’t know anything about the EEPROM properties. The defaults are covered by the online documentation

closed 2021-12-20 by timg236

lurch, 2021-12-20:
> It seems that the current behavior is for some options using the default value (e.g. BOOT_ORDER) to be omitted when running rpi-eeprom-config. It would be nice if all options could be captured

I think @jacobq might be assuming that if a value is "left at default" then it deliberately gets left out of the boot-config? Whereas the script isn't that "intelligent", it just blindly sets whatever you explicitly tell it to (regardless of whether that matches any default values or not).

>  adding some otherwise unnecessary complexity

It seems that an "easier" way of doing that might be to do something like:
```
  rpi-eeprom-config pieeprom.bin | grep -v BOOT_ORDER= > bootconf.txt
  # ^^^ deliberately omit BOOT_ORDER option
  echo "BOOT_ORDER=$DESIRED_BOOT_ORDER" >> bootconf.txt
```
?

timg236, 2021-12-20:
> I think @jacobq might be assuming that if a value is "left at default" then it deliberately gets left out of the boot-config? Whereas the script isn't that "intelligent", it just blindly sets [...]

The script is supposed to do with exactly what you tell it. 
Building a database of default options that tracks that across eeprom releases ands unnecessary complexity. You can either use the default behaviour or specify entire config according to the docs

jacobq, 2021-12-20:
> The script doesn’t know anything about the EEPROM properties. The defaults are covered by the online documentation

I guess what I am requesting then is that this tool have some kind of awareness of valid / invalid options to make things easier / less mistake-prone for a user. As somewhat of an outsider, I find it easy to get mixed-up between the various components involved in the boot process and their parameters (bootloader eeprom, kernel cmdline, config.txt, ...). If that's a big ask then obviously don't worry about it. 

@timg236 Thanks for mentioning the simpler "grep it out" approach. That is a bit better. Still, it'd be nice to be able to get the info right from the tool or at least a man page rather than have to open up a web browser and find the docs online.

timg236, 2021-12-20:
The docs are all in GitHub and can be cloned / grepped for convenience. If you look at the properties page it’s pretty much a list of properties which could be reformatted. It’s largely feature complete and I think we would want to avoid adding more properties

jacobq, 2021-12-20:
> The script is supposed to do with exactly what you tell it. Building a database of default options that tracks that across eeprom releases ands unnecessary complexity. You can either use the default behaviour or specify entire config according to the docs

How about a compromise like have a `default-bootconfig-sample.txt` shipped with the tool? It could be auto-generated just like the documentation so wouldn't add a lot of extra complexity and would still address the main goal of this request.