raspberrypi/rpi-eeprom #392: Raspberry Pi 4 8GB not booting from USB getting MSD error

krishnak, 2022-01-05:
I have booted the device with a 32GB SD card running Ubuntu 21.10 - this works well. HDMI port is working.

I installed Ubuntu 21.10 on a 1TB USB hard disk and also on a brand new 16TB USB hard disk.

When I connect the USB hard disk to the raspberry PI, the device doesn't bootup, it shows a 640x480 image in 60Hz - my tv is in PAL mode and I guess this is causing the image distortion.

The USB hard disk is mounted on USB 3 port.

After waiting for 30 minutes, I unplugged everything and inserted the  SD card, but that doesn't show any output on the monitor, so I ssh in to the device and the device lets me login and it is working except for the HDMI output.

Please advise  if there is a fix, I saw a similar post which was marked fixed almost 2 years ago

https://github.com/raspberrypi/rpi-eeprom/issues/131

[image]
[image]

peterharperuk, 2022-01-05:
> I installed Ubuntu 21.10 on a 1TB USB hard disk and also on a brand new 16TB USB hard disk.

What kind of hard drives are they?

krishnak, 2022-01-05:
Seagate Expansion 16TB Desktop External HDD - 3.5 Inch USB 3.0 for Windows and Mac with 3 yr Data Recovery Services, Portable Hard Drive (STKP16000402), Black 

I ran

```
$ vcgencmd bootloader_config
VCHI initialization failed

 $ rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
```

krishnak, 2022-01-05:
```
sudo rpi-eeprom-update 
[sudo] password for krishna: 
BOOTLOADER: up to date
   CURRENT: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
    LATEST: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

```

Do you want me to try the new stable firmware manually  ? -

pieeprom-2021-12-02.bin

peterharperuk, 2022-01-05:
Yes - you're right. Default is a symbolic link to the critical bootloader folder and that hasn't been updated since April 2021. So you'll have to switch to stable to pick up recent changes and then try setting USB_MSD_STARTUP_DELAY=5000
I think really the MSD boot mode is intended for SSD's and USB sticks rather than spinning disks.

krishnak, 2022-01-05:
I updated the firmware and added the MSD startup delay as advised. I have double checked that the bootloader has updated to december 2021 version and the configuration has the MSD startup delay. No luck, the green screen as posted before popup and then there is no video signal nor attempt to boot from HD.

krishnak, 2022-01-05:
```
$ rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
USB_MSD_STARTUP_DELAY=5000

$ rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Thursday 02 December 2021 11:08:03 AM UTC (1638443283)
    LATEST: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: version unknown. Try sudo rpi-eeprom-update
   CURRENT: 
    LATEST: 

```

Even with this setting the USB doesn't boot, however there is no HDMI output even after I re-plug in the SD card, is there an issue with the HDMI port, how do I figure it out? The pi though has booted and is accessible vis SSH. HDMI was working normally till I tried to boot with the USB hard drive.

krishnak, 2022-01-05:
Some googling has taken me to this [thread](https://www.reddit.com/r/RASPBERRY_PI_PROJECTS/comments/klsh2l/raspberry_pi_4_wont_boot_when_external_powered/)

This is has lot of similarity to my issue, my Pi doesn't boot from the SD card when the USB HD disk (seagate which has its own power supply) connected. It boots when the disk is removed.

timg236, 2022-01-05:
It sounds like the USB HD is back-powering the Pi. A powered USB hub may improve things, although, unfortunately, some USB hubs also back power the Pi.

krishnak, 2022-01-05:
Thanks, I just reformatted the hard disk completely, now it doesn't have the Ubuntu boot on it, it is a plain external powered USB hard disk with ext 4.

With this setup the PI boots from the SD card when  the USB hard disk is connected at the boot time. I have rebooted the PI few times with the SD card inserted along with the USB hard drive and it reboots every time. I guess it is not back powering issue, so I have eliminated one variable. However if the Ubuntu bootable image was present in the hard disk, the PI would not boot from the SD card unless the USB disk was disconnected and power cycled.

I will try increasing the delay to 10000 and update.

krishnak, 2022-01-06:
The issue of no HDMI output seems to be a cable issue - a cable which worked 24 hours ago seems to have broken now.

The issue with SD card unable to boot, while the USB hard disk was connected, has disappeared with the new boot loader. I also did a reformat of the USB hard disk to ext 4. Previously the Ubuntu OS image was in the harddisk, large portion of the HD appeared to be in some DOS format. 

I have not yet tested the USB hard disk boot though, as I am sure it will work as the hard disk mounts automatically after reboot using the SD card image.

closed 2022-01-06 by krishnak