raspberrypi/rpi-eeprom #393: Update

symbios24, 2022-01-06:
Hi there Happy new year
Its not really an issue i wanted to ask when you will put the latest stable release 2021-12-02 in the latest zipped fat32 disk images section for download ??
Thanks

timg236, 2022-01-06:
There's no plans to release zipped fat32 images on Github for anything except the default release which is used by the imager.

However,  you clone this repo and look in the imager directory then it's easy enough to run the scripts there and build arbitrary imager release zips.
https://github.com/raspberrypi/rpi-eeprom/tree/master/imager

timg236, 2022-01-20:
Closing - as "not going to fix" since the wrapper scripts are available to create custom images/zips. We are only going to maintain 'default' as a Github release.

closed 2022-01-20 by timg236