raspberrypi/rpi-eeprom #395: CM4 not rebooting with M.2 NVMe

DerAndereAndi, 2022-01-07:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63) forum

**Mandatory information**
* Raspberry Pi model
  CM4 with 8 GB Memory and no eMMC, no Wifi - attached to https://www.waveshare.com/wiki/CM4-IO-BASE-B
* Board revision (`cat /proc/cpuinfo | grep Revision`)
  Revision: d03140
* Operating system version
  Raspberry OS Lite Bullseye 64bit

**Describe the bug**
The system boots without any trouble from the attached NVMe disk. No problem in running the machine, no hiccups, everything works fine. However, a simple "reboot now" gets stuck after the splash screen. The only way to boot again is to unplug power, sometimes wait a bit (a few seconds), and power it on again

**Bootloader version and configuration**
Please include the bootloader version and config.
```bash
* vcgencmd bootloader_version
2021/12/02 11:08:03
version 78ec57468d6bb1dd7051698539ec814d22a98c64 (release)
timestamp 1638443283
update-time 1639478069
capabilities 0x0000007f
```

```bash
* vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```

**NVMe boot (please complete the following information):**

```bash
* sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     71DPCAAMQH71         KBG40ZNS256G TOSHIBA MEMORY              1         256.06  GB / 256.06  GB    512   B +  0 B   AEGA0102
```

```bash
* sudo nvme id-ctrl -H /dev/nvme0
NVME Identify Controller:
vid       : 0x1e0f
ssvid     : 0x1e0f
sn        : 71DPCAAMQH71
mn        : KBG40ZNS256G TOSHIBA MEMORY
fr        : AEGA0102
rab       : 3
ieee      : 8ce38e
cmic      : 0
  [3:3] : 0	ANA not supported
  [2:2] : 0	PCI
  [1:1] : 0	Single Controller
  [0:0] : 0	Single Port

mdts      : 9
cntlid    : 0
ver       : 0x10300
[... 224 lines cut ...]
fpi     : 0
  [7:7] : 0	Format Progress Indicator Not Supported

dlfeat  : 1
  [4:4] : 0	Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0	Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0x1	Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 256060514304
nsattr	: 0
nvmsetid: 0
anagrpid: 0
endgid  : 0
nguid   : 01000000000000008ce38e040291ff68
eui64   : 8ce38e040291ff68
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x3 Degraded (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0x1 Better
```

peterharperuk, 2022-01-07:
I was going to suggest it might be a hw problem https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/188
But if it works okay via USB then that probably rules that out. Does it panic before the reboot actually occurs?

DerAndereAndi, 2022-01-07:
The panic occurs during the boot process. See this image:
[image]

DerAndereAndi, 2022-01-07:
Maybe this is a hardware issue. Using the provided USB power plug I get this issue now way more often. Also seeing being stuck in the splash screen and nothing happening afterwards sometimes. Thanks for you help, I will check with the board vendor.

peterharperuk, 2022-01-07:
I wouldn't rule out an issue with the bootloader. If you have a UART cable, bootloader logging might show something interesting. You would have to enable it though. I would buy one of these drives for testing but can't immediately find any for sale. I don't suppose you know anywhere that's selling them?

DerAndereAndi, 2022-01-07:
I sadly don't have such a cable around. But I'll get one.

I got the SSD on Amazon in Germany: https://www.amazon.de/gp/product/B0866BPJPG/

M.2 NVMe in 2230 or 2242 sizes are quite hard to find. You don't happen to know a list with tested NVMe ? :)

peterharperuk, 2022-01-14:
I got one of these drives and can repro this crash although most of the time it seems to just stall boot as the kernel waits for rootfs to appear which it never does. I see warnings in the log about the nvme drive not being ready. I tested a Sabrent Rocket and it seems to reboot fine. I need to check if it's just an issue with the 64bit OS.

I'll have a think about compiling a list of drives. It's a bit tricky as it would imply I've fully tested them in all situations - which probably isn't the case. For what it's worth, my go to reliable drive is a Samsung 970 Evo plus but it's too big for the Waveshare board. I have a Sabrent drive that fits in the Waveshare board, but you'll need to update to the latest stable bootloader to fix an issue we had with that.

peterharperuk, 2022-01-14:
Oh and I was reproducing the problem in a RPI IO board, so it's not Waveshare specific.

DerAndereAndi, 2022-01-14:
@peterharperuk Thanks for getting back. Interesting that these things can be so picky. I can't use the Sabrent as it is double sided and thus doesn't fit. But I got a one sided Samsung 2242, and that seems to work nicely.

Regarding the list of drives: I agree, lots of people could interpret it wrongly, might be better to be user driven somehow.

Anyway, thanks a ton for your work here, for some use cases using only an NVMe is just better than having SD cards. And being able to boot from them, is even better on top of that :)

andrum993, 2022-01-20:
> I sadly don't have such a cable around. But I'll get one.

If your CM4 setup includes the built in Ethernet, you should be able to use the netconsole instead of a UART - see https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#NETCONSOLE.

closed 2022-04-09 by DerAndereAndi