raspberrypi/rpi-eeprom #396: Kingston DC500M SSD unrecognized/can't boot

ayo-x, 2022-01-08:
**Mandatory information**
* `Raspberry Pi model 4b 4GB`
* Board revision `c03112`
* Operating system version
 Raspberry Pi OS Lite, DietPi
* Details of any hardware attached e.g. links to USB 
 Kingston DC500M SSD /w USB 3.0 Adapter
* Photo of the HDMI diagnostics screen, UART trace.
[image]

**Describe the bug**
The Pi doesn't boot from this SSD, throwing USB MSD errors. While booted from another SSD running Dietpi, I connected this SSD with another USB adapter. The Pi sees the adapter (`lsusb`), but not the SSD itself. `lsblk` only shows the SSD it's booted from.

**Expected behaviour**
Boot normally.

**Bootloader version and configuration**
```vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1619712685
capabilities 0x0000001f

vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf14
```

**USB boot (please complete the following information):**  
lsusb -vvv
```Bus 002 Device 002: ID 174c:225c ASMedia Technology Inc. Ugreen Storage Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x174c ASMedia Technology Inc.
  idProduct          0x225c
  bcdDevice            1.00
  iManufacturer           2 Ugreen
  iProduct                3 Ugreen Storage Device
  iSerial                 1 20D11E80B118
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
[... 491 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

ayo-x, 2022-01-08:
I updated the bootloader by hanging the FIRMWARE_RELEASE_STATUS from "default" to "stable". Still not working.

lurch, 2022-01-08:
> I connected this SSD with another USB adapter. The Pi sees the adapter (lsusb), but not the SSD itself. lsblk only shows the SSD it's booted from.

I _suspect_ you'll need to concentrate on getting the SSD recognised by Linux, before you work on booting from the SSD? :shrug:

timg236, 2022-01-08:
Yep. If this doesn’t work from Linux it’s not going to work as a boot device. Could be a power issue with that adapter.
Closing because this seems to be a hardware issue

closed 2022-01-08 by timg236

ayo-x, 2022-01-08:
I installed the SSD into a desktop PC and have successfully installed Debian 11 on it. Everything works as you'd expect. Debian also detects it when I connect it with the USB adapter. I'll see if I can find a powered USB hub to test it with.

timg236, 2022-01-08:
One more thing. The lack of MSD INQUIRY or usb3 connect events suggest that the device is not visible to software. Good quality power hubs sometimes help.
The USB2 device is probably just the internal hub

ayo-x, 2022-01-08:
I'll be getting a powered hub in the next few days. Is there something else I could try in the meantime?

timg236, 2022-01-10:
It's best to ask about hardware interop questions on the Raspberry Pi forums
https://forums.raspberrypi.com/viewforum.php?f=63

ayo-x, 2022-01-12:
It was a power issue. Found a hub that doesn't back power the Pi and the SSD works normally. Thanks guys.

timg236, 2022-01-12:
@ayo-x Thanks for confirming.