raspberrypi/rpi-eeprom #397: CM4 - bootloader does not find NVME after reboot (worked with beta  from march 17 2021)

fanoush, 2022-01-10:
This is a bug for issue discussed in forum https://forums.raspberrypi.com/viewtopic.php?p=1958129#p1958129

NVME ssd boots fine from cold boot but after reboot it shows nvme error 1090584580 and boots from next device. If there is no other bootable device it **does** boot when next round comes - described also [here](https://forums.raspberrypi.com/viewtopic.php?p=1959144#p1959144).

Happens to me with CM4 4GB in official raspberry IO board and also with CM4 8GB in Waveshare CM4-IO-BASE-A. SSD is WD SN520 128GB but it happens to more people with different drives too.

After testing older versions looks like it works for me with beta version of nvme firmware from march 17 2021 but not with later stable versions.

peterharperuk, 2022-01-10:
Thanks for the investigation.

lurch, 2022-01-10:
See also #395 and https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/188 - all related to Waveshare CM4 base-boards.

fanoush, 2022-01-10:
This one is not related to Waveshare base boards, it happens with Raspberry IO board too. Also both work fine with that older beta version.

If bootloader serial output would really help I could enable it and retry.
And btw this is with 32bit raspbian buster with 64bit kernel enabled, not sure if that matters, probably not.

lurch, 2022-01-10:
My apologies - when I read "Happens to me with CM4 4GB in official raspberry IO board and also with CM4 8GB in Waveshare CM4-IO-BASE-A." I somehow missed out the "in official raspberry IO board" part!

peterharperuk, 2022-01-10:
@fanoush Leave it with me - it might help if you could test a fix when I have one available

peterharperuk, 2022-01-11:
@fanoush Any chance you could test a possible fix for me? I have a drive that fails on reboot but it sounds a bit different to the failure you're having.

[pieeprom.original.zip](https://github.com/raspberrypi/rpi-eeprom/files/7848740/pieeprom.original.zip)

cd usbboot/recovery
unzip pieeprom.original.zip
./update-pieeprom.sh
 ../rpiboot.exe -d .

It should mount nvme on reboot and not revert to an sd card. If it still fails the uart logs might be useful to have.

fanoush, 2022-01-11:
It still fails on first try and boots in second round but the nvme error number is different(?)  1073872900 but probably because boot order is different, now I use F561, the error number in first post is when boot order is F516 so nvme was first. will try to get serial output too.

fanoush, 2022-01-11:
oh, there is not much more than on screen
```
RPi: BOOTLOADER release VERSION:6a995251 DATE: 2022/01/11 TIME: 17:19:23 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1641921563 0x6d6adc75 0x00d03140 0x00064f36
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200

Boot mode: SD (01) order f56
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: NVME (06) order f5
nvme error 1073872900
Failed to open device: 'nvme'
Boot mode: BCM-USB-MSD (05) order f
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
xHC ports 1 slots 64 intrs 1
USB2[1] 000206e1 connected
USB2[1] 00200e03 connected enabled
USB2 root HUB port 1 init
DEV [01:00] 2.00 000000:01 class 9 VID 1a40 PID 0101
HUB init [01:00] 2.00 000000:01
HUB [01:00] 2.00 000000:01 init port 1 speed 2
DEV [02:01] 1.16 000001:01 class 0 VID 0458 PID 0186
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
USBSTS 18
USB MSD timed out after 25 seconds
Boot mode: RESTART (0f) order 0
Restart 0 max -1
Boot mode: SD (01) order f56
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: NVME (06) order f5
VID 0x15b7 MN WDC PC SN520 SDAPNUW-128G-1014
NVME on
MBR: 0x00002000,  524289 type: 0x0c
MBR: 0x00084000,249529008 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516191 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516191
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516191 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516191
Read config.txt bytes     2230 hnd 0x00000000
Read start4.elf bytes  2239680 hnd 0x00000000
Read fixup4.dat bytes     5352 hnd 0x00000000
Firmware: 71bd3109023a0c8575585ba87cbb374d2eeb038f Dec  1 2021 15:01:54
0x00d03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

```

peterharperuk, 2022-01-11:
Thanks for testing. It looks like it's failing somewhere different. Let me check the code and get back to you.

peterharperuk, 2022-01-12:
Can I impose on you again to do a test for me? In your case a nvme request is failing. I can't reproduce this and I can't seem to find any WD SN520 drives online. This version has some more logging enabled and it will retry the nvme request again - I'm curious if that fixes the problem for you and which request is failing. Thanks.

[pieeprom.original.zip](https://github.com/raspberrypi/rpi-eeprom/files/7854747/pieeprom.original.zip)

fanoush, 2022-01-12:
no problem, will test when I get back home in few hours

and BTW this happens also with other SSDs, [this guy](https://forums.raspberrypi.com/viewtopic.php?p=1957975#p1957975) mentioned also WD but SN550 - possible very similar to SN520.  [this guy](https://forums.raspberrypi.com/viewtopic.php?p=1940311#p1940311) has same symptoms but did not mention SSD

fanoush, 2022-01-12:
This one works! :-)
This is reboot output, tried twice, first attempt just log of nvme stuff, second attempt with full output (looks same)

```
Boot mode: NVME (06) order f5
MPS 4096-4096
nvme: new request 0 for queue 0
nvme: submit request 0 op 6 for queue 0
nvme: req 63042 completed
NVME req 0 fail 0x4002
nvme error active 2
nvme error active 3
nvme error 1073872900
Failed to open device: 'nvme'
Retry NVME 1
MPS 4096-4096
nvme: new request 0 for queue 0
nvme: submit request 0 op 6 for queue 0
nvme: req 0 completed
NSID 1
nvme: new request 1 for queue 0
nvme: submit request 1 op 6 for queue 0
nvme: req 1 completed
NCAP 0x00ee7c2b0
FMT 0x0
LB 512 B
NCAP 122104 MB
nvme: new request 2 for queue 0
nvme: submit request 2 op 6 for queue 0
nvme: req 2 completed
VID 0x15b7 MN WDC PC SN520 SDAPNUW-128G-1014
SSVID 0x15b7
SN 204046800421
MDTS 524288
SGLS 0x0
nvme: new request 3 for queue 0
nvme: submit request 3 op 9 for queue 0
nvme: req 3 completed
CQ 15 SQ 15
nvme: new request 0 for queue 0
nvme: submit request 0 op 5 for queue 0
nvme: req 0 completed
nvme: new request 1 for queue 0
nvme: submit request 1 op 1 for queue 0
nvme: req 1 completed
NVME on
```
2nd reboot, full output
```
RPi: BOOTLOADER release VERSION:10cade34 DATE: 2022/01/12 TIME: 13:01:09 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1641992469 0x6d6adc75 0x00d03140 0x00064743
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200

Boot mode: SD (01) order f56
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
[... 489 lines cut ...]
nvme: new request 4 for queue 1
nvme: submit request 4 op 2 for queue 1
nvme: req 4 completed
nvme: new request 5 for queue 1
nvme: submit request 5 op 2 for queue 1
nvme: req 5 completed
nvme: new request 6 for queue 1
nvme: submit request 6 op 2 for queue 1
nvme: req 6 completed
nvme: new request 7 for queue 1
nvme: submit request 7 op 2 for queue 1
nvme: req 7 completed
nvme: new request 0 for queue 1
nvme: submit request 0 op 2 for queue 1
nvme: req 0 completed
nvme: new request 1 for queue 1
nvme: submit request 1 op 2 for queue 1
nvme: req 1 completed
Read fixup4.dat bytes     5352 hnd 0x68a38
Firmware: 71bd3109023a0c8575585ba87cbb374d2eeb038f Dec  1 2021 15:01:54
0x00d03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+
```

peterharperuk, 2022-01-13:
Thanks again! I might push this workaround until I get some more drives to test (I've ordered a toshiba and that WD SN550).

nvme: submit request 0 op 6 for queue 0
nvme: req 63042 completed

This is pointing to something funny. A request with command id zero was submitted and it's telling me it completed a request with id 63042. I could believe that this is some left over request from linux but I see the same number in the next boot. Very odd.

fanoush, 2022-01-13:
Yes, noticed that later too. I am not very sure I did not copy same lines twice even if I tried not to. I can recheck today evening.

Maybe there is some reset id/counter command? - was thinking the 63042 command is actually your command 0 and maybe both sides just did not agree on the id? But that would not explain why 2nd retry gets the number right.

fanoush, 2022-01-13:
It is same, I even moved the ssd to the other board with different CM4 and the wrong request number after reboot is still the same - 63042 !
```
Boot mode: NVME (06) order f5
MPS 4096-4096
nvme: new request 0 for queue 0
nvme: submit request 0 op 6 for queue 0
nvme: req 63042 completed
NVME req 0 fail 0x4002
nvme error active 2
nvme error active 3
nvme error 1073872900
Failed to open device: 'nvme'
Retry NVME 1
MPS 4096-4096
nvme: new request 0 for queue 0
nvme: submit request 0 op 6 for queue 0
nvme: req 0 completed
NSID 1
nvme: new request 1 for queue 0
nvme: submit request 1 op 6 for queue 0
nvme: req 1 completed
NCAP 0x00ee7c2b0
FMT 0x0
LB 512 B
NCAP 122104 MB
nvme: new request 2 for queue 0
nvme: submit request 2 op 6 for queue 0
nvme: req 2 completed
VID 0x15b7 MN WDC PC SN520 SDAPNUW-128G-1014
SSVID 0x15b7
SN 204046800421
MDTS 524288
SGLS 0x0
nvme: new request 3 for queue 0
nvme: submit request 3 op 9 for queue 0
nvme: req 3 completed
CQ 15 SQ 15
nvme: new request 0 for queue 0
nvme: submit request 0 op 5 for queue 0
nvme: req 0 completed
nvme: new request 1 for queue 0
nvme: submit request 1 op 1 for queue 0
nvme: req 1 completed
NVME on
```

lurch, 2022-01-13:
@fanoush I wonder if it might be interesting to compare a power-on-from-cold boot log with a reboot boot log? :shrug:

fanoush, 2022-01-13:
It is same except that there is no error so following part with failed boot is not there
```
MPS 4096-4096
nvme: new request 0 for queue 0
nvme: submit request 0 op 6 for queue 0
nvme: req 63042 completed
NVME req 0 fail 0x4002
nvme error active 2
nvme error active 3
nvme error 1073872900
Failed to open device: 'nvme'
Retry NVME 1
```

fanoush, 2022-01-25:
Thanks for heads up. I now even see that there is stable release made from that beta today. I have tested the stable 2022-01-25 one and rebooting with NVME drive now works for me. Thank you :-)

peterharperuk, 2022-02-16:
Closing this as fixed. I have another issue showing NVMe boot loader issues, so intend to look further at this problem

closed 2022-02-16 by peterharperuk