raspberrypi/rpi-eeprom #403: No new firmware showing up when running rpi-eeprom

MopheusDG, 2022-02-03:
Hi, I'm trying to update my RPi4 to the latest stable firmware available via rpi-eeprom, but still shows only the "v2021.04.29-138a1" release. Is that expected or am I missing something ?

Should I be able to update to the "[2022-01-25-vl805-000138a1" already ?

OS: RaspiOS up to date.
HW: Raspberry Pi4 - 4 Gb model

Sorry is this is not the place for this question.
Thanks.

timg236, 2022-02-03:
Run apt update first

MopheusDG, 2022-02-03:
Sorry, my bad, bootloader was outdated, not the firmware. Now I'm updated, sorry for the confusion...

BOOTLOADER: up to date
CURRENT: Tue 25 Jan 14:30:41 UTC 2022 (1643121041)
LATEST: Tue 25 Jan 14:30:41 UTC 2022 (1643121041)
RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
       Use raspi-config to change the release.

VL805_FW: Dedicated VL805 EEPROM
VL805: up to date
CURRENT: 000138a1
    LATEST: 000138a1

closed 2022-02-04 by pelwell

timg236, 2022-02-04:
@MopheusDG Thanks for confirming