raspberrypi/rpi-eeprom #409: WARNING: BOOTFS: "/boot" contains no .elf files. Please check boot directory

gaamaaresosa, 2022-02-14:
### Describe the bug

In my Raspberry Pi-4 (OS: Buster 32bit lite)
I need to update eeprom bootloader firmware from python app.
My box is not rooted, but user pi only.

When I call my below code I get this failure message:
WARNING: BOOTFS: "/boot" contains no .elf files. Please check boot directory

On the rpi-eeprom-update source code I see sudo is missing at mount
       if blkid | grep -qE "/dev/mmcblk0p1.*LABEL_FATBOOT.*RECOVERY.*TYPE.*vfat"; then
          TMP_BOOTFS_MNT="$(mktemp -d)"
          mount /dev/mmcblk0p1 "${TMP_BOOTFS_MNT}"
          BOOTFS="${TMP_BOOTFS_MNT}"
       fi

I don't know why you haven't used sudo on mount line like below.
sudo mount /dev/mmcblk0p1 "${TMP_BOOTFS_MNT}"

What I guess is the /boot is not successfully mounted in /tmp folder because no root permission.

Can you please fix this issue ?

### Steps to reproduce the behaviour

Python3 code:
``
def execute(cmd):
    """
        Purpose  : To execute a command and return exit status
        Argument : cmd - command to execute
        Return   : exit_code
    """
    process = subprocess.Popen(cmd, shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    (result, error) = process.communicate()
    rc = process.wait()
    if rc != 0:
            print(error)
    return result
    
cmd = 'sudo rpi-eeprom-update -i -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2022-02-08.bin'
Res = execute(cmd).decode().strip()
print(Res)
``
I get this reply from Res:
WARNING: BOOTFS: "/boot" contains no .elf files. Please check boot directory

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

I use SD Card (32GB) for OS

vcgencmd bootloader_version
    Sep  3 2020 13:11:43
    version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
    timestamp 1599135103
    update-time 0
    capabilities 0x00000000
    
sudo rpi-eeprom-update:    
    BOOTLOADER: up to date
       CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
        LATEST: Thu  1 Jan 00:00:00 UTC 1970 (0)
       RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
                Use raspi-config to change the release.

      VL805_FW: Using bootloader EEPROM
         VL805: up to date
       CURRENT: 000138a1
        LATEST: 000138a1

sudo -E rpi-eeprom-config
    [all]
    BOOT_UART=0
    WAKE_ON_GPIO=1
    POWER_OFF_ON_HALT=0
    DHCP_TIMEOUT=45000
    DHCP_REQ_TIMEOUT=4000
    TFTP_FILE_TIMEOUT=30000
    ENABLE_SELF_UPDATE=1
    DISABLE_HDMI=0
    BOOT_ORDER=0xf41

    

### System

cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

 vcgencmd version
Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

uname -a
Linux RASPBERRY 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

peterharperuk, 2022-02-14:
What does "lsblk" show on your pi?

gaamaaresosa, 2022-02-14:
> What does "lsblk" show on your pi?

blkid
    /dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="DC3E-E470" TYPE="vfat" PARTUUID="a9db0ec6-01"
    /dev/mmcblk0p2: LABEL="rootfs" UUID="a7adb26a-8b87-4729-99c8-9f5ac069d51e" TYPE="ext4" PARTUUID="a9db0ec6-02"
    /dev/loop0: UUID="de734bf3-7c00-45c1-bd60-c4c61040a2c2" TYPE="crypto_LUKS"
    /dev/zram0: UUID="688cd56a-5dc6-4204-a982-bdd9aa374bfe" TYPE="swap"
    /dev/zram1: UUID="2a8b7c90-9174-4ce4-8d3c-d5d006bdecdf" TYPE="swap"
    /dev/zram2: UUID="5b261698-2267-4452-a45d-8df95dab9121" TYPE="swap"
    /dev/zram3: UUID="98ca3491-247e-410e-99d9-ff323b3bdf3b" TYPE="swap"
    /dev/mapper/L3Zhci9saWIvdmlydHVhbC9tZW0vZ3NyYW0=: UUID="3683f834-ae7c-4a41-9ef6-1ad1e369f79e" TYPE="ext4"
    /dev/loop1: UUID="d54dcab8-2239-4689-aa83-1e815369662e" TYPE="crypto_LUKS"
    /dev/mapper/L3Zhci9saWIvdmlydHVhbC9tZW0vc2Vuc29y: UUID="d1b5da5c-64fb-4706-8a4e-026223042637" TYPE="ext4"
    /dev/mmcblk0: PTUUID="a9db0ec6" PTTYPE="dos"

timg236, 2022-02-14:
"I don't know why you haven't used sudo on mount line like below."

scripts shouldn't randomly call 'sudo', otherwise, you get a horrible UI where the user potentially has to enter their password multiple times or worse, it does a thing with sudo privileges because you happened to recently run a command with sudo.

lurch, 2022-02-14:
Are you using an out-of-the-box setup of [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit) ? Is your `/dev/mmcblk0p1` mounted at `/boot` ? It [looks like](https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L32) you could do something like `sudo BOOTFS=/path/to/your/bootmount rpi-eeprom-update ...` if you have `/dev/mmcblk0p1` mounted to a non-default location?

gaamaaresosa, 2022-02-14:
> Are you using an out-of-the-box setup of [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit) ? Is your `/dev/mmcblk0p1` mounted at `/boot` ? It [looks [...]

I think my disk image seems something wrong.
I don't know how it happened.

please check this image
https://ibb.co/Wx5DqrX

peterharperuk, 2022-02-14:
"What it means ?"

/dev/mmcblk0 is the SD card. It has two partitions /dev/mmcblk0p1 and /dev/mmcblk0p2. The first partition will be small about 256MB. What does lsblk show? It should show that boot is mounted at /boot

NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 59.5G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59.2G  0 part /

If this is the case - what's in your /boot folder?

gaamaaresosa, 2022-02-14:
> "What it means ?"
>
> /dev/mmcblk0 is the SD card. It has two partitions /dev/mmcblk0p1 and /dev/mmcblk0p2. The first partition will be small about 256MB. What does lsblk show? It should show that boot [...]

This is my SD card image:
https://www.linkpicture.com/q/MySDCard.png

gaamaaresosa, 2022-02-14:
> "What it means ?"
>
> /dev/mmcblk0 is the SD card. It has two partitions /dev/mmcblk0p1 and /dev/mmcblk0p2. The first partition will be small about 256MB. What does lsblk show? It should show that boot [...]

blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="DC3E-E470" TYPE="vfat" PARTUUID="a9db0ec6-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="a7adb26a-8b87-4729-99c8-9f5ac069d51e" TYPE="ext4" PARTUUID="a9db0ec6-02"
/dev/mmcblk0: PTUUID="a9db0ec6" PTTYPE="dos"

gaamaaresosa, 2022-02-14:
> "What it means ?"
>
> /dev/mmcblk0 is the SD card. It has two partitions /dev/mmcblk0p1 and /dev/mmcblk0p2. The first partition will be small about 256MB. What does lsblk show? It should show that boot [...]

lsblk
NAME                                   MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
mmcblk0                                179:0    0  14.9G  0 disk  
├─mmcblk0p1                            179:1    0   256M  0 part  /boot
└─mmcblk0p2                            179:2    0  14.6G  0 part  /

gaamaaresosa, 2022-02-14:
> Are you using an out-of-the-box setup of [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit) ? Is your `/dev/mmcblk0p1` mounted at `/boot` ? It [looks [...]

You mean like this ?
cmd = 'sudo BOOTFS=/dev/mmcblk0p1 rpi-eeprom-update -i -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2022-02-08.bin'

Mine looks like below:
lsblk
NAME                                   MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
mmcblk0                                179:0    0  14.9G  0 disk  
├─mmcblk0p1                            179:1    0   256M  0 part  /boot
└─mmcblk0p2                            179:2    0  14.6G  0 part  /

lurch, 2022-02-14:
> This is my SD card image:

80GB sounds suspiciously like the size of a hard-drive, rather than the size of an SD-card?

> You mean like this ?
cmd = 'sudo BOOTFS=/dev/mmcblk0p1 rpi-eeprom-update -i -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2022-02-08.bin'

No, `/dev/mmcblk0p1` is a device-path (to the raw block-level partition), not a filesystem-path (where you can see the files and directories on the partition).

> lsblk
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
mmcblk0 179:0 0 14.9G 0 disk
├─mmcblk0p1 179:1 0 256M 0 part /boot
└─mmcblk0p2 179:2 0 14.6G 0 part /

That looks right. What does `ls -la /boot` show?

gaamaaresosa, 2022-02-14:
> > You mean like this ?
> > cmd = 'sudo BOOTFS=/dev/mmcblk0p1 rpi-eeprom-update -i -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2022-02-08.bin'

I have tried this also:
cmd = 'sudo BOOTFS=/boot rpi-eeprom-update -i -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2022-02-08.bin'

I see same message:
WARNING: BOOTFS: "/boot" contains no .elf files. Please check boot directory

gaamaaresosa, 2022-02-14:
> blkid /dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="DC3E-E470" TYPE="vfat" PARTUUID="a9db0ec6-01" /dev/mmcblk0p2: LABEL="rootfs" UUID="a7adb26a-8b87-4729-99c8-9f5ac069d51e" TYPE="ext4" [...]

I got a clue for PTTYPE="dos" in my blkid
If I use Pendrive (USB) and booted I get the additional /dev/mmcblk0: PTUUID="a9db0ec6" PTTYPE="dos"

Now without Pendrive (USB) after reboot it shows this:
blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="DC3E-E470" TYPE="vfat" PARTUUID="a9db0ec6-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="a7adb26a-8b87-4729-99c8-9f5ac069d51e" TYPE="ext4" PARTUUID="a9db0ec6-02"

gaamaaresosa, 2022-02-14:
Can any one suggest me another method to update bootloader from python code.
It can be any commandline.

Right now I have older version and I want it to update by my software from remote.

vcgencmd bootloader_version
Sep 3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

gaamaaresosa, 2022-02-14:
SOLVED !
Thanks you so much for all helped me suggesting tips.
The issue was /usr/bin/find was zero byte file on the SD card.
Which is mostly used by rpi-eeprom-update script.
After fixing /usr/bin/find now all works charm !

closed 2022-02-14 by gaamaaresosa

lurch, 2022-02-14:
Ah-ha! Well done for tracking down your problem. :+1: 
If `/usr/bin/find` had been truncated to a 0-byte file, then perhaps you've had some sort of filesystem-corruption, and you ought to re-image your SD card with a fresh copy of the OS, in case other files have been corrupted too?

gaamaaresosa, 2022-02-14:
> Ah-ha! Well done for tracking down your problem. 👍 If `/usr/bin/find` had been truncated to a 0-byte file, then perhaps you've had some sort of filesystem-corruption, and you ought to re-image your SD card with a fresh copy of the OS, in case other files have been corrupted too?

You are right !
I could have re imaged before, but I wanted to learn the issue and fix.
Thanks and I will do fresh image now.