raspberrypi/rpi-eeprom #410: The USB_MSD_*_TIMEOUT configs minimum 5s

bryceschober, 2022-02-15:
**Mandatory information**
* Raspberry Pi model: CM4 in custom hardware
* Board revision: d03140
* Operating system version: Custom BuildRoot w/ rpi-linux 5.10.x kernel
* Details of any hardware attached: VL805 with boot config `VL805=1`
* Photo of the HDMI diagnostics screen, UART trace.

**Describe the bug**
We have been configuring out bootloader to support USB MSD boot, and have configured:
```
USB_MSD_DISCOVER_TIMEOUT=500
USB_MSD_LUN_TIMEOUT=500
BOOT_ORDER=0x3154
VL805=1
```
But looking at the effective boot time, it sure seems like these variables have a minimum of 5000 ms from our testing.

**To Reproduce**
1. Time the boot with the above configuration
2. Time the boot with `BOOT_ORDER=0x31`

**Expected behaviour**
Setting USB MSD timeouts to 500 ms should mean that changing config from `BOOT_ORDER=0x3154` to `BOOT_ORDER=0x31` should reduce the boot time by 1 second, not 10 seconds.
 
**Bootloader version and configuration**
[usb-excluded_bootloader-config.txt](https://github.com/raspberrypi/rpi-eeprom/files/8073325/usb-excluded_bootloader-config.txt)
[usb_included-bootloader-config.txt](https://github.com/raspberrypi/rpi-eeprom/files/8073326/usb_included-bootloader-config.txt)
[timestamped_usb_boot_excluded.txt](https://github.com/raspberrypi/rpi-eeprom/files/8073327/timestamped_usb_boot_excluded.txt)
[timestamped_usb_boot_included.txt](https://github.com/raspberrypi/rpi-eeprom/files/8073328/timestamped_usb_boot_included.txt)

**SD card boot (please complete the following information):**  
 - CM4 internal or USB MSD flash stick

**USB boot (please complete the following information):**  
Verify that the the USB device works correctly when hot-plugged under Linux and attach the output of 'lsusb -vvv'.
:+1: :+1:

**Additional Context**
Built from the fork at https://github.com/DynonAvionics/rpi-usbboot, though I don't know why that would make a difference

peterharperuk, 2022-02-15:
Looking at the code, USB_MSD_DISCOVER_TIMEOUT has a minimum of 5 seconds and USB_MSD_LUN_TIMEOUT has a minimum of 100ms

timg236, 2022-02-15:
Iirc the documentation says 5s and I suspect it would take at least 1s even if no devices are present.
Btw if it does find an msd device then it the timeout won’t interrupt msd scsi commands

bryceschober, 2022-02-15:
The documentation for [`USB_MSD_DISCOVER_TIMEOUT`](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_DISCOVER_TIMEOUT) mentions no timeout, where the documentation for `USB_MSD_PWR_OFF_TIME` below does specify a timeout of 250ms. The `USB_MSD_LUN_TIMEOUT` documentation also mentions no timeout.

bryceschober, 2022-02-15:
Would it be possible to reduce the `USB_MSD_DISCOVER_TIMEOUT` minimum timeout to 1 second, if that is the effective limit?

Also, how does `USB_MSD_PWR_OFF_TIME` interact with these timeouts? Is the documented default of `USB_MSD_PWR_OFF_TIME=1000` a limiting factor in the other timeouts?

timg236, 2022-02-15:
No objection to reducing the timeouts for discover to say 1s. It will probably be ok except for weird setups like empty USB sd card readers. 
The PWR off turnout doesn’t do anything on CM4 or newer Pi4 or Pi400 because the USB port power is off from chip reset until sdram has initialised. Therefore there’s no need for any delays the first time usb is used

timg236, 2022-02-17:
The latest beta now sets the minimum timeout to 1s. Please could you give this a try to see if it gets to close one second in your setup.

bryceschober, 2022-02-24:
@timg236 Thanks! We've tried it out, and seen the expected minimum timeout behavior. We ended up going with the stable boot-loader and a GPIO-conditional trigger to enable USB MSD boot based on user input, which results in no extra delay in the normal case when MSD boot isn't desired. But thanks for your responsiveness!

timg236, 2022-02-24:
@bryceschober Thanks for getting back to us. Marking this one as "done" :)

closed 2022-02-24 by timg236