raspberrypi/rpi-eeprom #412: RP4B can't load from SSD due to being underpowered

MOZGIII, 2022-02-20:
### Describe the bug

I get a rainbow rectangle while trying to boot off of SSD in an ASM 2364-powered enclosure.

I've tried an official RPI4 power supply and Apple's 60W Type-C charger - doesn't help, so I'm pretty confident the power supply/cable is not the issue.

I've managed to boot off of an SD card successfully, and then mount and use the SSD without issues (read/write `config.txt`, etc).

I have managed to disable `arm_boost` (which was on by default) and boot off of SSD once, but then I've got some issues with block read timeouts from linux kernel. Next time I was unable to boot at all again.

### Steps to reproduce the behaviour

1. Take RPI4B, an ORICO 20GBPS NVME SSH enclosure (might be any enclosure with ASM 2364, idk, but I used one of these), and a 1.9 TB Samsung V-NAND 970 EVO Plus SSD., and an official power supply or Apple's 60W charger.
2. Write 64bit raspberry os line to the SSD.
3. Assemble SSD enclosure, put in the SSD, connect it to the RPI4.
4. Optionally, connect HDMI/UART for the debug output.
5. Connect the power adapter to the RPI4.
6. That's it, it's supposed to boot, but it's stuck.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

I've upgraded EEPROM to the latest beta while I was troubleshooting (2022-02-16 afaik).

### System

Can't get to that stage. :(

### Bootloader logs

TODO

### USB boot

It does work, and I have many of these (both SSDs and the enclosure).

### NVMe boot

For some reason, the SSD shows up as `/dev/sda`, not as `/dev/nvme*`. Not sure why is that.

lurch, 2022-02-20:
Can you try using a powered USB hub, and see if that solves the problem? https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#typical-power-requirements

MOZGIII, 2022-02-20:
That's what I wanted to try. It would be really undesirable as a final solution, cause adding the USB hub to the unit will push the cost out of the budget I'm aiming at. It's ok for testing. I don't have one laying around, so I'll have to wait while it arrives.

One more piece of data: it seems like the older version of the bootloader works better than the latest beta. Not sure which version I had, but I used the one from the Raspberry PI Imager's misc section just briefly - and it seemed to work another time. I'll experiment with the versions more.

lurch, 2022-02-20:
> push the cost out of the budget I'm aiming at

?? A powered USB hub is much cheaper than a 2TB SSD :wink: 

> One more piece of data: it seems like the older version of the bootloader works better than the latest beta.

If the old bootloader "worked better", then maybe you're sitting just on the edge of the power-threshold? In which case you can't guarantee that you'd get reliable operation even _with_ the old bootloader (e.g. perhaps the SSD would disconnect due to low-power, if the CPU suddenly needs to draw more power for performing an intensive task?).

MOZGIII, 2022-02-20:
> ?? A powered USB hub is much cheaper than a 2TB SSD 😉

I mean yeah, but these are "free" :D It's an odd choice of a disk for RPI too.

> If the old bootloader "worked better", then maybe you're sitting just on the edge of the power-threshold? In which case you can't guarantee that you'd get reliable operation even _with_ the old [...]

Yes, this seems to be what I'm experiencing. However, I bet if I can disable unneeded features like HDMI, WiFi, and BT (it's supposed a wire-connected no-display device) - I might get a few extra amps to get it to be stable enough. This disabling has to be at the boot level though.

What I've found was I can't boot even from an SD card with the SSD plugged in, so it suggests that I need to lower the boot power consumption (and then maybe tweak things at the linux end when I get to that).

timg236, 2022-02-21:
If you are seeing the rainbow screen then the bootloader is working. Since this is not a software issue I'm closing this.

closed 2022-02-21 by timg236

MOZGIII, 2022-02-21:
> not a software issue

I wouldn't be so sure...

pelwell, 2022-02-21:
It's definitely not an EEPROM issue.

You've been asked to try a powered hub and failed to do so, you've mentioned connecting a UART for debug output but not attached any, and you're in the wrong place. You won't get any help here.

MOZGIII, 2022-02-21:
> try a powered hub and failed to do so

I didn't fail to do so, I am in the process. Did anyone specify the timeframe for this feedback? No. So why are you blaming me for not doing my part?

> you've mentioned connecting a UART

That is in the reproducing section, you will be able to reproduce the issue without the UART, and I do not have the UART atm, I was using HDMI. I may be able to capture the logs with UART in the future, and obviously, that is what I was going towards.

> You won't get any help here.

I'm not sure you understand, but I'm seeking to find and solve the issue. If your code was open source, I'd fix it myself. I don't care if people here are in a rush to close the issues for whatever reason, but I'm having an issue and I'm solving it.

Not sure if you'll find anyone to spotlight this issue, and help to debug and push forward, but if you don't need my help - well, ok. I have other options.

timg236, 2022-02-21:
If it's loaded the Linux kernel then the bootloader and GPU firmware have finished and the bootloader isn't even in memory any more. Linux has its own USB stack