raspberrypi/rpi-eeprom #414: Regression: CM4 cannot find root partition on boot after eeprom update

biggreenogre, 2022-02-24:
### Describe the bug

Updated firmware from
  Tue Jan 25 14:30:41 UTC 2022 (1643121041)
to
  Tue Feb  8 17:24:46 UTC 2022 (1644341086)

System hangs "Waiting for root device PARTUUID=<PARTUUID_of_root_partition> ..."

The root partition cannot be found on boot.

[image]

Likely related to change "Avoid unnecessary PCIe probe on CM4". https://github.com/raspberrypi/rpi-eeprom/commit/cecc46f6878ff03ab87c1cdd3c17ca0b446546d2
I guess opinions on what is "unnecessary" differ.
Dr Shipman felt that breathing was "unnecessary" for many of his patients. :-}

Fortunately this issue was discovered on a test system before any production systems were "upgraded".

The bootloader only supports booting through PCIe from NVMe devices. Understanding the difficulty of supporting boot from magnetic devices, a sensible compromise is to boot from a boot partition on SD card (slow, rarely read or written to) with a root partition on magnetic storage (faster, frequently read and written too).
This has worked very well for us since last November until the most recent "stable" firmware update.

After finding the above release note, I tried moving the "USB PCIe MSD" boot method (5) ahead of SD boot, hoping that the PCIe bus would remain active. This does not appear to be the case.

Is there a method for "downgrading" to a working firmware? I quickly tried "rpi-eeprom-update" without success.

### Steps to reproduce the behaviour

CM4 Lite on CM4IO board
Firmware "Tue Jan 25 14:30:41 UTC 2022 (1643121041)"
Marvell 88SE9215 based PCIe SATA III controller (IOCrest SI-PEX40064 4 Port SATA III PCIe 2.0 X 1 Controller Card)
Seagate IronWolf SATA 1TB NAS HD 3.5 Inch SATA 6Gb/s
Raspberry Pi OS lite "buster"

Configure with:
   Boot partition on SD card in CM4IO slot
   Root partition on SATA disk.
/boot/cmdline.txt:
console=serial0,115200 console=tty1 root=PARTUUID=<root_partition_on_HDD> rootfstype=ext4 fsck.repair=yes rootwait

System boots.

Upgrade firmware to "Tue Feb  8 17:24:46 UTC 2022 (1644341086)"
Reboot.

System hangs on boot "Waiting for root device PARTUUID=<PARTUUID_of_root_partition> ..."

### Device (s)

Raspberry Pi CM4 Lite

### Bootloader configuration.

bootloader config:
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try  SD- > USB PCIe MSD -> USB 2.0 BCM XHCI -> Network ->  Loop
#BOOT_ORDER=0xf2541

# Attempt boot from USB-attached SD card first for firmware updates
# Try  USB 2.0 BCM XHCI -> SD -> USB PCIe MSD -> Network ->  Loop
BOOT_ORDER=0xf2415

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

### System

/etc/rpi-issue:
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2

bootloader_version:
2022/02/08 17:24:46
version f1e0407d74b5a2e9e20c1c5738e7b0b566ba0cf8 (release)
timestamp 1644341086
update-time 1645673246
capabilities 0x0000007f

firmware version:
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

kernel:
Linux stuttley 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

lurch, 2022-02-24:
Are you sure this is related to the bootloader update? Your logs above look like they're coming from the Linux kernel, and by the time that Linux is running the bootloader has already finished doing its job and got out of the way :wink: 
You mention that you've got the boot partition on an SD card and the root partition on a USB HDD, so perhaps the PARTUUID that the boot partition is looking for, doesn't match the actual PARTUUID of your root partition?

(Ahhh, I've just re-read your post and saw that you've connected the HDD via a PCIe SATA card rather than via USB, in which case my comment above may be irrelevant, sorry!)

> Is there a method for "downgrading" to a working firmware? I quickly tried "rpi-eeprom-update" without success.

https://www.raspberrypi.com/documentation/computers/compute-module.html#cm4bootloader
"The Compute Module 4 ROM never runs recovery.bin from SD/EMMC and the rpi-eeprom-update service is not enabled by default."
"Replace pieeprom.original.bin if a specific bootloader release is required."

timg236, 2022-02-25:
It seems that the bootloader and GPU have successfully loaded the kernel from SD but the root partition which is presumably on the PCIe SATA is not visible. The logs don't show this but there's probably a line saying link down or nothing showing the endpoint.

The firmware doesn't actually get involved in PCIe and Linux expects to see the RC in the HW reset state. This is actually how it always used to work before USB boot was added.

Since this works with other PCIe devices, it seems that there is a quirk of the card / IO board setup where the PCIe endpoint fails on the initial init, but additional PCIe fundamental resets / enumeration resolves this. That should really be addressed as a general kernel thing e.g. retrying PCIe enumeration.

However, you can replicate the old bootloader behaviour by adjusting the boot-order to probe PCIe first.
e.g. BOOT_ORDER=0xf15 would cause it to check enumerate PCIe to check for a non-existent XHCI controller before trying EMMC.

biggreenogre, 2022-02-25:
Thanks for your input. I figured out how to revert the eeprom to the previous version but the symptoms remain. :-{ I will have a look at the kernel and modules to see if something has changed there.

timg236, 2022-02-25:
I think the upstream kernel driver for the BRCM PCIe no-longer forces a fundamental reset when it's initialises because in theory it's unnecessary. However, it might be that the fundamental reset is helpful in some situations. If so, it might be feasible to add a quirk to force the reset.

biggreenogre, 2022-03-02:
Hi Tim,
I have the system booting from SD with HDD root on the latest "stable" bootloader and latest kernel:
kernel: Linux stuttley 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
BOOTLOADER: up to date
   CURRENT: Tue  8 Feb 17:24:46 UTC 2022 (1644341086)
    LATEST: Tue  8 Feb 17:24:46 UTC 2022 (1644341086)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 
    LATEST: 

Configured system from scratch with firmware "pieeprom-2021-07-06" and an SD card imaged with "2021-10-30-raspios-bullseye-arm64-lite" in the USB port and the system booted with HDD root again.
Upgraded through a range of f/w versions to "pieeprom-2022-02-08" and it continued to work.
Booted with the original SD card in the CM4IO SD slot (without the HDD root), re-configured for HDD root. Not working.

The fix was to update the "initramfs" line in /boot/config.txt to reflect the new kernel version. This was not updated when the new kernel was installed, followed by my own failure to notice and correct it.

Looking through the logs on the HDD I found entries corresponding to the upgrade on the 23rd:
/var/log/apt/history.1.gz:
Start-Date: 2022-02-23  16:59:33
Commandline: apt-get upgrade

/var/log/apt/term.1.gz (corresponding with the above history.1.gz entry):
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-5.10.92-v7+
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.92-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.92-v7l+ /boot/kernel7l.img

The initramfs hook did run (and the initrd.img* files were in the boot partition) but the /boot/config.txt still had "initramfs initrd.img-5.10.63-v7l+", corresponding to the old kernel version. After this was updated to reflect the new kernel ver, "5.10.92-v7l+" the system booted with HDD root.

I don't remember having to update this manually for an earlier raspberrypi-kernel upgrade (on December 13th).
dpkg.log.3.gz:2021-12-13 18:11:44 upgrade raspberrypi-kernel:armhf 1:1.20211029-1 1:1.20211118-3

lurch, 2022-03-03:
ping @XECDesign ?

XECDesign, 2022-03-04:
> ping @XECDesign ?

We don't add an initramfs line to config.txt, so it must've been added manually or by some other script. We don't ship with initramfs enabled and there are no hooks to update everything properly.

biggreenogre, 2022-03-08:
Yes, the initramfs was added manually.
I will look at some other  Debian-based distros and see where they put the hooks.
Would it be more appropriate to contribute this in https://github.com/raspberrypi/linux ?

lurch, 2022-03-08:
I think either https://github.com/RPi-Distro/pi-gen or https://github.com/RPi-Distro/repo would be the right place to ask about initramfs hooks?
But I suspect the answer _might_ be that if you're manually adding an initramfs, then you'll need to support it yourself? :shrug:

XECDesign, 2022-03-09:
It's just not supported by us at all right now.

timg236, 2022-03-09:
The best place to discuss this is the forums because initramfs is not a feature of Raspberry Pi OS.

closed 2022-03-09 by timg236

biggreenogre, 2022-03-11:
Thanks for your time.
I have created a postinst "run-parts" script for the raspberrypi-kernel package that will update /boot/config.txt. I guess we can't contribute directly to the source so I will raise it for discussion in the forums.