raspberrypi/rpi-eeprom #416: VL805 Configuration Issue on CM4-IO/CM4 platform when VL805 is behind a Switch

arqtv, 2022-03-04:
### Describe the bug

The Broadcom 2711 Videocore with Sept 23rd 2021 and later EEPROM firmware will, as part of the bootflow, configure a VL805 USB3 chip that does not have it's own dedicated EEPROM. This works as intended when the VL805 is the only device on the SoC PCIe bus. If the VL805 is behind a PCIe Switch it fails to be configured. The probable reason is that the VL805 address is currently hardcoded at 0x100000 and when the VL805 is at any other address it naturally fails to be configured. In the example attached when behind an ASM1182E  Switch the VL805 is at address 0x300000. 

A recent addition to config.txt is VL805=1, might it be possible to extend this to using the =1 to signify the address, as in =1 represents 0x100000 (the current hardcoded address) so perhaps =3 could equal 0x300000? 

@timg236 @peter

### Steps to reproduce the behaviour

[VL805 Configuration Issue behind ASM1182e Switch v2.docx](https://github.com/raspberrypi/rpi-eeprom/files/8187174/VL805.Configuration.Issue.behind.ASM1182e.Switch.v2.docx)

### Device (s)

Raspberry Pi CM4, Raspberry Pi CM4 Lite

### Bootloader configuration.

[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf25416
ENABLE_SELF_UPDATE=1
VL805=1

### System

[image]

timg236, 2022-03-04:
The address isn't hardcoded but there's no support for PCIe switches in the firmware or the bootloader so this isn't going to work. You'll need to add an SPI EEPROM

timg236, 2024-01-08:
Closed due to inactivity.

closed 2024-01-08 by timg236