raspberrypi/rpi-eeprom #419: CM4: NVMe boot fails with 'failed to open nvme'

mchaker, 2022-03-12:
### Describe the bug

At first, I thought my issue was #378 

However, it does not seem to be the case.
If my issue was the same as #378, I would need an initial boot off of NVMe to work.

In my case, using a CM4 in both the stock IO board and also a [Waveshare CM4 PoE board B](https://www.waveshare.com/wiki/Compute_Module_4_PoE_Board_(B)) yield the same result: 

The bootloader gets stuck with 'failed to open nvme'. No initial boot works.

This repros with both the latest stable and latest beta pieeprom.bin bootloaders.

I am using the [Waveshare PCIe switch](https://www.cnx-software.com/2022/02/07/add-four-pcie-x1-slots-to-raspberry-pi-cm4-io-with-waveshare-pcie-packet-switch-4p-board/) between the IO board and the NVMe drive, which uses an ASM1184e chip.

I have also tried other PCIe x1 switches from Amazon.

_(Booting from the NVMe drive directly plugged into the IO board works, but trying to boot using the PCIe switch in front of the NVMe drive does not work.)_

### Steps to reproduce the behaviour

1. Plug [Waveshare PCIe switch](https://www.cnx-software.com/2022/02/07/add-four-pcie-x1-slots-to-raspberry-pi-cm4-io-with-waveshare-pcie-packet-switch-4p-board/) in to IO board PCIe slot
2. Plug NVMe drive into PCIe switch
3. Attempt boot
4. Get stuck at 'failed to open nvme' message and a boot loop

### Device (s)

Raspberry Pi CM4

### Bootloader configuration.

_Both the latest stable and beta pieeprom.bin files do not work.
Here is `boot.conf`:_

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf5641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```

_And here is `config.txt`:_

```
uart_2ndstage=1
dtoverlay=dwc2,dr_mode=host
```

### System

OS and version: no boot, so no OS load
bootloader version: latest stable and beta (2022-03-10)
firmware version: whatever comes with the latest stable or beta bootloaders (otherwise, latest firmware)
kernel version: no boot, no kernel

### USB boot

N/A

### NVMe boot

NVMe works when directly plugged into the IO board, but not behind a PCIe switch (in this case using an ASM1184e chip)

### Network (TFTP boot)

N/A

lurch, 2022-03-12:
> NVMe works when directly plugged into the IO board, but not behind a PCIe switch (in this case using an ASM1184e chip)

That's expected behaviour. See #416

mchaker, 2022-03-12:
Ah, ok. Was hoping it would be possible to boot with >1 PCIe device connected. 😅
Thanks!

closed 2022-03-12 by mchaker

lurch, 2022-03-13:
You could always do the old "boot partition on SD card (or eMMC), root partition on NVMe SSD (behind a PCIe switch)" trick :slightly_smiling_face:

timg236, 2022-03-14:
Yes, for advanced boot configurations you need to boot Linux (or u-boot) from local storage. The bootloader and GPU firmware don't currently support enumeration of devices behind PCIe bridges

mchaker, 2022-03-14:
Thanks for the suggestion -- I've tried for hours now to install Ubuntu Server 21.10 with /boot on the local eMMC storage, and / (root) on an NVMe drive, to no luck -- the /boot partition is never picked up by the raspi bootloader :(

Are there guides or notes you know of that can help with "/boot on eMMC, root on NVMe or USB" setups? 😅 

Thank you so much for the guidance so far, even though the expected behavior is not to work 😄

timg236, 2022-03-14:
I don't know about a guide but if you have the bootloader loading start4.elf file from EMMC (BOOT_ORDER=1) and are getting the rainbow splash screen then it should be just a matter of editing cmdline.txt on the NVMe device to have something like
rootfs=/dev/nvmn0p1 

Adding uart_2ndstage=1 to config.txt on the EMMC will cause start4.elf to a description of the files loaded to the UART

procount, 2022-03-14:
I concur with Lurch about starting off with Raspberry Pi OS first.
All I know about NVME booting can be found in the following thread -> https://github.com/procount/pinn/issues/517
I don't know if Ubuntu includes the appropriate Kernel drivers for PCIe or NVMe. If not, it would need a new Kernel to include these.
PINN now works on CM4 with NVMe as I have included the drivers. I have versions of Ubuntu that it can install with UBoot removed, so you could try those. Many of my OSes have not been tested with the CM4 as I don't have one, so you may have to select the `showall` option for it to show up as being installable. If you do get one installed, do let know how you got on - if it works I can add the CM4 to the list of models that support Ubuntu,

mchaker, 2022-03-15:
Well, that worked with Raspberry Pi OS 64 Lite - the CM4 just has to boot-loop twice for every time it wants to boot (because of the PINN recovery partition)

(Is there a way to avoid the "boot into PINN, then reboot into RaspiOSLite" loop and just directly boot into RaspiOSLite?)

What I did:
1. Plug CM4 into computer and `./rpiboot`
2. Write the latest PINN image to the CM4 eMMC
3. Unmount/unplug, replug, `./rpiboot`, and make sure that PINN has `otg_mode=1` at the end of `config.txt`
4. Unplug CM4 from computer and boot normally with HDMI and NVMe attached (behind PCie switch)
5. PINN sees the NVMe properly -- configure the PINN settings to enable the `showall` checkbox in the settings window
6. Reboot PINN
7. In PINN, select Raspberry Pi OS Lite 64
8. Select the NVMe SSD as the target
9. Refresh the sources list
10. Install Raspberry Pi OS Lite 64 to the NVMe SSD

I think the `/boot` partition gets made... on the eMMC SSD, and the data partition gets made on the NVMe SSD?

It would be nice to be able to choose btrfs (with the SSD option/flag enabled) instead of ext4 for the root partition but I'm at least thankful it works now!!! 🎉 

Not sure if I will go through the steps of crafting Ubuntu Server 21.10 to be like RaspiOSLite since I already have the drivers and lightweight commandline interface I need with RaspiOSLite. However hopefully whoever in the future DOES want to craft Ubuntu Server to work on the CM4 with an NVMe-behind-PCIe-switch finds this thread 😄 

Thank you all so much for your help!

procount, 2022-03-15:
> (Is there a way to avoid the "boot into PINN, then reboot into RaspiOSLite" loop and just directly boot into RaspiOSLite?)

Yes, add an autoboot.txt file to PINN's partition with the line `boot_partition=6` in it (IIRC). Remove it to restore normal operation.

>I think the /boot partition gets made... on the eMMC, and the data partition gets made on the NVMe SSD?

Yes, the /boot partition must always be on the same partition as PINN.

> It would be nice to be able to choose btrfs (with the SSD option/flag enabled) instead of ext4 for the root partition but I'm at least thankful it works now!!!

I'm sure I supported BTRFS in PINN to allow XBIAN to be installed. So if you crafted your own meta files for RaspiOSLite you could probably do it.

mchaker, 2022-03-15:
Thank you for the tips @procount ! What is a "meta file"? (a link to an example would be fine of course -- I'm guessing there's probably one in the PINN repo?)

mchaker, 2022-03-15:
I don't mind! Got overfocused and sidetracked 😅 thanks for the reminder!