raspberrypi/rpi-eeprom #421: Turning off the watchdog causes the system to not reboot or shutdown

GoingTime, 2022-03-19:
### Describe the bug

After turning off the watchdog with `dtparam=watchdog=off` in the `config.txt file`, the system fails on the second reboot or shutdown. After sending the `sudo reboot` command through SSH, the red power light and green status light are off, but the two LED lights of the network port are always on.

### Steps to reproduce the behaviour

* Write new system in SD card
* Connect to Raspberry Pi via SSH
* Add `dtoverlay=disable-wifi dtoverlay=disable-bt dtparam=watchdog=off` in `/boot/config.txt`
* At first I thought it was caused by the language switch, but according to the experiment, this step seems to have no effect.
```
$ sudo raspi-config
Generating locales (this might take a while)...
  en_US.UTF-8... done
  zh_CN.UTF-8... done
Generation complete.
```
* Reboot
* When rebooting for the second time, the failure occurs
* Cut off the power, reconnect the power, the system starts normally, but still cannot shut down or restart normally
* Remove `dtparam=watchdog=off` in `config.txt`, roboot normally, the problem doesn't reappear

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```shell
$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf41
```

config.txt:
```shell
[cm4]
# Enable host mode on the 2711 built-in XHCI USB control>
# This line should be removed if the legacy DWC2 control>
# (e.g. for USB device mode) or if USB support is not re>
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtoverlay=disable-wifi
dtoverlay=disable-bt
dtparam=watchdog=off
```

### System

```
$ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux bookworm/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"

[... 668 lines cut ...]
cat /etc/rpi-issue
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage2
```

```
$ vcgencmd bootloader_version
2022/01/25 14:30:41
version 6efe41bd9d1e5546fa3715e72e1775b7bd813237 (release)
timestamp 1643121041
update-time 0
capabilities 0x0000007f
```

```
 $ vcgencmd version
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
```

```
$ uname -a
Linux hostname 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux
```

pelwell, 2022-03-19:
The watchdog reset mechanism is used during a reboot, so I can see how disabling one may prevent the other.

Why do you feel the need to disable the watchdog?

timg236, 2022-03-19:
Expanding slightly on what pelwell said the reboot/restart/shutdown all go via the __bcm2835_restart function to reset the chip using the PM watchdog register and therefore disabling the driver will prevent a reboot
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/watchdog/bcm2835_wdt.c#L100

 There's no default "heartbeat"  mechanism where the watchdog is used to enable a reset if the system is frozen so it's unnecessary to set dtparam=watchdog=off

N.B. If the green activity LED doesn't go off briefly during reboot and there is no UART output with BOOT_UART=1 in the EEPROM config then the system probably hasn't rebooted and therefore the bootloader hasn't run.

GoingTime, 2022-03-22:
> The watchdog reset mechanism is used during a reboot, so I can see how disabling one may prevent the other.
> 
> Why do you feel the need to disable the watchdog?

When using OpenWrt as a container, due to the many kmod, device and firewall operations involved, and the lack of sufficient technical ability of the person making the image, it is usually necessary to give the container --privileged permission to maintain the normal operation of OpenWrt, which means that the container will Has root level privileges.
OpenWrt will initialize the watchdog on startup. However, OpenWrt seems to have some flaws and once the watchdog is started it cannot be stopped. If the restart operation of the container is not completed within 30 seconds, the watchdog will force the system to restart, resulting in file system corruption. Similarly, if the shutdown operation is not completed within 30 seconds, the restart operation will also be triggered.
The flaws in OpenWrt are not only in containers, but many people who install directly in routers or hosts also encounter this situation.

GoingTime, 2022-03-22:
> When using OpenWrt as a container, due to the many kmod, device and firewall operations involved, and the lack of sufficient technical ability of the person making the image, it is usually necessary [...]

I tried removing the watchdog with rm -rf, but the container reinitializes when it starts and the watchdog still appears.

```shell
# ubus call system watchdog
{
        "status": "running",       
        "timeout": 30,          
        "frequency": 5         
		}
ubus call system watchdog
# Change timeout to 60s
ubus call system watchdog '{"timeout":60}'
 
# Change frequency to 1s
ubus call system watchdog '{"frequency":1}'

# to stop watchdog:
ubus call system watchdog '{ "stop": true }'
```

Modifications to `status` and `timeout` within OpenWrt are invalid

GoingTime, 2022-03-22:
> Expanding slightly on what pelwell said the reboot/restart/shutdown all go via the __bcm2835_restart function to reset the chip using the PM watchdog register and therefore disabling the driver will [...]

Thank you for your reply. In fact, the PWR and ACT lights are off, and the FDX and LNK lights of the network port are on. WoL is not turned on.

timg236, 2022-03-22:
Closing because this isn't related to the bootloader

closed 2022-03-22 by timg236