raspberrypi/rpi-eeprom #422: Pi4B - Intermittent hang on USB SSD boot

CubicEarth, 2022-03-21:
### Describe the bug

9 out of 10 times, my Pi is booting just fine into Ubuntu 21.10 of an SSD connected to the USB 3.0 port. Sometimes it hangs. The readout is indicating the USB Device is failing to acquire and address, and it is not being enabled. So the SSD never mounts, and the file system is not loaded. Eventually the boot loader gives up trying and the BusyBox shell loads.

Perhaps the initially problem is rooted in power supply issue or a balky USB-SATA chip. When this issue arises, I am curious why the boot loader can't resolve it better in the end. Is there no option to have the boot loader itself initiate a reboot after if it fails to load a file system? Or at least to power cycle the USB ports? In my situation, the odds are pretty good that a reboot will result in the file system loading the system booting as intended.

I can imagine a workaround of having an alternate file system on an SD card, and that file system itself could just trigger a reboot after it loads. But would it be possible, and wouldn't it be desirable, to have this capability builtin to the boot loader itself? For a remote or embedded system, it is unfortunate to have a mode on boot where the Pi becomes permanently unresponsive.

Or maybe it already is there, and I am not making use of it correctly?

### Steps to reproduce the behaviour

Rebooting the system periodically with a ping-based watchdog. Most of the time it boots successfully, but sometimes the USB / SSD MSD fails to load, and the system hangs.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf14
USB_MSD_DISCOVER_TIMEOUT = 10000
MAX_RESTARTS = 1

### System

2022/03/10 11:57:12
version 778c182c82ab673c60a476da6f4e73a03854cacd (release)

(Also happened with earlier bootloader versions)

vcgencmd version
version d9b293558b4cef6aabedcc53c178e7604de90788

#24-Ubuntu SMP PREEMPT Wed Mar 16 07:19:33 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

timg236, 2022-03-22:
The issue doesn't describe the hardware configuration or provide any logs so closing as an interop question. These are best asked on the Raspberry Pi forums.

closed 2022-03-22 by timg236

lurch, 2022-03-22:
> Eventually the boot loader gives up trying and the BusyBox shell loads.

If it has loaded a BusyBox shell, then that implies that the Linux kernel is running, and the bootloader has already finished doing its job? In which case this sounds like a kernel problem rather than a bootloader problem? :shrug:

CubicEarth, 2022-03-22:
Sure enough, I see now that the problem is immediately after the rainbow screen. Boot loader working correctly!

It would be interesting if there was an option for the boot loader to enable the hardware watchdog on the SoC, and reset it immediately before handing over control to the kernel. Then if the kernel failed to manage the watch dog... reset!

lurch, 2022-03-22:
It might not be quite what you're asking for, but does this help? https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#fail-safe-os-updates-tryboot

CubicEarth, 2022-03-26:
Thanks for the pointers lurch. My current line of inquiry is with the init.d configuration settings. I'm hoping to be able to tell Linux it should reboot if it can't find a file system, rather than permanently enter the busy box shell.

CubicEarth, 2023-04-28:
I still haven't found a solution to this.
There is almost no doubt it is a power issue related to the USB SSD. It happens maybe 1 in 15 boots.

"device descriptor read/64, error -71" endlessly prints on the screen

If it would only try to reboot itself, there would be 14/15 chance of a successful startup.

lurch, 2023-05-03:
> There is almost no doubt it is a power issue related to the USB SSD.

Have you tried plugging the SSD into a powered hub, rather than powering it directly from the Raspberry Pi?

CubicEarth, 2023-05-14:
I could do that, but I am interested in finding a solution that doesn't involve extra hardware. But I see there is nothing more the boot loader can do. It is unfortunate the Pi's internal watchdog times out after 15 seconds.  It it could run for a minute or so, it seems like enabling it with a script after the initramfs loads would be the way to go, and then it would reboot if the OS never loaded.

It is just unfortunate there exists a gap in the booting process where only a manual, physical intervention can cause the system to recover.

CubicEarth, 2024-01-31:
Just wanted to follow up - I finally solved this by putting a script in init-premount. Essentially it waits 180 seconds and then checks to see if the root file system is mounted. If it doesn't detect it, it reboots the system. It took plenty of trial and error to get the script working properly.
Indeed the boot loader works fine in this respect. But I am not sure why Ubuntu doesn't enable something like this by default, or have it been an option at least.