raspberrypi/rpi-eeprom #427: rpi-eeprom not being updated in Buster repository

reraikes, 2022-05-07:
### Describe the bug

rpi-eeprom not being updated in Buster repository

### Steps to reproduce the behaviour

Latest version in Bullseye repository is: 13.14-1

root@FreePBX:~# apt search rpi-eeprom
Sorting... Done
Full Text Search... Done
rpi-eeprom/oldstable 13.12-1~buster armhf
  Raspberry Pi 4 boot EEPROM updater

rpi-eeprom-images/oldstable 13.12-1~buster arm64
  transitional package

root@FreePBX:~# apt update
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease
Hit:3 https://deb.nodesource.com/node_10.x buster InRelease
Hit:4 http://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://packages.sury.org/php buster InRelease
Hit:6 http://deb.debian.org/debian buster-updates InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.

root@FreePBX:~# dpkg -s rpi-eeprom
Package: rpi-eeprom
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 28036
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: arm64
Version: 13.12-1~buster
Replaces: rpi-eeprom-images (<< 7.2)
Provides: rpi-eeprom-images
Depends: libraspberrypi-bin, python3, binutils, raspberrypi-bootloader (>= 1.20190819), pciutils
Recommends: flashrom
Breaks: rpi-eeprom-images (<< 7.2)
Conffiles:
 /etc/default/rpi-eeprom-update 0d657c25176302dace9384ca77aa3328
Description: Raspberry Pi 4 boot EEPROM updater
 Checks whether the Raspberry Pi bootloader EEPROM is up-to-date and updates
 the EEPROM.
Homepage: https://github.com/raspberrypi/rpi-eeprom/

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

root@FreePBX:~# vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf14

### System

root@FreePBX:~# cat /etc/rpi-issue
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage2

root@FreePBX:~# vcgencmd bootloader_version
2022/03/10 11:57:12
version 778c182c82ab673c60a476da6f4e73a03854cacd (release)
timestamp 1646913432
update-time 1651963807
capabilities 0x0000007f

root@FreePBX:~# vcgencmd version
Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

root@FreePBX:~# uname -a
Linux FreePBX 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

lurch, 2022-05-07:
Buster is now the legacy release of Raspberry Pi OS, and only receives security updates. See https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

reraikes, 2022-05-08:
@lurch,

I find that response surprising considering:

1. The Raspberry Pi 4B bootloader is not O/S specific and is totally unrelated to Buster.

2. The article you referenced states:

Although we will not support new products on the legacy image, we will make sure any new revisions of older products continue to be supported. So, for example, if we were to release a (currently imaginary) new Raspberry Pi 4 rev 1.5 (which usually means component changes for supply reasons), it would be supported on the legacy image, whereas a new Raspberry Pi product (a future 5 for example, also currently imaginary) would not.

Raspberry Pi OS (Legacy) will remain supported while the various components continue to receive updates. For Debian Buster, support will be available until June 2024. For Linux 5.10 kernel, December 2026.

It's not a problem for me -- I simply downloaded the rpi-eeprom package under Bullseye, moved it to Buster, and installed it with dpkg.  I (apparently incorrectly) assumed there would be more interest in not requiring the typical user to have to jump through such hoops.

lurch, 2022-05-08:
I should probably clarify that I'm not directly involved with rpi-eeprom, and neither am I directly involved with either the Buster or Bullseye versions of Raspberry Pi OS; I'm just trying to help out by explaining things as I see them. So I'll shut up now until somebody more qualified replies :wink: 

But just out of curiosity, why did you feel that you **needed** to upgrade the rpi-eeprom package on Buster?
(to get a clearer view of the differences, you can compare https://github.com/raspberrypi/rpi-eeprom/blob/debian/buster/firmware/release-notes.md and https://github.com/raspberrypi/rpi-eeprom/blob/debian/bullseye/firmware/release-notes.md )

pelwell, 2022-05-08:
The Raspberry Pi Imager makes it easy to download an EEPROM update - all you need is a spare SD card.

timg236, 2022-05-09:
Yes, this is expected.

closed 2022-05-09 by timg236