raspberrypi/rpi-eeprom #430: failed to set config

actraiser30, 2022-05-23:
### Describe the bug

Hello im trying to change the bootloader configuration so it can cut power to my usb hdd when its turned off

using the command: sudo rpi-eeprom-config --edit

[all]
BOOT_UART=1
WAKE_ON_GPIO=1
NET_INSTALL_ENABLED=0
BOOT_ORDER=0xf14

the line: POWER_OFF_ON_HALT=0 is not there i add it manually:POWER_OFF_ON_HALT=1
and also change the:WAKE_ON_GPIO=1 to WAKE_ON_GPIO=0

but upon exit saving and reboot the bootloader remains the same if i check it and also the: POWER_OFF_ON_HALT=1 is not there and power still goes to to usb and gpio upon shutdown

why i t cannot be saved?

thank you

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

.

### System

raspbian buster

### Bootloader logs

.

### USB boot

.

### NVMe boot

.

### Network (TFTP boot)

.

timg236, 2022-05-23:
Please can you provide bootloader uart logs and details of how you are booting the system e.g. sd / usb?

actraiser30, 2022-05-23:
i boot from usb all i did was to edit the bootloader with rpi-eeprom-config --edit i dont have logs i just want to cut the power on usb ports on shutdown

timg236, 2022-05-23:
Unfortunately, without logs we can't see what might be different. If you have a spare SD card then you could try installing Raspberry Pi OS lite, apt update + apt upgrade and configure the bootloader from that since that's known/clean configuration.

adevoss, 2022-06-12:
That does not work for me. The boot order is Network then USB.
The `-f` option does not ignore `FREEZE_VERSION=1`

Any suggestions?

timg236, 2022-06-13:
recovery.bin does not have any code that processes FREEZE_VERSION so using RPi Imager EEPROM update with an SD card will clear FREEZE_VERSION. Just select the boot-mode that you want (SD, USB, Network)

lurch, 2022-06-13:
@adevoss And just to clarify - the check for `recovery.bin` on an SD card happens _before_ the second-stage bootloader (from EEPROM) gets run, and (AFAIK) it's only the second-stage bootloader that reads config values such as `BOOT_ORDER` or `FREEZE_VERSION` https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-4-boot-flow

adevoss, 2022-06-13:
The pi does not boot from sdcard because the boot order is Network - USB.

When I use an usb drive with (fresh install of) PiOS the pi boots via USB.
When I use the same usb drive to boot rpi-boot-eeprom-recovery-2022-04-26-vl805-000138a1 the pi does not boot.
(something with ...mkfs.vfat....label '   NO NAME   '...)

Tried to write the image with the RPI Imager under Windows 10 and PiOS (other pi) and using `dd` under Linux.

pelwell, 2022-06-13:
I think you are mistaken - the ROM will load a recovery.bin or bootcode.bin from SD card. Do as @timg236 says and use Imager to write an SD card to update your EEPROM configuration.

Unless you just enjoy arguing with people trying to help you, in which case do whatever you want.

adevoss, 2022-06-14:
> I think you are mistaken

Yes I was mistaken. Sorry. I thought the sd card was not read because of the boot order.
All is up to date now and in the right boot order.
Thanks for your help.

adevoss, 2022-06-14:
> N.B. There's no good reason to use FREEZE_VERSION anymore

I set it to 1 because I used a particular HAT. When that worked I wanted to stay at that configuration.
Now I don't use that anymore so no need for setting it to 1.

timg236, 2022-06-14:
Thanks for confirming that this is now working for you.

Re: FREEZE_VERSION, my preference is to disable the systemd service rather than tagging the EEPROM. 
systemctl disable rpi-eeprom-update

Although, unsurprisingly since I flash the EEPROM quite often I have a spare SD card with pre-canned configs :)

closed 2022-06-14 by timg236