raspberrypi/rpi-eeprom #431: Ed25519 secure boot public key support

FallingSnow, 2022-05-27:
### Describe the bug

The python code seems to be written to support only RSA. Is it as simple as making this code key type agnostic or is this a deeper issue in the bootloader firmware?

Traceback (most recent call last):                                                                                                                                                                                                           
  File "/root/rpi-bootloader/rpi-eeprom/./rpi-eeprom-config", line 497, in <module>                                                                                                                                                          
    main()                                                                                                                                                                                                                                   
  File "/root/rpi-bootloader/rpi-eeprom/./rpi-eeprom-config", line 484, in main                                                                                                                                                              
    image.update_key(args.pubkey, PUBKEY_BIN)                                                                                                                                                                                                
  File "/root/rpi-bootloader/rpi-eeprom/./rpi-eeprom-config", line 334, in update_key                                                                                                                                                        
    pubkey_bytes = pemtobin(src_pem)                                                                                                                                                                                                         
  File "/root/rpi-bootloader/rpi-eeprom/./rpi-eeprom-config", line 83, in pemtobin                                                                                                                                                           
    key = RSA.importKey(f.read())                                                                                                                                                                                                            
  File "/usr/lib/python3.9/site-packages/Cryptodome/PublicKey/RSA.py", line 770, in import_key                                                                                                                                               
    return _import_keyDER(der, passphrase)                                                                                                                                                                                                   
  File "/usr/lib/python3.9/site-packages/Cryptodome/PublicKey/RSA.py", line 685, in _import_keyDER                                                                                                                                           
    raise ValueError("RSA key format is not supported")                                                                                                                                                                                      
ValueError: RSA key format is not supported

### Steps to reproduce the behaviour

```sh
step crypto keypair secure-boot.pub secure-boot.key --kty OKP # Generate keypair
./rpi-eeprom-config --config ../boot.conf --pubkey secure-boot.pub --out ../pieeprom.bin ./firmware/latest/pieeprom-2022-04-26.bin # Insert public key into firmware
```

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

boot.conf
```conf
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf741
HTTP_HOST=master1.universalis.dev
HTTP_PATH=pi4_net_install
NET_INSTALL_ENABLED=0
```

lurch, 2022-05-28:
> ValueError: RSA key format is not supported

https://pycryptodome.readthedocs.io/en/latest/src/public_key/rsa.html#Crypto.PublicKey.RSA.import_key documents the RSA key  formats that Cryptodome supports.

FallingSnow, 2022-05-28:
Yes, but I'm wondering about OKP and EC keys. Specifically Ed25519. It seems the code is written to only support RSA.
I don't know much about secure boot. Is RSA the only key type that is supported?

peterharperuk, 2022-05-28:
Yes - it's all that is supported in the boot rom.

FallingSnow, 2022-05-28:
Okay, thank you.

closed 2022-05-28 by FallingSnow