raspberrypi/rpi-eeprom #433: HTTP_HOST doesn't accept local IP address

Bourne-ID, 2022-06-29:
### Describe the bug

Per https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#HTTP_HOST - I've attempted to set HTTP_HOST to a private IP address (192.168.50.51) however this seems to be ignored and instead defaults to the global network installed hosted on the official website. 

When changing this to a DNS entry referencing the IP address, HTTP_HOST appears to work as intended, IE downloading from the internal server. 

### Steps to reproduce the behaviour

Set boot.conf to the following:
```
[all]
HTTP_HOST=192.168.50.51
HTTP_PORT=80
HTTP_PATH=images
NET_INSTALL_ENABLED=0
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=0
BOOT_ORDER=0xf71
```
and apply with something similar to the following, rebooting afterwards. 
`sudo rpi-eeprom-config --apply boot.conf pieeprom.upd` 

Which previews the following:

[image]

----
When setting HTTP_HOST to a DNS entry pointing to this IP, the following occurs, showing signs it's downloading locally instead of across the internet:
[image]

(Similarly 
`192.168.50.200 - - [29/Jun/2022:04:36:06 +0000] "GET /images/boot.img HTTP/1.1" 200 134545962 "-" "rpiboot/1.0 (d03114; version:507b2360 timestamp:1656477201)" "-"`
is reported in my NGINX logs)

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
[all]
HTTP_HOST=192.168.50.51
HTTP_PORT=80
HTTP_PATH=images
NET_INSTALL_ENABLED=0
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=0
BOOT_ORDER=0xf71
```

### System

https://pastebin.com/VeDDqP6J

peterharperuk, 2022-06-29:
Yes, I think I see the problem. The code is looking for a valid host name in this field and this fails for an IP address.

peterharperuk, 2023-02-13:
I believe this is fixed in the bootloader January 2023 updated

closed 2023-02-13 by peterharperuk