raspberrypi/rpi-eeprom #450: rpi-eeprom-update is broken again if xxd is not installed

HiassofT, 2022-11-05:
### Describe the bug

Trying to run rpi-eeprom-update on a system without xxd installed no longer works, eg using rpi-eeprom 1197a4a on LibreELEC (which has openssl, but not xxd)
```
LibreELEC:~ # FIRMWARE_RELEASE_STATUS=stable rpi-eeprom-update -a
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Tue Oct 18 10:57:44 UTC 2022 (1666090664)
    LATEST: Fri Nov  4 15:19:15 UTC 2022 (1667575155)
   RELEASE: stable (/usr/lib/kernel-overlays/base/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
   CURRENT: Tue Oct 18 10:57:44 UTC 2022 (1666090664)
    UPDATE: Fri Nov  4 15:19:15 UTC 2022 (1667575155)
    BOOTFS: /flash
xxd not found. Try installing the xxd package.
```
The issue has already been fixed a year ago in commit /86eee571e269e360bafd36128e80117a93838bc7 but has now reappeared as commit eb3bb76a81f0512280769a0f87b926db1cf06129 incorrectly reverted it - I suggest uodating rpiboot repo with the fix and aligning both repos to the fixed version

### Steps to reproduce the behaviour

run latest rpi-eeprom-update on a system without xxd installed

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

[all]
BOOT_UART=1
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
BOOT_ORDER=0xf241
NET_INSTALL_ENABLED=0

### System

LibreELEC, but likely all other systems are affected as well unless xxd is installed

```
LibreELEC:~ # cat /etc/os-release 
NAME="LibreELEC"
VERSION="devel-20221106000142-4052ef0"
ID="libreelec"
VERSION_ID="11.0"
PRETTY_NAME="LibreELEC (HiassofT): devel-20221106000142-4052ef0"
HOME_URL="https://libreelec.tv"
BUG_REPORT_URL="https://github.com/HiassofT/LibreELEC.tv"
BUILD_ID="4052ef019d786939e9cd2239e0149bdf2f3a1fc4"
LIBREELEC_ARCH="RPi4.arm"
LIBREELEC_BUILD="HiassofT"
LIBREELEC_PROJECT="RPi"
LIBREELEC_DEVICE="RPi4"
BUILDER_NAME="HiassofT"
```

MichaIng, 2022-11-06:
The reason it got explicitly removed from dependencies: https://github.com/raspberrypi/rpi-eeprom/issues/383
Need to check whether some recent commit now checks for it/requires it despite no explicit key signing used.

timg236, 2022-11-07:
I'll remove the check from checkDependencies but new flags such as -v will fail without it i.e for the new operations coreutils is required

timg236, 2022-11-07:
https://github.com/raspberrypi/rpi-eeprom/commit/eb3bb76a81f0512280769a0f87b926db1cf06129
https://github.com/raspberrypi/usbboot/commit/e071bbf9c764170d1267120b7418909491ea45eb

peterharperuk, 2022-11-07:
Looks ok

MichaIng, 2022-11-07:
> i.e for the new operations coreutils is required

`coreutils` was required before, but it's "essential", hence forcefully installed by APT. `xxd` however is a dedicated package. Thanks for the fix 👍.

closed 2022-11-07 by HiassofT