raspberrypi/rpi-eeprom #451: CM4: Not booting from SK hynix BC711 NVMe

agners, 2022-11-17:
### Describe the bug

I am trying to boot from a SK hynix BC711 NVMe but it seems the CM4 is not able to boot from that device. The device gets successfully detected in Linux (when booting the system from another media). Also the same setup boots fine from a Samsung 970 EVO plus.

### Steps to reproduce the behaviour

- Use SK hynix BC711 NVMe
- Flash the device using rpiboot/mass storage gadget
- Try to boot from it

### Device (s)

Raspberry Pi CM4 Lite

### Bootloader configuration.

```
$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```

Updated to latest version using rpiboot
```
SIG pieeprom.sig 476be6589f29bef71b1218f232475b3a6a17f6d8feab78a865f7de9a45bdf00d 1668674865
Reading EEPROM: 524288
Writing EEPROM
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++......+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED
```

### NVMe boot

```
pi@raspberrypi:~$ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1        FJB1N694511301Q3Z BC711 NVMe SK hynix 128GB                1         128.04  GB / 128.04  GB    512   B +  0 B   41002131
pi@raspberrypi:~$ sudo nvme id-ctrl -H /dev/nvme0
NVME Identify Controller:
vid       : 0x1c5c
ssvid     : 0x1c5c
sn        :    FJB1N694511301Q3Z
mn        : BC711 NVMe SK hynix 128GB
fr        : 41002131
rab       : 3
ieee      : ace42e
cmic      : 0
  [3:3] : 0     ANA not supported
  [2:2] : 0     PCI
  [1:1] : 0     Single Controller
  [0:0] : 0     Single Port

mdts      : 6
cntlid    : 0x1
ver       : 0x10300
rtd3r     : 0x7a120
rtd3e     : 0x1e8480
oaes      : 0x200
[... 216 lines cut ...]
  [7:7] : 0     Format Progress Indicator Not Supported

dlfeat  : 0
  [4:4] : 0     Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0     Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0     Bytes Read From a Deallocated Logical Block and its Metadata are Not Reported

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
nsattr  : 0
nvmsetid: 0
anagrpid: 0
endgid  : 1
nguid   : 00000000000000000000000000000000
eui64   : ffffffffffffffff
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best
pi@raspberrypi:~$
```

agners, 2022-11-17:
Tested with the latest beta firmware `pieeprom-2022-11-04.bin`, no success.

```
SIG pieeprom.sig 9221b44ba796ba4cd2f60a27990afc66c74d7e2f72b2f7be02e9f8ab13f39481 1668678754
Reading EEPROM: 524288
Writing EEPROM
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*************************....+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED
```

When enabling UART, I get the following:

```
...
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Read config.txt bytes     2075 hnd 0x163
Read start4.elf bytes  2249280 hnd 0x3cdb
Read fixup4.dat bytes     5399 hnd 0x169
0x00b03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16
Starting start4.elf @ 0xfec00200 partition 0
NVME off
+
```

So it seems start4.elf crashing actually?

peterharperuk, 2022-11-17:
>> Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16

Any chance you can try a recent version of start4.elf? Something after "Oct 5 2022" It sounds a bit like https://github.com/RPi-Distro/repo/issues/309

You can get the most recent here https://github.com/raspberrypi/firmware/tree/master/boot

If you add enable_uart=1 and uart_2ndstage=1 to config.txt it'll tell us if it's getting as far as firmware.

agners, 2022-11-17:
After realizing that the EEPROM is actually handing off too start4.elf, that is exactly what I was start doing :smile: 

I manually replaced the files on the NVMe with the ones from the `1.20221104` tag, and it boots! The firmware which is currently shipped with the Raspberry Pi OS (release date 2022-09-22 according to RPi Imager, start4.elf seems to be from `1.20220830` tag).

closed 2022-11-17 by agners

peterharperuk, 2022-11-17:
> but maybe the git log is incomplete

Yes - my fault. The change was to turn nvme off before switching from bootloader to firmware and firmware to kernel. That "NVME off" line is showing the fix working in the bootloader. I failed to mark the change as affecting the firmware so there's no comment in the git log - apologies for that. Thanks for testing.

agners, 2022-11-17:
Ok, I see. I was actually wondering if the "NVME off" log entry could be a problem. Thanks for the insight!