raspberrypi/rpi-eeprom #452: Failed boot producing unconfigured netconsole messages

masto, 2022-11-25:
### Describe the bug

While running a tcpdump of the boot process on a CM4 disconnected from the Internet, I observed the following unexpected traffic, in addition to normal DHCP requests:

```
23:04:00.076748 e4:5f:01:ba:ea:a3 > 55:33:2c:d0:b5:49, ethertype IPv4 (0x0800), length 60: (tos 0x0, ttl 64, id 21952, offset 0, flags [none], proto UDP (17), length 45)
    23.58.181.11.4613 > 128.32.74.11.11410: [udp sum ok] UDP, length 17
	0x0000:  4500 002d 55c0 0000 4011 8e8f 173a b50b  E..-U...@....:..
	0x0010:  8020 4a0b 1205 2c92 0019 0f41 5374 6f70  ..J...,....AStop
	0x0020:  7069 6e67 206e 6574 776f 726b 0a00       ping.network..
23:04:00.076805 e4:5f:01:ba:ea:a3 > 55:33:2c:d0:b5:49, ethertype IPv4 (0x0800), length 127: (tos 0x0, ttl 64, id 21954, offset 0, flags [none], proto UDP (17), length 113)
    23.58.181.11.4613 > 128.32.74.11.11410: [udp sum ok] UDP, length 85
	0x0000:  4500 0071 55c2 0000 4011 8e49 173a b50b  E..qU...@..I.:..
	0x0010:  8020 4a0b 1205 2c92 005d 9539 5258 3a20  ..J...,..].9RX:.
	0x0020:  3234 3020 4950 3a20 3020 4950 5634 3a20  240.IP:.0.IPV4:.
	0x0030:  3234 3020 4d41 433a 2032 3038 2055 4450  240.MAC:.208.UDP
	0x0040:  3a20 3230 3820 5544 5020 5245 4356 3a20  :.208.UDP.RECV:.
	0x0050:  3020 4950 5f43 5355 4d5f 4552 523a 2030  0.IP_CSUM_ERR:.0
	0x0060:  2055 4450 5f43 5355 4d5f 4552 523a 2030  .UDP_CSUM_ERR:.0
	0x0070:  0a
```

This was flashed with the firmware from https://github.com/raspberrypi/usbboot/tree/ec8294ffa46931e5b1b322b04ca26863a98a40cb/recovery using the following boot.conf:

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

#NET_INSTALL_ENABLED=1
```

As you can see, NETCONSOLE is not enabled. This is a Lite module with no storage attached, so it is not getting any configuration from disk, it's just continuously boot looping.

Given the nature of the IPs and destination MAC address (it has the multicast bit set), I suspect this is coming from some sort of uninitialized memory. I found only one Google result for these addresses, comments on a Reddit thread about netbooting that suggest others are seeing this behavior: https://www.reddit.com/r/raspberry_pi/comments/t0pwmd/can_not_netboot_rpi_4b_odd_network_traffic/

### Steps to reproduce the behaviour

As far as I know, flashing the abovementioned firmware/boot.conf to a CM4001000 and letting it boot loop while observing the network traffic it generates for a couple of minutes.

I have tried three of them and they all produce these packets.

### Device (s)

Raspberry Pi CM4 Lite

### Bootloader configuration.

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

#NET_INSTALL_ENABLED=1
```

### System

(OS information should be irrelevant as this is pre-boot, but I have temporarily booted the system to obtain these for completeness)

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage2
```

```
$ vcgencmd bootloader_version
2022/11/04 15:19:15
version febe323f4d5604ba122f914ae1909b033fc1a70b (release)
timestamp 1667575155
update-time 1669338198
capabilities 0x0000007f
```

```
$ vcgencmd version
Aug 26 2022 14:03:16
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)
```

```
$ uname -a
Linux pi-cm4-2 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
```

timg236, 2022-11-25:
I think I can see what's going on here, please could you see if this fix resolves the issue for you?

[rpi-eeprom-recovery-netconsole.zip](https://github.com/raspberrypi/rpi-eeprom/files/10090306/rpi-eeprom-recovery-netconsole.zip)

timg236, 2022-11-25:
@masto Please could you confirm that the fix works for you, if all is good we can push an APT update.

masto, 2022-11-25:
Thank you for the amazingly quick response. I set out to test this and my compute module appears to have fried. It was working, since I used it to double check I could still see the netconsole packets before changing anything. Then all I did was install the RPIBOOT jumper and powered it back up, and it's now completely dead with or without the jumper. (This was not a consequence of installing the firmware, I wasn't even able to get to that point).

I am not sure how much time I can spend on this, but I will let you know if I'm able to test the new firmware. I can use one of the remaining units, but I was planning to set up my cluster this weekend and now I've kind of lost enthusiasm for the project.

timg236, 2022-11-25:
Sorry to hear the CM4 is not working. Is the red LED on and is there power on the 3v3 header pin?

I’ll update APT with this early next week. Thanks for the detailed bug report.

masto, 2022-11-25:
Thanks again for the quick fix. I can confirm now that I'm no longer seeing these packets after flashing `pieeprom-2022-11-25.bin` with my same `boot.conf`.

I don't know what's wrong with this one, but the other three I have all work in the same socket (https://youtu.be/IogJu65c_kY), and this module also fails to do anything when put into a CM4IO (although in that case, the green and red lights just come on and stay on). On the 40-pin header, there's 5V, but on the 3.3V pin I'm getting 1.9V. So I guess that's a clue. Should have thought to check that first. Hopefully Digi-Key will take it back. I've only had this thing 6 days.

closed 2022-11-25 by masto

timg236, 2022-11-26:
Thanks for verifying the fix.

If the 3.3V is only 1.9V then the PMIC on the CM4 or related component has likely failed.  On this board it sounds like it might be able to load some code from and enable the green LED but will likely fail shortly after when peripherals or DDR are initialised. The exact failure mode can be a bit random depending on what's happening on the power supply outputs.