raspberrypi/rpi-eeprom #458: HID error and network install

peterharperuk, 2022-12-15:
### Describe the bug

See https://forums.raspberrypi.com/viewtopic.php?t=339219

It seems some users are seeing "HID error" and that's stopping the device booting normally. It seems that it's possible to get stuck waiting for the ethernet cable to be plugged in? 

Looking at the code. After a HID error we reply to the find a keyboard event. If the bootloader doesn't see that usb has an error the it will think a key has been pressed. It'll then wait 10 mins for the ethernet cable to be plugged in*. It doesn't look like anything is setting the error in the usb host after a HID error. We also reset usb after a hid error - I don't know if that's relevant at all.

* The 10 mins to plug the cable in seems too long, maybe reduce to 2 mins?

### Steps to reproduce the behaviour

Tricky - need a keyboard that reproduces the problem. Might be easier to reproduce with wireless keyboards/mice

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

Latest

lurch, 2022-12-15:
> Tricky - need a keyboard that reproduces the problem. Might be easier to reproduce with wireless keyboards/mice

If you're able to identify what the "bad" USB HID sequence is, perhaps you could program a Pico to emulate that? :man_shrugging:

cleverca22, 2022-12-15:
@lurch if somebody with an offending keyboard, runs `modprobe usbmon` and then fires up wireshark and captures the usb descriptors while plugging it in

then a pico could be programmed to repeat those same descriptors

fgibaux, 2022-12-18:
Hi, I'm getting this error with my remote keyboard (ID 1997:2433 Shenzhen Riitek Technology Co., Ltd wireless mini keyboard with touchpad) I did a capture while plugging it in my laptop: 
[rii_remote_port2.tar.gz](https://github.com/raspberrypi/rpi-eeprom/files/10254208/rii_remote_port2.tar.gz)

HTH

peterharperuk, 2022-12-20:
I have one of these Riitek keyboards now (model RT-MWK01, mini wireless keyboard mouse combo). I saw the problem once, However it's now working fine, which confuses me. I do see the odd HID error, but it doesn't seem to stop boot happening.

lurch, 2022-12-21:
Just a random guess - is it possible that it sends out "bad" HID packets when it gets low on battery? :battery: :shrug:

peterharperuk, 2022-12-21:
Got it again. It's starting net install because the bootloader hasn't been passed the error from usb.

peterharperuk, 2022-12-21:
I have a fix for this

peterharperuk, 2023-01-03:
Test bootloader image attached. Format an UNUSED sd card with this image using RPI Imager and power up a rpi4 and it will update the bootloader. In network install it should show the version 2b8fa59f in the bottom right

[rpi-eeprom-recovery.zip](https://github.com/raspberrypi/rpi-eeprom/files/10338498/rpi-eeprom-recovery.zip)

ascot7654321, 2023-02-12:
Hello.
I have the same Problem. Not every time i reboot, but often!
but my english is not very good. I dont understand if u found the solution in this thread or are you still trying things?
I got a rapoo E2710 Keyboard.
You think this is causing the error?
What can i do? 
thank you

peterharperuk, 2023-02-13:
Do you have the latest version of the bootloader?

ascot7654321, 2023-02-13:
I have no idea. I got 2 SD cards, one with raspberryOS, other one with libreelec. Both got this mistake. Where can i Check the bootloader version and how can i update it?

peterharperuk, 2023-02-13:
On Raspberry Pi OS run: rpi-eeprom-update

ascot7654321, 2023-02-13:
_rpi-eeprom-update:_
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: version unknown. Try sudo rpi-eeprom-update
   CURRENT:
    LATEST:

ascot7654321, 2023-02-13:
How to update?

peterharperuk, 2023-02-13:
sudo rpi-eeprom-update -a

ascot7654321, 2023-02-13:
_sudo rpi-eeprom-update -a_
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
   CURRENT: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)
    UPDATE: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    BOOTFS: /boot
Using recovery.bin for EEPROM update

EEPROM updates pending. Please reboot to apply the update.
To cancel a pending update run "sudo rpi-eeprom-update -r".

_reboot_

thx  a lot. i will check if the error occurrs aggain.
Can i do the same with the libreelec sd card?

peterharperuk, 2023-02-13:
The bootloader is in an eeprom, not on the SD card, so you shouldn't have to do it again.

closed 2023-02-13 by peterharperuk