raspberrypi/rpi-eeprom #459: First reboot after update from 1650968668 to 1670423055 fails

hailfinger, 2022-12-18:
### Describe the bug

I upgraded the EEPROM of my Raspberry Pi 4 Model B Rev 1.4 with 4 GB RAM.
Old version: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
Version after update: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)

Directly after the EEPROM update the Pi booted normally. However, the first reboot after that did hang with a black screen and no network. Pulling GLOBAL_EN low for a short moment reset the Pi successfully and no reboot since has had that problem.

Reporting because across hundreds of reboots of this Pi I haven't seen anything like this. I'm afraid this may happen to some of the 500+ remote and physically inaccessible devices in our fleet if I upgrade.

### Steps to reproduce the behaviour

```
root@pi-test-cdh-ses1:~# rpi-eeprom-update -a
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    LATEST: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    UPDATE: Wed  7 Dec 14:24:15 UTC 2022 (1670423055)
    BOOTFS: /boot
Using recovery.bin for EEPROM update

EEPROM updates pending. Please reboot to apply the update.
To cancel a pending update run "sudo rpi-eeprom-update -r".
root@pi-test-cdh-ses1:~# reboot
```

EEPROM updates will be installed and the system will be booted again. Works. Now reboot again.

`ssh pi-test-cdh-ses1 reboot`

The system shuts down but does not boot again. The HDMI stays blank and the machine doesn't respond to ping.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
root@pi-test-cdh-ses1:~# vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
```

### System

```root@pi-test-cdh-ses1:~# raspinfo 
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage2
[... 685 lines cut ...]
[    5.798168] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[    6.077475] Console: switching to colour frame buffer device 160x45
[    6.103478] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[    6.396853] usb 1-1.3: Found UVC 1.00 device USB3. 0 capture (534d:2109)
[    6.461311] usbcore: registered new interface driver uvcvideo
[    6.750927] usbcore: registered new interface driver snd-usb-audio
[    7.368845] 8021q: 802.1Q VLAN Support v1.8
[    7.545623] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.749150] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    7.794621] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.820187] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    8.076621] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.077861] bcmgenet fd580000.ethernet eth0: Link is Down
[   10.109392] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   10.109436] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.231430] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   16.417414] NET: Registered PF_ALG protocol family
[   17.328293] Initializing XFRM netlink socket
[   22.259744] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[   31.709254] cam-dummy-reg: disabling
```

### Bootloader logs

No bootloader logs because I haven't been able to reproduce it since.

timg236, 2022-12-19:
I tried various downgrade / upgrade / reboot-loop combinations as part of the last release tests (and again now) but it never failed for me.

We don't recommend remotely updating the bootloader if physical access is difficult e.g. the EEPROM is too small for A/B bootloaders and the bootloader + firmware would need a lot of changes for proper HW watchdog support.

However, if you can reproduce this some for of logging to indicate where the failure occurred (bootloader, start.elf, Linux etc) then please add details here.

hailfinger, 2023-02-17:
After 350 updated devices, I only have one possible other case similar to this one, but I haven't been able to get a screenshot from the other failing device yet (it's remote). I'll report back once I know more.

hailfinger, 2023-03-24:
I was unable to get a screenshot from the other failing device, so I'll assume this was just a fluke.
Closing.

closed 2023-03-24 by hailfinger