raspberrypi/rpi-eeprom #466: DISABLE_HDMI=1 prevents BOOT_ORDER to be honoured

goldyfruit, 2023-01-22:
### Describe the bug

When `DISABLE_HDMI=1` is defined in the boot loader configuration, the `BOOT_ORDER` is not followed anymore.

### Steps to reproduce the behaviour

Micro-SD and USB drive with both Raspberry Pi OS Bullseye 64-bit, `PARTUUID` for `rootfs` partition must be different between the Micro-SD and USB drive.

1) Initial boot loader configuration, everything works as expected, the Micro-SD is the first boot.
```
[all]
BOOT_UART=0
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

2) Boot loader configuration with USB drive as first boot, everything works as expected, the USB drive is the first boot.
```
[all]
BOOT_ORDER=0xf14
BOOT_UART=0
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

3) Boot loader configuration with USB drive as first boot and `DISABLE_HDMI=1` set to disable the boot diagnostic screen. Everything works as expected, I removed the USB drive, I don't get the boot diagnostic screen and then the Micro-SD kicks in.
```
[all]
BOOT_ORDER=0xf14
BOOT_UART=0
DISABLE_HDMI=1
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

4) Boot loader configuration with Micro-SD as first boot and `DISABLE_HDMI=1` set to disable the boot diagnostic screen and the USB drive is connected. Nothing works as expected, the USB drive still the first boot even if `0xf41` is configured.
```
[all]
BOOT_ORDER=0xf41
BOOT_UART=0
DISABLE_HDMI=1
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

5) Boot loader configuration with Micro-SD as first boot, `DISABLE_HDMI=1` has been  removed from the boot loader configuration and the USB drive is connected. Everything works as expected, Micro-SD is the first boot.
```
[all]
BOOT_ORDER=0xf41
BOOT_UART=0
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

I tried different combinations, every time `DISABLE_HDMI=1` is defined, the `BOOT_ORDER` is not respected anymore.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
[all]
BOOT_ORDER=0xf14
BOOT_UART=0
DISABLE_HDMI=1
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```

```
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

### System

Raspberry Pi OS Bulleyes 64-bit

jamesregis, 2023-01-22:
Hello,

I tried to boot on the USB storage while the SD card was inserted. and I had the same problem. 

Whenever `DISABLE_HDMI=1` is present in the eeprom configuration, the boot order is not respected (`BOOT_ORDER=0xf14` ). The Pi boots on the SD card instead of the USB memory. 

By deleting `DISABLE_HDMI=1` in the eeprom configuration, the Pi boots on the USB memory.

It seems that `DISABLE_HDMI=1` influences something in the boot order.

pelwell, 2023-01-23:
It seems to be working for me:
```
[all]
BOOT_ORDER=0xf14
BOOT_UART=0
DISABLE_HDMI=1
POWER_OFF_ON_HALT=0
WAKE_ON_GPIO=1
```
```
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```
Using a deliberately incompatible firmware on the USB drive to see the failover:
```
[  106.034883] reboot: Restarting system
Read start4.elf bytes  2240608 hnd 0x5705
Read fixup4.dat bytes     5354 hnd 0x39d
0x00b03114 0x00000000 0x00000fff
Firmware: bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 Jan 20 2022 13:56:48
start4.elf: is not compatible
This board requires newer software
Get the latest software from https://www.raspberrypi.com/software/
ERROR: 44
Retry USB-MSD 1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
USB MSD stopped. Timeout: 25 seconds
Boot mode: SD (01) order f
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [13]
CID: 000353445343313647801abb0d840144
CSD: 400e00325b59000076b27f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,30583808 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Read config.txt bytes     2385 hnd 0x29f2
Read start4.elf bytes  2250848 hnd 0x44
Read fixup4.dat bytes     5396 hnd 0x631d
0x00b03114 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+
```

pelwell, 2023-01-23:
With `BOOT_ORDER=0xf41` it goes straight to the SD card and boots as normal.

goldyfruit, 2023-01-23:
This is really strange, once I got `DISABLE_HDMI=1` in my configuration, any changes related to the boot order are not respected.

I can see the change applied via `rpi-eeprom-config` but if before `DISABLE_HDMI=1` the boot order was `0xf41` and then I want to change to `0xf14` then the boot will still acting as if `0xf41` was configured. But if I removed  `DISABLE_HDMI=1` from the configuration then the boot order now acts as  `0xf14`.

pelwell, 2023-01-23:
I'm not disputing that you are experiencing a problem, but I can't reproduce it, and reading the EEPROM code I can't see how DISABLE_HDMI could affect boot order.

Does changing the order of the config lines make any difference? I used the same order as you without incident, but I'm clutching at straws.

pelwell, 2023-01-23:
And how about `DISABLE_HDMI=0`?

goldyfruit, 2023-01-23:
> I'm not disputing that you are experiencing a problem, but I can't reproduce it

No worry, I totally understand _(I spent few days on this issue and I maybe rough in my description, if you felt that way I apologize :+1:)_

> Does changing the order of the config lines make any difference? I used the same order as you without incident, but I'm clutching at straws.

I tried different order, same behaviour.

> And how about DISABLE_HDMI=0?

Let me try.

lurch, 2023-01-23:
Just a random guess, but could `DISABLE_HDMI=1` mean that the bootloader timings are slightly changed, and that has edge-case effects on how some SD cards and/or USB drives are initialised / detected?

goldyfruit, 2023-01-23:
> And how about `DISABLE_HDMI=0`?

Everything works fine with `DISABLE_HDMI=0`, boot order is respected.

lurch, 2023-01-23:
@goldyfruit It _might_ also be worth testing to see if you have different behaviour between a "reboot/reset with the power still applied" and a "fresh bootup with the power previously totally disconnected" ? :shrug:

timg236, 2023-01-23:
What does the board-revision say? I suspect this is specific to older (pre 1.4) model B with bootloader versions that have network install.

goldyfruit, 2023-01-23:
> @goldyfruit It _might_ also be worth testing to see if you have different behaviour between a "reboot/reset with the power still applied" and a "fresh bootup with the power previously totally disconnected" ? shrug

Thanks @lurch, same behaviour.

> Do you have either any other computer and a USB serial cable or a second Pi and three IDC patch cables (wires to connect one header pin to another)? If so, the best way to debug this would be to set BOOT_UART=1 and watch the output from pins 6 (GND) and 8 (TXD) (for completeness, pin 10 is RXD but no characters will be read during booting).

I do, I'll try to get the boot loader logs.

> What does the board-revision say? I suspect this is specific to older (pre 1.4) model B with bootloader versions that have network install.

Rev 1.2

goldyfruit, 2023-01-23:
> Do you have _either_ any other computer and a USB serial cable _or_ a second Pi and three IDC patch cables (wires to connect one header pin to another)? If so, the best way to debug this would be to [...]

Here is the boot loader log with `DISABLE_HDMI=1`:
```
RPi: BOOTLOADER release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 25a12658 boardrev c03112 stc 544141
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order f1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 9
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
HUB [01:00] 2.16 000000:01 init port 3 speed 3
DEV [02:01] 2.00 000003:01 class 0 VID 0bda PID 8176
HUB [01:00] 2.16 000000:01 init port 4 speed 1
DEV [03:01] 2.00 000004:01 class 0 VID 046d PID c534
HID [03:01] 2.00 000004:01 register HID
USB3[2] 00021203 connected enabled
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
USB MSD stopped. Timeout: 25 seconds
Boot mode: SD (01) order f
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [2]
CID: 001b534d474231515430b0ce5338013b
CSD: 400e00325b590000ee7f7f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02b58002 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,61988831 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' recovery-b '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' recovery-b '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Read config.txt bytes      426 hnd 0x7a10
gpio_cmd: '14=op,dl' pins: 
```

Here is the boot loader log with `DISABLE_HDMI=0`:
```
RPi: BOOTLOADER release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 25a12658 boardrev c03112 stc 369039
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order f1
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
HUB [01:00] 2.16 000000:01 init port 3 speed 3
DEV [02:01] 2.00 000003:01 class 0 VID 0bda PID 8176
HUB [01:00] 2.16 000000:01 init port 4 speed 1
DEV [03:01] 2.00 000004:01 class 0 VID 046d PID c534
HID [03:01] 2.00 000004:01 register HID
USB3[2] 00021203 connected enabled
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 19
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3[2] 000002b1 connected
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3[2] 00281203 connected enabled
USB3 root HUB port 2 init
DEV [02:00] 3.32 000000:02 class 0 VID 0781 PID 5583
MSD device [02:00] 3.32 000000:02 conf 0 iface 0 ep 81#1024 02#1024
MSD [02:00] 3.32 000000:02 register MSD
MSD [02:00] 3.32 000000:02 LUN 0
MSD INQUIRY [02:00] 3.32 000000:02
MSD [02:00] 3.32 000000:02 lun 0 block-count 120176640 block-size 512
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,119644127 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 8 fat-sectors 256 c-count 65467 c-size 8
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 65467
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 8 fat-sectors 256 c-count 65467 c-size 8
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 65467
Read config.txt bytes      391 hnd 0x834
gpio_cmd: '14=op,dl' pins: 

```

timg236, 2023-01-23:
Looks like fallout from network install. Is there an older version that works for you?
Boot order 0xf144 might be a workaround

goldyfruit, 2023-01-23:
> Looks like fallout from network install. Is there an older version that works for you?

Which one do you want me to try?

>  Boot order 0xf144 might be a workaround

I can try.

timg236, 2023-01-23:
Just the default releases. My guess is that before April 22 might work

goldyfruit, 2023-01-23:
>  Boot order 0xf144 might be a workaround

Indeed, the Pi boots on the USB drive but I got the 30 seconds delay _(but as you mentioned, it's  workaround)_. 

Here is the boot loader log with `DISABLE_HDMI=1` and `BOOT_ORDER=0xf144`:
```
RPi: BOOTLOADER release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 25a12658 boardrev c03112 stc 369048
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order f14
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 9
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
HUB [01:00] 2.16 000000:01 init port 3 speed 3
DEV [02:01] 2.00 000003:01 class 0 VID 0bda PID 8176
HUB [01:00] 2.16 000000:01 init port 4 speed 1
DEV [03:01] 2.00 000004:01 class 0 VID 046d PID c534
HID [03:01] 2.00 000004:01 register HID
USB3[2] 00021203 connected enabled
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
USB MSD stopped. Timeout: 25 seconds
Boot mode: USB-MSD (04) order f1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 19
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3[2] 000002b1 connected
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3[2] 00281203 connected enabled
USB3 root HUB port 2 init
DEV [02:00] 3.32 000000:02 class 0 VID 0781 PID 5583
MSD device [02:00] 3.32 000000:02 conf 0 iface 0 ep 81#1024 02#1024
MSD [02:00] 3.32 000000:02 register MSD
MSD [02:00] 3.32 000000:02 LUN 0
MSD INQUIRY [02:00] 3.32 000000:02
MSD [02:00] 3.32 000000:02 lun 0 block-count 120176640 block-size 512
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,119644127 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 8 fat-sectors 256 c-count 65467 c-size 8
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 65467
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 8 fat-sectors 256 c-count 65467 c-size 8
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 65467
Read config.txt bytes      391 hnd 0x834
gpio_cmd: '14=op,dl' pins: 
```

What is the purpose of `0xf144`? Is it kind of a retry on the USB?

goldyfruit, 2023-01-23:
> Just the default releases. My guess is that before April 22 might work

I'll try.

goldyfruit, 2023-01-23:
> Just the default releases. My guess is that before April 22 might work

Same behaviour with a release from March 2022

Here is the boot loader log with `DISABLE_HDMI=1` and `BOOT_ORDER=0xf14`:
```
RPi: BOOTLOADER release VERSION:778c182c DATE: 2022/03/10 TIME: 11:57:12 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1646913432 0x25a12658 0x00c03112 0x000499e4
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order f1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 9
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
HUB [01:00] 2.16 000000:01 init port 3 speed 3
DEV [02:01] 2.00 000003:01 class 0 VID 0bda PID 8176
HUB [01:00] 2.16 000000:01 init port 4 speed 1
DEV [03:01] 2.00 000004:01 class 0 VID 046d PID c534
HID [03:01] 2.00 000004:01 register HID
USB3[2] 00021203 connected enabled
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
USB MSD stopped. Timeout: 25 seconds
Boot mode: SD (01) order f
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
OCR c0ff8000 [2]
CID: 001b534d474231515430b0ce5338013b
CSD: 400e00325b590000ee7f7f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02b58002 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,61988831 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' recovery-b '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' recovery-b '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Read config.txt bytes      426 hnd 0x7a10
gpio_cmd: '14=op,dl' pins: 
```

timg236, 2023-01-25:
I tried it on an earlier Pi4B and did not see this problem so I don't think DISABLE_HDMI=1 is actually having an effect on the BOOT_ORDER.

The failing log includes this line "USB MSD stopped. Timeout: 25 seconds" which indicates that MSD boot was attempted but the device did not respond for reasons unknown. It looks as though there is a connection on the USB3 port but nothing more.

It might be interesting to see what happens if you connect the USB3 device to the USB2 port or try a powered hub / removing other USB devices.

Alternatively, if you just want to avoid seeing a boot-splash with the slow booting USB device you could try setting HDMI_DELAY to a large number (seconds). This will only show the HDMI UI if a fatal error has occurred.

goldyfruit, 2023-01-27:
> I tried it on an earlier Pi4B and did not see this problem so I don't think DISABLE_HDMI=1 is actually having an effect on the BOOT_ORDER.

Yeah, in my head it sound like you, how `DISABLE_HDMI=1` could impact the boot order but in my case for some weird reason it does... Thanks for trying @timg236 :+1: 

Did you try with Raspberry Pi OS on both _(Micro-SD and USB)_ at the same time?

> The failing log includes this line "USB MSD stopped. Timeout: 25 seconds" which indicates that MSD boot was attempted but the device did not respond for reasons unknown. It looks as though there is a connection on the USB3 port but nothing more.

I tried with different USB drive, same behaviour.

> It might be interesting to see what happens if you connect the USB3 device to the USB2 port or try a powered hub / removing other USB devices.

I can try that.

> Alternatively, if you just want to avoid seeing a boot-splash with the slow booting USB device you could try setting HDMI_DELAY to a large number (seconds). This will only show the HDMI UI if a fatal error has occurred.

What will happen if everything is fine _(from a boot order perspective)_, will I miss the boot sequence from the screen until the `HDMI_DELAY` is reached?

timg236, 2023-01-27:
Logically, the display is always on. The bootloader just mutes display updates / HDMI hardware init until HDMI_DELAY seconds after power on. The default is 8 seconds which is enough for most devices to boot but too short for network boot and some USB hardware / NAS setups.

timg236, 2024-01-08:
Closed due to inactivity.

closed 2024-01-08 by timg236