raspberrypi/rpi-eeprom #469: rpi-eeprom-update does not list latest VLI eeprom

hailfinger, 2023-02-07:
### Describe the bug

rpi-eeprom-update doesn't list the latest VL805 firmware update until after it has been installed.

```
root@pi-taucha:~# rpi-eeprom-update 
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
root@pi-taucha:~# rpi-eeprom-update -a
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    UPDATE: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    BOOTFS: /boot
Using recovery.bin for EEPROM update

EEPROM updates pending. Please reboot to apply the update.
To cancel a pending update run "sudo rpi-eeprom-update -r".
root@pi-taucha:~# reboot
Connection to pi-taucha closed by remote host.
Connection to pi-taucha closed.
user@laptop:~$ ssh pi-taucha
Linux pi-taucha 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l
root@pi-taucha:~# rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

I would have expected "LATEST: 000138c0" in the first call to rpi-eeprom-update instead of "LATEST: 000138a1"

Environment: Raspberry Pi OS 32-bit Buster
rpi-eeprom package is version 16.0-1~buster
Raspberry Pi 4 Model B Rev 1.4 with 4 GB RAM

### Steps to reproduce the behaviour

Run rpi-eeprom-update in Buster on a Raspberry Pi 4 Model B Rev 1.4 with 4 GB RAM while the VL805 EEPROM is not up to date.
Not sure if there are other ways to reproduce.

### Device (s)

Raspberry Pi 4 Mod. B

[... 697 lines cut ...]
[    5.338059] usbcore: registered new interface driver snd-usb-audio
[    5.351837] Console: switching to colour frame buffer device 160x45
[    5.376612] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[    5.541816] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    5.541866] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    5.549507] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    5.549556] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    5.559880] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    5.559940] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    7.911446] random: crng init done
[    7.911476] random: 7 urandom warning(s) missed due to ratelimiting
[    7.979284] 8021q: 802.1Q VLAN Support v1.8
[    8.175663] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.322442] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.495971] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.496380] bcmgenet fd580000.ethernet eth0: Link is Down
[    8.544567] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   12.634694] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.634725] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.403998] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   19.080174] NET: Registered protocol family 15
[   19.397117] NET: Registered protocol family 38
[   19.741501] cryptd: max_cpu_qlen set to 1000

root@pi-taucha:~#

hailfinger, 2023-02-07:
Thank you for the explanation. Makes sense, I can work with that.

timg236, 2023-02-13:
I think it's probably possible to scrape the VL805 FW version for lspci -xxx

hailfinger, 2023-02-20:
Indeed. The following snippet reads the currently running VL805 FW version:

lspci -nxxx -d 1106:3483|grep "^50"|sed "s/^..: \(..\) \(..\) \(..\) \(..\).*/\4\3\2\1/"

rbeede, 2024-01-25:
Would this explain why I see LATEST not following the current stable?

Current shows the May 2023, but latest states Jan 2023.

```
root@raspberrypi:/# rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Thu 11 May 06:26:03 UTC 2023 (1683786363)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```