raspberrypi/rpi-eeprom #471: Cannot load UEFI firmware from a WD 6TB hard-disk

cquike, 2023-02-19:
### Describe the bug

I am trying to boot my raspberry pi 4 from a WD My Book 6TB disk. The disk is GPT formatted and has 13 partitions. The first one is a fat16 partition with the boot,esp flags that contains the UEFI firmware (https://github.com/pftf/RPi4). However when I try to boot it seems to bail out and then it tries the sd card.
I attach a screenshot of the error shown on screen:
[image]
If I put a 8 GB pendrive with also GPT formatted with the same fat16 partition as the first one then I get into the UEFI firmware without problems.

### Steps to reproduce the behaviour

- Boot from a pendrive with UEFI firmware and Debian installer.
- Install Debian in the large hard disk
- Create a partition in the large disk also with the UEFI firmware.
- Remove the pendrive.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

Unfortunately I have installed vanilla Debian and those commands are not available there, only in raspbian.

### System

Debian GNU/Linux 11 (bullseye)
Bootloader version: 2022/04/26

### USB boot

# lsusb -vvv

Bus 002 Device 002: ID 1058:25ee Western Digital Technologies, Inc. My Book 25EE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x1058 Western Digital Technologies, Inc.
  idProduct          0x25ee My Book 25EE
  bcdDevice           40.09
  iManufacturer           2 Western Digital
  iProduct                3 My Book 25EE
  iSerial                 1 575831324437304654385434
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15

timg236, 2023-02-20:
This is unrelated to UEFI since it's not able to load start.elf. Best guess is that this hard-drive consumes too much power and is not suitable for use as a boot device. I'd recommend asking on the forums to see if other people use this drive.

cquike, 2023-02-20:
Right, it is not really related to UEFI. I have changed the tittle.
The hard disk has its own AC charger to power it, not charged via USB. Would that still be a problem?

timg236, 2023-02-20:
Does it boot if you leave it cycling through the boot modes? Spinning drives can be slow to start.
There’s a newer version of the bootloader if you want to try that (use rpi imager boot loader recovery to upgrade)

cquike, 2023-02-21:
I have removed also the sd card to force it to try several times on the large USB disk but I still get the same problem. Thank you very much for the suggestion anyway! I might try the newer bootloader version later on.

cquike, 2023-02-22:
I have now updated the bootloader and now it boots! It does not boot at the first try, as you mentioned it needs a second try but then it properly boots from the large USB disk. Now I don't need to have any sd card or extra small USB pendrive.
Thank you very much for the help!

closed 2023-02-22 by cquike