raspberrypi/rpi-eeprom #475: HTTP BOOT http client doesn't follow redirects

dividuum, 2023-04-25:
### Describe the bug

I'm currently investigating how creating a completely remote booting OS might be implemented. While playing around with the HTTP Boot feature (see also [my forum post](https://forums.raspberrypi.com/viewtopic.php?t=350469)), I noticed that the HTTP client within the EEPROM doesn't follow HTTP redirects. Instead the boot process stops with the following error:

```
No free connections
HTTP: web client failed
```

This happens both for 302 redirects for the boot.sig and boot.img requests. I think it would be very useful to support redirects for all boot requests. Let me explain:

Using the HTTP boot feature basically means that you flash the EEPROM once and then have to work with how the HTTP_HOST and HTTP_PATH settings are set. It's time consuming or impossible to make large changes later on. Having the boot.img url fixed in that way means that there is no way to later move the distribution of the boot image file to a different host without reflashing the Pi. Supporting redirects would make that easy: You could still point to a central fixed URL, but it could then, potentially depending on the requesting client, redirect the delivery of the boot.img to different CDNs or potentially customer-local HTTP servers.

Right now the only way to implement this is to proxy through all requests which does not feel good and isn't as flexible. I see basically two options to implement this:

 * Support following 302 redirects in the EEPROM http client. Maybe that's just raising the number of parallel connections from 1 to 2 in the code somewhere?
 * An alternative might to to allow overwriting the boot.img URL within the initial boot.sig response. So the boot.sig response could have an additional field `image` containing the url of the boot.img like this:
   ```
   <sha256>
   rsa2048: <signature>
   image: http://example.net/some-path/boot.img
   ```
   The advantage of that might also be that it would be possible to link both boot requests. Given unique boot.img urls, a response to the boot.sig requests could ensure that it links to the correct boot.img file matching the signature. Right now due to both request being independant, it could happen that boot.img was updated between the boot.sig and boot.img requests happening.

### Steps to reproduce the behaviour

Point HTTP_HOST to an endpoint responding with a 302 response.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
BOOT_ORDER=0xf7
HTTP_HOST=192.168.1.10
HTTP_PATH=netboot
```

### System

pieeprom.bin version is `8ba17717`, running on a Pi4 model `d03115`.

peterharperuk, 2023-04-25:
I think redirects are implemented, but our stack is configured to use only one TCP connection. I suspect that might be the issue, although it wouldn't surprise me if there were other issues.

dividuum, 2023-04-25:
Figured it might be that. If that's not something that can be easily modified: Would the alternative I suggested be something that can be implemented?

peterharperuk, 2023-04-25:
I think it's unlikely we'd want to modify the boot sig file. It's probably easier to just fix redirects.

dividuum, 2023-04-25:
Thanks for the quick responses so far. I guess the boot.sig method would have the advantages mentioned and I figured the existing parser that already splits `<key>: <value>` pairs for `ts`, `rsa2048` might provide the basic mechanism for parsing additional lines already. But obviously I don't have any insight on how any of that is actually implemented. Regardless of how (if?) there is a solution: I'm looking forward to test it as soon as it's available.

timg236, 2023-04-25:
Seems a reasonable feature, but I can't see it happening very soon.

peterharperuk, 2023-04-25:
What does your access log look like? I just tested this on a couple of local devices and it worked...

10.3.14.227 - - [25/Apr/2023:15:58:09 +0100] "GET /pete_test/boot.sig HTTP/1.1" 302 555 "-" "rpiboot/1.0 (b03114; version:8ba17717 timestamp:1682434333)"
10.3.14.227 - - [25/Apr/2023:15:58:10 +0100] "GET /net_install/boot.sig HTTP/1.1" 200 868 "-" "rpiboot/1.0 (b03114; version:8ba17717 timestamp:1682434333)"
10.3.14.227 - - [25/Apr/2023:15:58:11 +0100] "GET /pete_test/boot.img HTTP/1.1" 302 555 "-" "rpiboot/1.0 (b03114; version:8ba17717 timestamp:1682434333)"
10.3.14.227 - - [25/Apr/2023:15:58:12 +0100] "GET /net_install/boot.img HTTP/1.1" 200 44040426 "-" "rpiboot/1.0 (b03114; version:8ba17717 timestamp:1682434333)"

I had my device configured to use /pete_test/ on an apache server running on a rpi4. I setup a redirect from this to /net_install/

dividuum, 2023-04-25:
Oh no. Well that's embarrassing. I must have managed to mess up the redirect somehow so it ended up as localhost when testing from the device. Obviously that doesn't work and not sure why I didn't catch that :-\

Sorry about the noise. I still see the `No free connection` message, but it does indeed follow the redirect regardless and boots correctly.

peterharperuk, 2023-04-25:
> I still see the No free connection message

Where do you see this error? I'm surprised it works if you see that message.

dividuum, 2023-04-26:
I've reworked my infrastructure and can confirm that it works mostly. I still get the "No free connections" message. Looking at the access logs, it seems the `boot.sig` (not redirected) works as expected. Then `boot.img` endpoint is requested twice(!?) and the following messages appear while the second attempt is being made. The main download then seems to complete but then the boot process aborts with `Error 3 loading boot.img` at the very end.

While taking the picture there was also a new message I've not seen before ("Buffer too large 1513 > 1500"). Not sure what that means.

Example output:

[image]

Due to unlimited restarts, the cycle repeats and the download completes during the second attempt but it's still showing "No free connections" while requesting `boot.img`:

[image]

`bootconf.txt` is:

```
BOOT_ORDER=0xf7
HTTP_HOST=testbox.local
HTTP_PATH=netboot/af557d41d87fe300ba0de6477ff9fe0d
NET_INSTALL_ENABLED=0

[config.txt]
[all]
initramfs initramfs.gz followkernel
```

Response to `http://testbox.local/netboot/af557d41d87fe300ba0de6477ff9fe0d/boot.sig`:

```
< HTTP/1.1 200 Boot
< Server: nginx
< Date: Wed, 26 Apr 2023 12:53:30 GMT
< Content-Type: text/plain
< Content-Length: 587
< Connection: close
< 
5775cacc2785620b13db52e70ca9d6075fa5b1169401cd70fc09ed3f03b5bc57
rsa2048: 485c3f26bc8aa3c0875c314[....]
* Closing connection 0
```

Response to `http://testbox.local/netboot/af557d41d87fe300ba0de6477ff9fe0d/boot.img`:

```
< HTTP/1.1 302 Boot away
< Server: nginx
< Date: Wed, 26 Apr 2023 12:53:35 GMT
< Content-Type: text/plain
< Content-Length: 0
< Connection: close
< Location: http://[s3-via-cloudfront-domain]/boot.img
< 
* Closing connection 0
```

with the redirect target responding like:

```
< HTTP/1.1 200 OK
< Content-Type: application/octet-stream
< Content-Length: 46137344
< Connection: keep-alive
< Date: Wed, 26 Apr 2023 12:59:47 GMT
< Last-Modified: Wed, 26 Apr 2023 10:24:18 GMT
< ETag: "2227927c1529e0df48b82568bc38bc16"
< x-amz-server-side-encryption: AES256
< x-amz-meta-s3cmd-attrs: atime:1682504646/ctime:1682504646/gid:1000/gname:ib-sync/md5:2227927c1529e0df48b82568bc38bc16/mode:33204/mtime:1682504640/uid:1000/uname:ib-sync
< Cache-Control: max-age=31536000
< x-amz-version-id: a1QgFQdt2pCaeFm8bvIPvyj5LzQGlrdo
< Accept-Ranges: bytes
< Server: AmazonS3
< X-Cache: Miss from cloudfront
< Via: 1.1 e1dafdc346c45e381eadf0a96d4694c6.cloudfront.net (CloudFront)
< X-Amz-Cf-Pop: TXL50-P4
< X-Amz-Cf-Id: XUFzqOkCtVwC5puffLWUEtQ71oOXtTv-JNQnorHTjE4awwHfTJLo6Q==
< 
[....]
```

Are there any quirks one might look out for? Like expected HTTP response formatting? Timeouts? Headers? Or maximum sizes for urls? Is there any list of the error codes produced (e.g. what's the `3`?). Thanks!

peterharperuk, 2023-04-27:
As far as I can tell it's just looking for content type and content length and ignoring everything else. The "No free connections" may be a bug when it retries. Another bug is it complaining about too large packet size when 1514 should be valid? The "3" error is a generic invalid value error which gets thrown in a number of places. This could be the packet size problem again.

Do you have a public url I can use for testing?

dividuum, 2023-04-27:
I do indeed. Was just finishing this comment: Maybe this is helpful for reproducing this exact test case. This is an API that generates a customized pieeprom image zip on request for the next 7 days:

```
wget --header 'Authorization: Bearer 3733b01a9bebf3d51b7b7191028dbbb1' \
     --post-data='redirect=true' https://info-beamer.com/api/v1/device/create/netboot \
     -O netboot.zip
```

Extracting the generated netboot.zip onto an empty SD card and putting it in a Pi4 will flash the eeprom. Screen should turn green. As usual, remove SD card, then power cycle and it should boot into the custom OS over the internet and show some demo content. During the boot process at least in my testing, I always see the download made twice. Note that the running Pi is  self-registering with my account on my company's service info-beamer where I can fully remotely control it. Depending on your policies, you might want to firewall the device from the rest of your network. Reverting to the stock eeprom and boot process requires (for example) using one of the image files from [the official releases](https://github.com/raspberrypi/rpi-eeprom/releases).

peterharperuk, 2023-04-27:
Hmm, that sounds complicated. I guess I could just not be lazy and test using s3

dividuum, 2023-04-27:
Oh no. I thought I made it as simple as possible :-o

Speaking of content type. I'm currently responding with `Content-Type: text/plain` and correct `Content-Length` to the `boot.sig` request. The `boot.img` also has `Content-Type: text/plain` and correct `Content-Length: 0` with an empty body and the `Location` header pointing to the main image.

dividuum, 2024-07-11:
The duplicate downloads and "Error 3" I've seen in earlier version is now fixed. Not sure since when but with

```
2024/05/17 12:26:58
version 72caf66729df313801bcefe9b1ff7099c71bb5ce (release)
```

everything works as expected.

closed 2024-07-11 by dividuum

peterharperuk, 2024-07-11:
Yes, I think we fixed it when we added network install support to Pi5 as we wanted to be able to download images from github. Thanks for testing and marking it as fixed.