raspberrypi/rpi-eeprom #479: CM4 bootloader does not appear to read or honor EDID

JinShil, 2023-06-08:


timg236, 2023-06-08:
Moved to rpi-eeprom repo

timg236, 2023-06-08:
The bootloader only supports HDMI and DVI. The HDMI compliance tests require VGA support and the bootloader uses the standard timings. If EDID is not available (e.g. short SCL to fake this) then the standard VGA timings will be used

{1,  640,  HDMI_NEGATIVE_SYNC,   16,  96,  48,  480,  HDMI_NEGATIVE_SYNC, 10,  2,  33,    0,    0,       0,       60,  25200000};

timg236, 2023-06-08:
I future improvement might be to have an EEPROM config option to force an explicit CEA mode e.g. if the panel / driver-chip can't support VGA 720 / 1080p could be forced. Although, I think the set of modes will be quite limited

JinShil, 2023-06-09:
We need to display the bootloader diagnostics screen on our LCDs.  Anything that doesn't allow us to do that is not helpful for us.  I don't see how forcing a specific mode will help us, if the mode is not compatible with our LCDs.
  
Why not just read and honor the settings in the EDID like the OS and firmware does?  What's the limitation?  Is the bootloader EEPROM too small to store the code?

timg236, 2023-06-09:
Do the panels not support VGA / DVI?

JinShil, 2023-06-09:
Our LCDs have a 40-pin TTL RGB interface or an LVDS interface.  We run the HDMI output of the CM4 through a converter IC.  The LCD resolutions include 800x480, 1024x600, 1024x768, and 1280x1024.

VGA 640x480 would be better than nothing even if it displayed in the top left corner of the screen or was tiled across the screen, but I don't think we can get anything to appear on the screen if we can't set pixel clocks, front/back porch, etc.  Here's an example of what we see without such settings:

[image]

timg236, 2023-06-09:
Looking at the EDIDs it seems that they don't explicitly declare support for VGA. It may be possible to add support for some common panel resolutions 1024x768 and 800x480 but that's a new feature so can't promise when anyone would be available to look at that.
If we were to do that then I would probably add an option similar to hdmi_mode= to force resolutions if EDID is not found since VGA is not necessarily supported by non-HDMI displays.

JinShil, 2023-09-19:
> If we were to do that then I would probably add an option similar to hdmi_mode= to force resolutions if EDID is not found since VGA is not necessarily supported by non-HDMI displays.

I'm sorry, but I don't understand this.  The EDID has the resolution coded in it, so the firmware should just read it and honor it, correct?

Our LCD panels are connected to the compute module through a TFP401A HDMI to TTL RGB converter.  So, from the perspective of the compute module it is an HDMI display.  I think the firmware just needs to read and transmit the timing settings as pictured below:
[image]
I that possible?

timg236, 2023-09-20:
It's technically feasible, but right now the bootloader HDMI setup is literally just some pre-canned HDMI register setup which was all that was required for 640x480.
Marked as enhancement because it seems reasonable but can't say when this might be done.