raspberrypi/rpi-eeprom #480: Network boot taking nearly 2 minutes if STP is active on managed Ethernet switch

maxnet, 2023-06-18:
### Describe the bug

On managed Ethernet switches that have the Spanning Tree Protocol enabled, every time the network link goes up it tests for network loops for +/- 30 seconds before it lets normal traffic through.

In an ideal world, everyone buying a managed Ethernet switch would know how to properly configure it, and only enable full STP on uplink switch ports that connect to other Ethernet switches, and disable it (or set it to portfast/rapid STP flavour) on Ethernet ports leading to host devices like computers and Pi.
We could provide some education on how to do configuration, but some users may still struggle with the console cable and CLI commands necessary (not all switches offer a convenient option to set this in the management web interface).
And therefore it is realistic to expect that some will end up using a switch that has STP enabled, and suffer from this problem.

In that case my expectancy would be that network boot is delayed by 30 seconds, but not much more.
But in reality is if STP is enabled it currently takes the boot firmware like 1 minutes 45 seconds instead:

[image]

It waits 25 seconds for USB MSD to timeout, goes to network boot, does not succeed because switch port is not forwarding traffic yet, goes to other boot methods again having to wait 25 seconds for USB MSD again, does succeed to network boot the second time.

If I look at the events from the switch:

```
Switch#
*Mar  1 00:20:27.412:     pm_vp 2/14(1): during state not_present, got event 0(add)
*Mar  1 00:20:27.412: @@@ pm_vp 2/14(1): not_present -> present
*Mar  1 00:20:27.412: *** vp_statechange: single: added: 2/14(1)
*Mar  1 00:20:27.412:     pm_vp 2/14(1): during state present, got event 8(linkup)
*Mar  1 00:20:27.412: @@@ pm_vp 2/14(1): present -> authentication
*Mar  1 00:20:27.421:     pm_vp 2/14(1): idle during state authentication
*Mar  1 00:20:27.421: @@@ pm_vp 2/14(1): authentication -> authentication
*Mar  1 00:20:27.421:     pm_vp 2/14(1): during state authentication, got event 18(authen_disable)
*Mar  1 00:20:27.421: @@@ pm_vp 2/14(1): authentication -> notforwarding
*Mar  1 00:20:27.421: STP SW: Fa2/0/12 new blocking req for 1 vlans
*Mar  1 00:20:27.421: *** vp_linkchange: single: up: 2/14(1)
*Mar  1 00:20:29.417: %LINK-3-UPDOWN: Interface FastEthernet2/0/12, changed state to up
*Mar  1 00:20:30.424: %LINEPROTO-5-UPDOWN: Line protocol on Interface FastEthernet2/0/12, changed state to up
*Mar  1 00:20:42.428: STP SW: Fa2/0/12 new learning req for 1 vlans
*Mar  1 00:20:51.991:     pm_vp 2/14(1): during state notforwarding, got event 4(remove)
*Mar  1 00:20:51.991: @@@ pm_vp 2/14(1): notforwarding -> present
*Mar  1 00:20:51.991: *** vp_linkchange: single: down: 2/14(1)
*Mar  1 00:20:51.991: @@@ pm_vp 2/14(1): present -> not_present
*Mar  1 00:20:51.991: *** vp_statechange: single: remove: 2/14(1)
*Mar  1 00:20:52.997: %LINEPROTO-5-UPDOWN: Line protocol on Interface FastEthernet2/0/12, changed state to down
*Mar  1 00:20:54.004: %LINK-3-UPDOWN: Interface FastEthernet2/0/12, changed state to down
*Mar  1 00:20:54.759:     pm_vp 2/14(1): during state not_present, got event 0(add)
*Mar  1 00:20:54.759: @@@ pm_vp 2/14(1): not_present -> present
*Mar  1 00:20:54.759: *** vp_statechange: single: added: 2/14(1)
*Mar  1 00:20:54.759:     pm_vp 2/14(1): during state present, got event 8(linkup)
*Mar  1 00:20:54.767: @@@ pm_vp 2/14(1): present -> authentication
*Mar  1 00:20:54.767:     pm_vp 2/14(1): idle during state authentication
*Mar  1 00:20:54.767: @@@ pm_vp 2/14(1): authentication -> authentication
*Mar  1 00:20:54.767:     pm_vp 2/14(1): during state authentication, got event 18(authen_disable)
*Mar  1 00:20:54.767: @@@ pm_vp 2/14(1): authentication -> notforwarding
*Mar  1 00:20:54.767: STP SW: Fa2/0/12 new blocking req for 1 vlans
*Mar  1 00:20:54.767: *** vp_linkchange: single: up: 2/14(1)
*Mar  1 00:20:56.764: %LINK-3-UPDOWN: Interface FastEthernet2/0/12, changed state to up
*Mar  1 00:20:57.771: %LINEPROTO-5-UPDOWN: Line protocol on Interface FastEthernet2/0/12, changed state to up
*Mar  1 00:21:09.775: STP SW: Fa2/0/12 new learning req for 1 vlans
*Mar  1 00:21:24.782: STP SW: Fa2/0/12 new forwarding req for 1 vlans
*Mar  1 00:21:24.782:     pm_vp 2/14(1): during state notforwarding, got event 14(forward_notnotify)
*Mar  1 00:21:24.782: @@@ pm_vp 2/14(1): notforwarding -> forwarding
*Mar  1 00:21:24.782: *** vp_list_fwdchange: forward: 2/14(1)
```

- It seems that when the boot firmware starts, network link goes up, and it start testing for network loops.
- However 25 seconds later, link goes down and up again. Causing it to start testing for network loops again from the beginning.
- It seems you are only sending DHCP requests for a short time. So network boot times out in the first round, before the switch forwards traffic.

Would it be possible to change the boot firmware so that it only set the network link carrier to up at the very beginning, and no longer resets the network link when it goes from USB MSD to network boot mode?
So that the 25 seconds spent waiting for USB MSD to timeout, would count towards the forwarding delay.

And it would be nice if it would try DHCP a little longer. The PXE standard demands DHCP discover is retried at 4, 8, 16 and 32 seconds for a reason.

### Steps to reproduce the behaviour

- Get a managed Ethernet switch that has STP enabled by default. Most notorious brand for having it on by default is Cisco.
- Use default settings (on most Cisco models you can hold the "mode" button for some seconds typically to reset to factory settings)
- Network boot a Pi

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### Bootloader configuration.

-

pelwell, 2023-06-18:
One could argue that the longer delay is harder to ignore, and therefore more likely to be detected and solved by turning off STP on that port.

maxnet, 2023-06-18:
>One could argue that the longer delay is harder to ignore

There should be a 30 second delay, yes.
That it takes 1 minute 50 second instead, I do consider a firmware bug.

And if that 30 second delay could be done while we are waiting 25 seconds for USB MSD to timeout, the difference between a user that has STP disabled and one that has STP enabled would only be 5 seconds...

maxnet, 2023-06-18:
And if detection is the goal: then I suggest the boot firmware prints a big fat warning message if it detects STP is enabled on the network port to alert the user.
While the switch does not forward traffic, the Pi do can see the packets the switch sends to test for network loops, so can test for that.

(In the Piserver code I am already displaying displaying a warning on the server if it detects STP is active in the network.
https://github.com/raspberrypi/piserver/blob/master/src/stpanalyzer.cpp
However that detection is better done on the client. As it can only test if STP is active on the server's port, while it is configurable on a per-port basis, and it is the client's port that counts.
And the testing also does not work if the server OS is run under virtualization)