raspberrypi/rpi-eeprom #484: An special case not listed in the LED Warning Flash Codes table

xgqfrms, 2023-09-12:
### Describe the bug

After follow the official doc `Using Raspberry Pi Imager to update the bootloader`, I got an special case not listed in the LED Warning Flash Codes table.

The green LED blink `6 times`, and the red LED not blink (`long flashes`).

So, what's wrong with that?

## refs

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#updating-the-bootloader
https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes

### Steps to reproduce the behaviour

## use Raspberry Pi Imager  to format a SD card (SanDisk 64GB, class10)

1. Launch Raspberry Pi Imager
2. Select `Erase` under Operating System
3. Select SD card and then Write

## use Raspberry Pi Imager  to create an Raspberry Pi 4 EEPROM bootloader rescue image

1. Launch Raspberry Pi Imager
2. Select `Misc utility images` under Operating System
3. Select Bootloader
4. Select a boot-mode `SD`
5. Select SD card and then Write

## try recover the EEPROM

1. Boot the Raspberry Pi with the new image and wait for at least 10 seconds.

The green activity LED will blink `6 times` with a steady pattern and the HDMI display `no signal`.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

I can not use `vcgencmd` command.

### System

2023-05-03-raspios-bullseye-arm64.img.xz

pelwell, 2023-09-12:
From the video you posted (thanks!) that looks like 2 (not very) long and 4 short flashes.

1. Does it boot if you remove the update card and put your normal SD card image back?
2. Have you enabled boot signing on your Pi 4?

timg236, 2023-09-12:
This error can also be raised if a pieeprom.sig file is there but corrupted (possibly broken disk).

Delete any pieeprom.*, recovery.* files from the boot partition and run chkdsk or similar to check for FAT corruption

xgqfrms, 2023-09-13:
> From the video you posted (thanks!) that looks like 2 (not very) long and 4 short flashes.
> 
> 1. Does it boot if you remove the update card and put your normal SD card image back?
> 2. Have you enabled boot signing on your Pi 4?

1. I thought my SD card was bad, I bought a new SD card and installed the new image but it still doesn't work.
2. I don't know what "launch signature" is

I think it's most likely that there's something wrong with my board, maybe some bad circuit contact error.

PS: It suddenly stopped working after it fell off my desk yesterday. Everything was fine until then.

xgqfrms, 2023-09-13:
raspberry pi 4b red light always on and  green light always off, if remove the SD card.

xgqfrms, 2023-09-13:
I think it's a poor contact in the CPU circuit.

pelwell, 2023-09-13:
I think it's excessive contact with the ground - gravity-induced blunt-force trauma. The likely cause of the error message is that some of the data lines from SD card are compromised, and the EEPROM-update code is experiencing that as an incorrect signature. You might be able to get it to boot by putting your SD card into a USB card reader and plugging that into the Pi - it depends on what the default boot order is - but it sounds like you will no longer be able to use the SD card slot.

timg236, 2023-09-13:
So 6 lights basically can happen any time a required SHA256 doesn't match 

Hardware cause

- SPI FLASH EEPROM fails
- SD card mangled - reads a corrupted .sig file when doing an EEPROM update
- SDRAM works enough to get past DDR init and basic tests but causes some bit differences computing hashes. The built in quick memory test normally catches this. 

Checking 3V3 is a good indicator of PMIC damange

xgqfrms, 2023-09-13:
> This error can also be raised if a pieeprom.sig file is there but corrupted (possibly broken disk).
> 
> Delete any pieeprom._, recovery._ files from the boot partition and run chkdsk or similar to check for FAT corruption

I have tried that, still does not work.

xgqfrms, 2023-09-13:
> I think it's excessive contact with the ground - gravity-induced blunt-force trauma. The likely cause of the error message is that some of the data lines from SD card are compromised, and the [...]

You are right, I will try to use the USB boot-mode the Pi, Thank s a lot.

xgqfrms, 2023-09-13:
> So 6 lights basically can happen any time a required SHA256 doesn't match
>
> Hardware cause
>
> * SPI FLASH EEPROM fails
> * SD card mangled - reads a corrupted .sig file when doing an EEPROM update
> * [...]

1. I'm using the Official Power Supply
2. as @pelwell said, maybe my SD card slot had broken.

closed 2023-09-13 by pelwell