raspberrypi/rpi-eeprom #486: repo reorganized - rpi-eeprom/releases.md paths outdated, where have stable etc gone?

draeath, 2023-10-16:
### Describe the bug

It appears that the repository recently was reorganized, but some of the documenmtation still points to old paths (such as `https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/stable`).

Please update the documentation and/or fix the paths.

Note: I do not use any utilities to manage my Pi firmware, nor do I use Raspbian. I prefer to do this manually.

### Steps to reproduce the behaviour

N/A

### Device (s)

Other

### Bootloader configuration.

N/A, this is a project-level issue.

draeath, 2023-10-16:
Links/paths were busted in c2918a8ec67dcb5c2ea7495cd52312a549ab152e

draeath, 2023-10-16:
I've found where these files now live, but I'm still grumpy that this was reorganized instead of a new branch or new repository for the Pi 5.

lurch, 2023-10-16:
This change to the directory-structure was discussed in #485

> I think I found that `vl805-000138c0.bin` is the most recent.

It's a bit hidden away, but https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L642 says:
```
   # The VL805 version number is an eight character hex string. Select the
   # largest number for the newest version.
```

> some of the documenmtation still points to old paths

If you'd like to point out where that incorrect documentation is, we'll do our best to get it updated.

draeath, 2023-10-16:
Any chance we can get some basics on the two (and presumably more in the future, if more generations are developed later) firmware directories added to the main repo readme? It took some sleuthing to work out what it was, and I didn't find that PR in the process.

I think that would help anyone else who runs into this the hard way update their scripts etc?

I'll keep an eye out for stale docs on your end. In this case it was one of my own private notes.

timg236, 2023-10-17:
"Note: I do not use any utilities to manage my Pi firmware, nor do I use Raspbian. I prefer to do this manually."

The supported interface rpi-eeprom-update / rpi-eeprom-config - third-party tools should wrap these. If there's specific documentation that refers to the old directory structure then we'll fix that.

closed 2023-10-17 by timg236