raspberrypi/rpi-eeprom #493: Pi 4 conditional configs seem parsed before halt and get confusing when booting later

jheeks, 2023-10-27:
### Describe the bug

It seems that bootloader conditional configurations are evaluated and stored when entering halt mode.
Later, when starting from halt mode it seems that the conditionals are not re-evaluated but the stored status is used.

The expectation is that the conditionals are evaluated when starting from halt mode.

### Steps to reproduce the behaviour

I have described this observation in [forum post](https://forums.raspberrypi.com/viewtopic.php?p=2149953#p2149554)
The key points:
* The configuration allows halt mode.
* Based on state of a GPIO some boot options are changed.
* It works as expected in power on reset case and in reboot case but not in halt mode case.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
[gpio3=1]
BOOT_ORDER=0xf14
[all]

### System

[raspinfo.txt](https://github.com/raspberrypi/rpi-eeprom/files/13193128/raspinfo.txt)

jheeks, 2023-10-29:
I have captured a UART log from bootloader
[bootloaderLog.txt](https://github.com/raspberrypi/rpi-eeprom/files/13198848/bootloaderLog.txt)
With following steps:
* power off to halt mode with button not pressed, means GPIO3=1
* start with button always pressed, means GPIO3=0
  -> boots from USB, which fits to the condition GPIO3=1
* power off to halt mode with button pressed, means GPIO3=0
* start with very short button press, means GPIO3=1 afterwards
  -> boots from SD, which fits to the condition GPIO3=0

Unfortunately, i cannot see any indication of the conditional's evaluation in the UART log.
But the log shows that first boot is using USB and second boot is using SD.

timg236, 2023-10-30:
The EEPROM config is not re-evaulated after coming out of vpu-HALT (WAKE_ON_GPIO) so this is the expected behaviour.

jheeks, 2023-10-30:
Ok, I understand that this behaviour is expected and will not be changed.
Is it then reasonable to make this clear in the documentation with a small note at [WAKE_ON_GPIO](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#WAKE_ON_GPIO) ?
Or is it already documented elsewhere?

timg236, 2023-10-30:
> Ok, I understand that this behaviour is expected and will not be changed. Is it then reasonable to make this clear in the documentation with a small note at [WAKE_ON_GPIO](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#WAKE_ON_GPIO) ? Or is it already documented elsewhere?

It's not documented, but I would agree that it's non-obvious so tagging this bug as documentation as a reminder to update the docs