raspberrypi/rpi-eeprom #495: Strict dependency on "raspi-firmware" is causing problems on distro-upgraded systems

MichaIng, 2023-11-15:
### Describe the bug

The latest version of the `rpi-eeprom` package strictly depends on the `raspi-firmware` package, i.e. is not satisfied with the old `raspberrypi-bootloader` package anymore. The `raspi-firmware` package however strictly requires the FAT partition to be mounted at `/boot/firmware`, which is not the case for e.g. originally Bullseye images upgraded to Bookworm. Related reports can be found:
- https://forums.raspberrypi.com/viewtopic.php?t=359533
- https://forums.raspberrypi.com/viewtopic.php?t=359622
- https://bugs.debian.org/1055901

I had a look into the package, and I see no reason why it would not work with the old mount style, actually I see no reason why it would depend on the bootloader package at all, since it contains all EEPROM bootloader/firmware binaries self-contained, and does not R/W anything to the FAT partition, or do I miss something?

As you can see in the forum and Debian bug tracker, people are now "solving" this in various different ways, which are all, AFAICS, incomplete and will lead to follow-up errors, especially since the old kernel package does not copy files to the new mount, and the config symlinks are usually not created, so that expected and actual boot configs/cmdline and kernel images can diverge.

I am working on a script for an as safe as possible and complete migration to the new kernel and firmware packages, but it would be great if this was not enforced by an `apt upgrade`, if possible.

### Steps to reproduce the behaviour

From a Bullseye => Bookworm upgraded image, or one which was generated with FAT mounted to `/boot` and old kernel/bootlaoder packages:
```sh
apt update
apt upgrade
# pr
apt install rpi-eeprom
```

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### Bootloader configuration.

irrelevant

### System

irrelevant

XECDesign, 2023-11-15:
@timg236 Do you remember why the raspberrypi-bootloader dependency was added in the first place? I can't think of a reason for it.

MichaIng, 2023-11-15:
I know distro upgrades are not officially supported, but many do it regardless, and it usually works pretty well, at least on headless/server systems. If the dependency of the package on `raspi-firmware` indeed turns out do have no specific reason, other than saying "you need a compatible kernel+dtb+boot config on the FAT partition for this EEPROM bootloader to work" (which is obvious), then it would help/prevent users running into this issue without any downsides. So I am not asking for supporting it (I know especially with the desktop/GUI packages, it is more complicated), but just making the life of those easier who do it, knowing that they are on their own of something breaks, and in this particular case: Preventing them from actually breaking something by applying incomplete workarounds/migrations.

timg236, 2023-11-15:
> @timg236 Do you remember why the raspberrypi-bootloader dependency was added in the first place? I can't think of a reason for it.

Not off hand. It may have been for /boot mount points in the past but that's probably no longer true

XECDesign, 2023-11-16:
Updated package is in the internal repo now and should go public next week.

timg236, 2023-12-14:
I think this is resolved now.

closed 2023-12-14 by timg236