raspberrypi/rpi-eeprom #496: How should the 2712 EEPROM firmware be packaged?

classabbyamp, 2023-11-16:
### Describe the bug

I am working on updating the packaging of rpi-eeprom for Void Linux.

Does the 2712 firmware also go in /lib/firmware/raspberrypi/bootloader mixed with the 2711 firmware?

### Steps to reproduce the behaviour

.

### Device (s)

Other

### Bootloader configuration.

,

timg236, 2023-11-16:
I recommend you install RPi OS and look there.
It is now /lib/firmware/raspberrypi/bootloader-2711 and /lib/firmware/raspberrypi/bootloader-2712

closed 2023-11-16 by timg236

classabbyamp, 2023-11-16:
I was looking at the contents of the RPi OS packages but I didn't see it there. thanks

timg236, 2023-11-16:
> I was looking at the contents of the RPi OS packages but I didn't see it there. thanks

If you are familiar with building .debs (I'm not :)) then the package meta data is here
https://github.com/raspberrypi/rpi-eeprom/tree/debian/bookworm/debian