raspberrypi/rpi-eeprom #497: Raspberry Pi 5 PMIC does not negotiate 5A current via USB-PD PPS profile

l3iggs, 2023-11-21:
### Describe the bug

The raspberry Pi 5 PMIC does not make use of 5000 mA current offered by a charger via USB-PD PPS profile. Instead, it gets stuck at 3000mA even when a charger that advertises a PPS PDO of 3.3V - 20V @ 5A (even when a 5A e-marked cable is used).

### Steps to reproduce the behaviour

1) Buy a decent 3rd party USB type-c charger. For example, Anker 150W 747 Gan Prime. This charger happens to advertise the following PDOs [1]:
[image]
2) Power a rpi 5 with it using a properly e-marked 5A cable.
3) Inspect the bootloader's `power:` line. It will show the power source as being only `3000 mA` capable

As you can see in the image above, the charger offers a PPS PDO profile of 3.3 - 11V at 5A. 5V @ 5A falls squarely in that range and the pi5 should make use of this.

\[1]: https://www.chargerlab.com/teardown-of-anker-150w-ganprime-747-charger-a2340/

### Device (s)

Other

### Bootloader configuration.

versions 30de0ba5 2023/10/30 and earlier

timg236, 2023-11-21:
This is expected. PPS is not supported.

greyltc, 2023-11-21:
So you're saying there's a chance! (at some point) :-)

timg236, 2023-11-21:
> So you're saying there's a chance! (at some point) :-)

Maybe ... not an expert on PPS, IIRC there are some hardware constraints plus more time-critical messaging to and from the power supply which will cause an unexplained reset (go to 'safe' 0V) if messaging goes wrong which is a disaster if the sink doesn't have a battery.

greyltc, 2023-11-21:
oof yeah. too bad the safe fallback wasn't something like 5V@3A

greyltc, 2023-11-21:
@timg236 I've got a Satechi model ST-TC100GM charger which has a 5V fixed 3A PDO and a PPS 3.3-20V @ 5A one. When the time comes, I'd be willing to help in testing out any pre-alpha firmwares you might have for this feature. I can recover from bad power up negotiations with a benchtop power supply.

timg236, 2023-11-21:
@greyltc

timg236, 2023-11-21:
@greyltc I've re-opened this as a feature request so that anyone interested can track it. Warning, it's probably 50:50 whether this is actually doable :)

dev-drprasad, 2023-11-23:
@timg236 is there any cheaper 5V/5A charger works with Pi5 apart from official one that we can purchase ?

timg236, 2023-11-23:
> @timg236 is there any cheaper 5V/5A charger works with Pi5 apart from official one that we can purchase ?

Sorry, I've no idea

timg236, 2023-11-23:
Removed off-topic comments about product recomendations

TudbuT, 2024-03-24:
Just to add my own 2 cents to this:

I've been observing a lot of voltage-related throttling on all of my non-official power supplies, so I checked with a lab bench PSU and it actually never exceeds 2A of current draw without anything plugged into the USB ports - it shouldn't be receiving too low a voltage. But it is, because of cable resistance. My cables are all unmarked and one of them is long but of good quality, and the rest are shorter but of dubious quality, so it's no wonder they're all pretty high resistance.

Indeed, I also have another 3A power supply that has a built-in high quality cable, and it gives me absolutely no issues. The power supplies therefore aren't even the main issue when it comes to throttling (that is, unless you have a bunch of stuff on your USB ports) - the cables are. (And they really are a big one: I'm seeing things like 4.5V despite the PSU still being at 5.)

Using PPS, the pi could ask for 5.1 or 5.2 volts, which would make the voltage drop on the cable less of a problem. It could even regulate the voltage dynamically depending on load, though I'm not sure how realistic that part would be. Some power supplies with fixed cables might even do this on their own, but I have no idea if that's actually the case.

greyltc, 2024-03-24:
> Using PPS, the pi could ask for 5.1 or 5.2 volts
The official 27W one runs at 5.1V and has a permanently attached cable, I'm sure those design choices are for mitigating the voltage drop over the cabling. 5A is a lot. Using PPS to request a flat 5.2V no matter what seems like it would be a good idea to me. It's too bad the USB(/PD) spec didn't include some sort of provision for proper remote voltage sensing so that the source voltage supply could automatically and universally compensate for the voltage drop in the cables.

actraiser30, 2024-09-06:
That's why everytime I use a flash drive the os freezes?

lurch, 2024-09-06:
@actraiser30 For help with troubleshooting flash drives, you should ask for help on the [forums](https://forums.raspberrypi.com/), rather than on GitHub.

fanoush, 2024-10-12:
> it gets stuck at 3000mA even when a charger that advertises a PPS PDO of 3.3V - 20V @ 5A (even when a 5A e-marked cable is used).

just wondering what that means exactly.  if it can otherwise  do 5V/5A via PPS will it really limit the current if the device draws more than 3A at negotiated  5V mode? For pi5 the limit can be overriden in config.txt via `usb_max_current_enable=1`

actraiser30, 2024-10-12:
Big issue with original psu even not enought current for usb ports still is not fixed.

timg236, 2024-10-14:
The feature request for USD-PD PPS is noted. Locking the issue to comments because the forums are a more appropriate place to have tech support discussions about USB-PD.

timg236, 2026-08-21:
Closing NGTF.

closed 2026-08-21 by timg236