raspberrypi/rpi-eeprom #501: RPi5 active cooler fan starts full-throttle

HiassofT, 2023-11-30:
### Describe the bug

When powering up the RPi5 with an active cooler the fan runs full-throttle, being annoyingly noisy like 19" rack servers, until the kernel pwm-fan driver is finally loaded which puts it to silent/off mode.

Especially when netbooting (and having CONFIG_SENSORS_PWM_FAN set to "m" in kernel config) the fan is noisy for a very long time.

IMHO it would be better to initialize the fan PWM output to "off" (or at most a very low level) as early-stage boot shouldn't be too challenging for the CPU/VPU and having the fan running shouldn't be necessary.

### Steps to reproduce the behaviour

Power-up an RPi5 with official active cooler installed and fan connected

### Device (s)

Other

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=1
BOOT_ORDER=0xf421

### System

```
root@raspberrypi:~# vcgencmd bootloader_version
2023/11/20 19:40:17
version 361205c60f6077e57aa9b01e9d7d93d79c921dad (release)
timestamp 1700509217
update-time 1701386666
capabilities 0x0000007f

root@raspberrypi:~# cat /etc/rpi-issue 
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
root@raspberrypi:~# vcgencmd bootloader_version
2023/11/20 19:40:17
version 361205c60f6077e57aa9b01e9d7d93d79c921dad (release)
timestamp 1700509217
update-time 1701386666
capabilities 0x0000007f

root@raspberrypi:~# vcgencmd version
2023/11/20 19:40:17 
Copyright (c) 2012 Broadcom
version 361205c6 (release) (embedded)
root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

pelwell, 2023-12-03:
I kind of like this behaviour - it lets you know that the fan is working. It also blows any dust out of the systemas in the rack-mount servers you refer to.

regorxxx, 2023-12-12:
Having some side-benefits for a bug doesn't make it a feature. 

This should be fixed; another thing is if there should be some kind of configuration for it.

pelwell, 2023-12-12:
> Having some side-benefits for a bug doesn't make it a feature.

The fact that you don't like it doesn't make it a bug.

closed 2023-12-12 by pelwell

regorxxx, 2023-12-12:
When there are tons of reports on forums by regular users asking why the fan seems a rocket at init, without any real reason for the final user, with low temps... that looks to me a bug or an unwanted behavior.

Your argument can be used with your own answer, you like it, that doesn't make it "right". And most regular users DON'T use raspi for rack servers. They just want a silent thing on their room.

Closing this issue is totally arbitrary.