raspberrypi/rpi-eeprom #502: Auto-detect support for PCIe expansion HAT - missing specs

mikegapinski, 2023-12-04:
### Describe the bug

Hi,

I am wondering how we can make our HAT with NVMe automatically detected by the Pi 5. Before updating we had to change the boot order and now we also need to force the PCIe enumeration with the new flag.

We have the EEPROM on our board and it is connected to both the I2C on the pcie connector and the HAT ID pins. 

I know that the specs are not our yet but I figured I'd ask before trying to reverse-engineer the detection logic used in the upcoming official board. We are shipping a huge batch of boards this week and it is the last moment for us to put something in the EEPROM.

I initially figured that the bootloader will at some point automatically enumerate the pcie bus when there is no USB/SD Card present, this is a much cleaner solution but there has to be a good reason why it does not work like this.  

### Steps to reproduce the behaviour

1. Connect a "unofficial NVMe HAT/HAB"
2. The pi should boot from it without changing the EEPROM config

### Device (s)

Other

### Bootloader configuration.

NVMe added as a first device in the boot order
PCIE_PROBE added after updating to the current eeprom

### System

Bookworm

### Bootloader logs

Unsupported Boot Order 6

### NVMe boot

NVMe is detected and the pi boots from it with PCIE_PROBE enabled

timg236, 2023-12-04:
The answer is that a spec will be published and this Issue won’t be the spec because that will risk people wasting time and money on the wrong thing.

closed 2023-12-04 by timg236

mikegapinski, 2023-12-04:
@timg236  Hmm, that was harsh. I think you misunderstood my intention. I am trying to figure out a way for everyone in the market to use the automatic detection since it is very much technically possible.

We'll get it working regardless, thank you for your answer

surak, 2023-12-04:
+1 here

timg236, 2023-12-04:
That’s exactly why a spec will be published and there is no change to the boot order stuff

mikegapinski, 2023-12-04:
I am not denying that the specs will be out at some point... But what is the point of publishing bootloader changes that target the not yet existing spec that already affects the community? People who already got their hands on NVMe boards.

Since the bootloader support is already there what is holding back the release of the specs(at least this part)? A few companies are already pumping out boards that work fine and it would be great if we could benefit from the existing mechanism. 

I won't persue this topic any further. Please consider this issue as closed, there is no point in continuing this conversation

timg236, 2023-12-04:
Preliminary changes which are subject to change because at the time the spec wasn’t finalised.
I believe it is close to being finalised.

pelwell, 2023-12-04:
> Since the bootloader support is already there

And that's where you're wrong - there is enough bootloader support for _something_ to work _now_, but _not everything_, and _perhaps not forever_.

As has been stated, the spec is almost finished, but until it's published it is subject to change. If, in order to support the full vision of PCIe expansion HATs, it is necessary to do something that happens to stop prototypes working, then _we will do that_. If we were to describe what we've done so far than a reader might reasonably expect it to remain supported in the final spec, we would lose some freedom, and everyone would lose out.