raspberrypi/rpi-eeprom #505: CM4 Lite 8GB with KBG40ZNT256G TOSHIBA MEMORY 256,0 GB

DigitalSmile, 2023-12-09:
### Describe the bug

I have 6 same NVME drives, but only two are able to boot with Waveshare CM4-IO-BASE-B rev4 (I have 4 of them, tried on each). There are no differences (except manufacturing date, firmware is the same). The strange thing is they are all can succesfully boot from usb-pcie adapter and works just fine. It can be viewed in windows/linux as a drive both within pcie on motherboard and USB adapter.
The two working cards can be booted without any issues anytime. Others simply don't.
I have power adapter that shows voltage/ampers and all looking good. Cards themselves consumes 3,3V 1.1A

Bootlog sometimes say 'Timeout ...'  and some times just 'Error 8' and failed to load.

### Steps to reproduce the behaviour

1) Update rpi-eeprom to the latest version
2) Do not touch any boot configs, order, etc
3) Succesfully flash the firmware
4) Use rpi-imager to flash the drive with usb-pcie adapter (or with pci-e adapter on motherboard). Tried vanilla raspberry and Ubuntu Server.
5) Put  it in CM4-IO-BASE-B rev4
6) Tried to boot with no success (see booltog)

### Device (s)

Raspberry Pi CM4 Lite

### Bootloader configuration.

Vanilla

### Bootloader logs

RPi: BOOTLOADER release VERSION:4fd8f1f3 DATE: 2023/05/11 TIME: 07:26:03
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1683786363 serial 24497a6d boardrev a03140 stc 478846
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
DDR 3200 1 0 64 152

Boot mode: SD (01) order f2564
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: USB-MSD (04) order f256
USB xHC init failed
Boot mode: NVME (06) order f25
Timeout 00008000 00000000 00000000 00000000
nvme: error 8
Failed to open device: 'nvme'
Retry NVME 1
Timeout 00008000 00000000 00000000 00000000
nvme: error 8
Failed to open device: 'nvme'

lurch, 2023-12-10:
Is there any chance it's related to one of these other issues?
https://github.com/search?q=repo%3Araspberrypi%2Frpi-eeprom+waveshare&type=issues

DigitalSmile, 2023-12-14:
> Is there any chance it's related to one of these other issues? https://github.com/search?q=repo%3Araspberrypi%2Frpi-eeprom+waveshare&type=issues

Thanks for the tip, but I don't think those are related, since I cannot boot at all with those cards...
I understand the idea of having rootfs on NVMe and bootloader on SD card, though it seems to be just a workaround, but not  the solution.
Am I understand correctly, that the bootloader itself is closed source? Any chance I can play with source and some local debug?

pelwell, 2023-12-14:
Yes - the bootloader is closed source and will remain so, containing as it does IP from Broadcom.

DigitalSmile, 2023-12-14:
> Yes - the bootloader is closed source and will remain so, containing as it does IP from Broadcom.
Thanks...
Anyway, can you please describe the logic behind bootloader? It's definitely related to card <-> ARM chip peripherals communications...
Are you probing the devices before loading device tree? Those 'Timeout' errors are looking strange to me, since it should work for generic purpose cards. Is there any chance, that I need to dig into device tree configuration rather than bootloader code?

timg236, 2023-12-14:
The bootloader is completely independent of Linux. When debugging hardware / interop issues it's best to boot Linux from SD and stress test the storage device first from Linux. If it doesn't work there then there is no chance of it working in the VPU firmware.

DigitalSmile, 2023-12-14:
> The bootloader is completely independent of Linux. When debugging hardware / interop issues it's best to boot Linux from SD and stress test the storage device first from Linux. If it doesn't work there then there is no chance of it working in the VPU firmware.

I will try the workaround with SD card, but high chances are all will be working good, since it is working fine with USB

DigitalSmile, 2023-12-24:
Okay, so I managed to boot up with SD card and M2 card is just another partition int the system.
First of all, sometimes it goes with kernel panic in the function of reading/writing pcie configuration (sorry did not make any photo).
When I am able to boot, I have an AER spam in the logs. I have added two options to cmdline to test, but no effect (`pcie_aspm=off` and `pci=nommconf` found in internet with related symptoms)
```
ds@raspberrypi:~ $ dmesg | grep pci
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:BA:9D:97 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 pcie_aspm=off pci=nommconf root=PARTUUID=7125204e-02 rootfstype=ext4 fsck.repair=yes rootwait
[    1.224071] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.224117] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.224209] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.224300] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x01ffffffff -> 0x0400000000
[    1.224856] brcm-pcie fd500000.pcie: setting SCB_ACCESS_EN, READ_UR_MODE, MAX_BURST_SIZE
[    1.225310] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.225341] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.225369] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.225453] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.225689] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.229335] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.229577] pci_bus 0000:01: supply vpcie3v3 not found, using dummy regulator
[    1.229737] pci_bus 0000:01: supply vpcie3v3aux not found, using dummy regulator
[    1.229823] pci_bus 0000:01: supply vpcie12v not found, using dummy regulator
[    1.340066] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.340160] pci 0000:01:00.0: [1e0f:0001] type 00 class 0x010802
[    1.340244] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    1.340614] pci 0000:01:00.0: PME# supported from D0 D3hot
[    1.340774] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    1.341115] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.341169] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.341203] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600003fff 64bit]
[    1.341258] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.341288] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.341685] pcieport 0000:00:00.0: enabling device (0000 -> 0002)
[    1.341896] pcieport 0000:00:00.0: PME: Signaling with IRQ 31
[    1.342371] pcieport 0000:00:00.0: AER: enabled with IRQ 31
[    1.342733] brcm-pcie fd500000.pcie: clkreq control enabled
[    1.398742] nvme nvme0: pci function 0000:01:00.0
[    1.540473] pcieport 0000:00:00.0: AER: Multiple Corrected error received: 0000:00:00.0
[    1.540639] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[    1.540727] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[    1.540807] pcieport 0000:00:00.0:    [12] Timeout               
[    2.192269] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[    2.194929] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[    2.197533] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[    2.200170] pcieport 0000:00:00.0:    [12] Timeout               
[    2.513331] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[    2.516002] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[    2.518669] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[    2.521320] pcieport 0000:00:00.0:    [12] Timeout               
[   64.588782] pcieport 0000:00:00.0: AER: Multiple Corrected error received: 0000:00:00.0
[   64.591548] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[   64.596997] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[   64.602533] pcieport 0000:00:00.0:    [12] Timeout               
[  198.321270] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[  198.321321] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[  198.321335] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[  198.321352] pcieport 0000:00:00.0:    [12] Timeout               
[  198.390693] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[  198.390738] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[  198.390749] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00001000/00002000
[  198.390761] pcieport 0000:00:00.0:    [12] Timeout               
[  198.394743] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[... cut ...]
		Max no snoop latency: 0ns
	Capabilities: [300 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: LaneErr at lane: 0
	Capabilities: [400 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1- ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
			  PortCommonModeRestoreTime=60us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
```

Any hints or direction where to go from there?

timg236, 2024-01-02:
Unfortunately those errors indicate that it’s not going to work and would not be fixable with software.

DigitalSmile, 2024-01-07:
Yes, I recently gave up and bought Transcend TS256GMTE300S cards and they worked like a charm.
For future explorers: it seems the problem lies between driver software, firmware and hardware. In certain circumstances it can work, e.g. they work great on my PC both with usb adapter and pcie x1 on motherboard.

closed 2024-01-07 by DigitalSmile