raspberrypi/rpi-eeprom #508: SSD failing to boot in USB-3 slots (it boots in USB-2 slots)

sudo-splinter-cell, 2023-12-18:
### Describe the bug

Device is Pi 5.

My boot drive (sata SSD, plugged with a sata-usb adapter) fails to boot in Pi-5 when connected to a usb-3 slot. It boots perfectly fine when connected to a usb-2 slot.

### Steps to reproduce the behaviour

Plug the ssd to a usb-3 port and boot. Here is the boot screen after that:

[image]

### Device(s)

Other

### Kernel logs

The following is the output for when the drive is connected to a USB-2 port, as the system does not start when it is in USB-3.

[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:C0:AA:6C vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=88a1d4e6-02 rootfstype=ext4 fsck.repair=yes zswap.enabled=1 usbhid.mousepoll=0 rootwait
[    0.799164] usbcore: registered new interface driver usbfs
[    0.804680] usbcore: registered new interface driver hub
[    0.810021] usbcore: registered new device driver usb
[    0.815191] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    1.340768] usbcore: registered new interface driver r8152
[    1.346298] usbcore: registered new interface driver lan78xx
[    1.351996] usbcore: registered new interface driver smsc95xx
[    1.364017] usbcore: registered new interface driver uas
[    1.369371] usbcore: registered new interface driver usb-storage
[    1.454154] usbcore: registered new interface driver usbhid
[    1.459756] usbhid: USB HID core driver
[    2.041160] xhci-hcd xhci-hcd.0: new USB bus registered, assigned bus number 1
[    2.070524] xhci-hcd xhci-hcd.0: new USB bus registered, assigned bus number 2
[    2.078207] xhci-hcd xhci-hcd.0: Host supports USB 3.0 SuperSpeed
[    2.084825] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    2.093578] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.101263] usb usb1: Product: xHCI Host Controller
[    2.106587] usb usb1: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.113572] usb usb1: SerialNumber: xhci-hcd.0
[    2.118621] hub 1-0:1.0: USB hub found
[    2.127504] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    2.136244] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.143929] usb usb2: Product: xHCI Host Controller
[    2.149323] usb usb2: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.156307] usb usb2: SerialNumber: xhci-hcd.0
[    2.161307] hub 2-0:1.0: USB hub found
[    2.237130] xhci-hcd xhci-hcd.1: new USB bus registered, assigned bus number 3
[    2.266630] xhci-hcd xhci-hcd.1: new USB bus registered, assigned bus number 4
[    2.274390] xhci-hcd xhci-hcd.1: Host supports USB 3.0 SuperSpeed
[    2.280997] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    2.289732] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.297419] usb usb3: Product: xHCI Host Controller
[    2.302745] usb usb3: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.309730] usb usb3: SerialNumber: xhci-hcd.1
[    2.314739] hub 3-0:1.0: USB hub found
[    2.323624] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    2.332353] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.340027] usb usb4: Product: xHCI Host Controller
[    2.345392] usb usb4: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.352368] usb usb4: SerialNumber: xhci-hcd.1
[    2.357362] hub 4-0:1.0: USB hub found
[    2.384328] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    2.549936] usb 1-1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice= 6.56
[    2.558609] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.566199] usb 1-1: Product: USB2.1 Hub
[    2.579036] usb 3-2: new high-speed USB device number 2 using xhci-hcd
[    2.586026] usb 1-1: Manufacturer: GenesysLogic
[    2.628972] hub 1-1:1.0: USB hub found
[    2.675654] usb 2-1: new SuperSpeed USB device number 2 using xhci-hcd
[    2.702646] usb 2-1: New USB device found, idVendor=05e3, idProduct=0626, bcdDevice= 6.56
[    2.711292] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.718900] usb 2-1: Product: USB3.1 Hub
[    2.727887] usb 2-1: Manufacturer: GenesysLogic
[    2.748725] usb 3-2: New USB device found, idVendor=13fd, idProduct=3960, bcdDevice= 1.10
[    2.757420] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.757424] usb 3-2: Product: External
[    2.757429] usb 3-2: Manufacturer: Generic 
[    2.757431] usb 3-2: SerialNumber: 3136333732353434313630382020202020202020
[    2.789407] usb-storage 3-2:1.0: USB Mass Storage device detected
[    2.796182] scsi host0: usb-storage 3-2:1.0
[    2.821335] hub 2-1:1.0: USB hub found
[    2.847591] usb 1-2: new low-speed USB device number 3 using xhci-hcd
[    3.013004] usb 1-2: New USB device found, idVendor=1c4f, idProduct=0002, bcdDevice= 1.50
[    3.021953] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.029576] usb 1-2: Product: USB Keyboard
[    3.034735] usb 1-2: Manufacturer: SIGMACHIP
[    3.092909] input: SIGMACHIP USB Keyboard as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.0/0003:1C4F:0002.0001/input/input1
[    3.115539] usb 1-1.1: new full-speed USB device number 4 using xhci-hcd
[    3.167711] hid-generic 0003:1C4F:0002.0001: input,hidraw0: USB HID v1.10 Keyboard [SIGMACHIP USB Keyboard] on usb-xhci-hcd.0-2/input0
[    3.188852] input: SIGMACHIP USB Keyboard Consumer Control as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.1/0003:1C4F:0002.0002/input/input3
[    3.261541] usb 1-1.1: New USB device found, idVendor=1ea7, idProduct=0064, bcdDevice= 2.00
[    3.269939] usb 1-1.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    3.277302] usb 1-1.1: Product: 2.4G Mouse
[    3.281456] input: SIGMACHIP USB Keyboard System Control as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.1/0003:1C4F:0002.0002/input/input4
[    3.297419] hid-generic 0003:1C4F:0002.0002: input,hidraw1: USB HID v1.10 Device [SIGMACHIP USB Keyboard] on usb-xhci-hcd.0-2/input1
[    3.348375] input: 2.4G Mouse as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:1EA7:0064.0003/input/input6
[    3.362402] input: 2.4G Mouse as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:1EA7:0064.0003/input/input7
[    3.376425] hid-generic 0003:1EA7:0064.0003: input,hiddev96,hidraw2: USB HID v1.10 Mouse [2.4G Mouse] on usb-xhci-hcd.0-1.1/input0
[    7.645634] usbcore: registered new interface driver brcmfmac

timg236, 2023-12-18:
Please provide details of the driver / SATA adapter and verify that the driver behaves correctly when connected to the USB-3 when the Pi is booted from SD. E.g. copy large files to / from the mounted drive.

sudo-splinter-cell, 2023-12-18:
> Please provide details of the driver / SATA adapter and verify that the driver behaves correctly when connected to the USB-3 when the Pi is booted from SD. E.g. copy large files to / from the mounted drive.

Thanks for the attention. I purchased a new sata-usb adapter today. It worked out of the box as usb-3. The "non-working" adapter probably had one of its usb-3 wires busted so it was only working as usb-2. It was an "initio inic-3639" by the way. I discarded it. The new one i bought is again an initio (there are a lot of these chips around i guess) but this time it is inic-3619PL model.

Anyway, sorry for the confusion. I will close this as resolved.

closed 2023-12-18 by sudo-splinter-cell

timg236, 2023-12-18:
Thanks for taking the time to update the issue. Mystery solved :)