raspberrypi/rpi-eeprom #518: pi5 thermal throttling raises clock

cleverca22, 2024-01-12:
### Describe the bug

if you switch cpufreq to manual mode:
```
echo userspace > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
echo > /sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed 2000000
```
the system will then be locked to 2ghz

if you then cause thermal throttling with for example `stress-ng --cpu 4` and a lack of heatsink
the system will RAISE! the clock
```
root@raspberrypi:~# vcgencmd measure_clock arm ; vcgencmd measure_temp
frequency(0)=2000005120
temp=76.8'C
root@raspberrypi:~# vcgencmd measure_clock arm ; vcgencmd measure_temp
frequency(0)=2000006656
temp=77.4'C
root@raspberrypi:~# vcgencmd measure_clock arm ; vcgencmd measure_temp
frequency(0)=2256007936
temp=82.9'C
```

i suspect the problem, is that the soft-throttling uses a percentage of the max clock, rather then a percentage of what cpufreq requested

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```
root@raspberrypi:~# vcgencmd bootloader_config
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf41
```

### System

```
root@raspberrypi:~# vcgencmd bootloader_version
2024/01/05 15:57:40
version 30cc5f37d65b279e820adb1d9840ad3c1cc98229 (release)
timestamp 1704470260
update-time 1704905906
capabilities 0x0000007f
```

timg236, 2024-02-16:
Reproduced. The soft-limit scaling was based arm_freq config.txt rather than the current max freq.

timg236, 2024-05-01:
Closing as fixed

closed 2024-05-01 by timg236